betaflight/betaflight #12193: Custom cuild won't allow ESC protocol and PWM for servos

ramiss, 2023-01-14:
### Describe the bug

Since the cloud/custom build only lets you choose one motor protocol there is no way to use a servo on a quad.  Please allow us to choose a protocol and PWM.

### To Reproduce

1. Go to Update Firmware
2. Turn off Classic Mode
3. Choose Motor Protocol drop down.
4. Can't choose ESC protocol of choice and PWM.

### Expected behavior

Allow us to choose ESC protocol of choice and PWM.

For now I have worked around this by using Classic Mode, which is what the Support data will show.

### Support ID

```txt
Id: "e3c077cd-89ac-4fb5-bfa8-ac785865a409"
```

### Diff configuration

```txt
diff all

# version
# Betaflight / STM32F745 (S745) 4.4.0 Jan 13 2023 / 23:01:33 (2417d50d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name FLYWOOF745NANO
manufacturer_id FLWO
mcu_id 003a00413039511638313632
signature 

# resources
resource MOTOR 5 NONE
resource SERVO 1 C09

# feature
feature -LED_STRIP
feature SERVO_TILT
feature GPS
feature TELEMETRY
feature CHANNEL_FORWARDING

# serial
serial 0 64 115200 57600 0 115200
serial 1 2 115200 57600 0 115200
serial 2 1 115200 57600 0 115200
serial 4 1 115200 57600 0 115200

# servo
servo 0 1000 2000 1500 100 8

# beeper
beeper -GPS_STATUS
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE

# aux
aux 0 0 0 1725 2100 0 0
aux 1 1 1 900 1150 0 0
aux 2 46 1 1900 2100 0 0
aux 3 13 2 1850 2100 0 0
aux 4 35 2 1300 1700 0 0

# adjrange
adjrange 0 0 3 900 2100 29 3 0 0

# master
set acc_trim_pitch = -1
set acc_calibration = 97,-54,-100,1
set baro_hardware = AUTO
set serialrx_provider = CRSF
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT300
set bat_capacity = 850
set ibata_scale = 380
set small_angle = 180
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_ground_speed = 3000
set gps_rescue_min_sats = 6
set gps_rescue_allow_arming_without_fix = ON
set pid_process_denom = 16
set osd_rssi_alarm = 40
set osd_cap_alarm = 5000
set osd_vbat_pos = 384
set osd_rssi_pos = 474
set osd_link_quality_pos = 2532
set osd_rssi_dbm_pos = 399
set osd_tim_1_pos = 53
set osd_tim_2_pos = 2519
set osd_flymode_pos = 2508
set osd_throttle_pos = 6592
set osd_vtx_channel_pos = 16
set osd_crosshairs_pos = 236
set osd_ah_pos = 4206
set osd_current_pos = 247
set osd_mah_drawn_pos = 6635
set osd_craft_name_pos = 4160
set osd_gps_speed_pos = 271
set osd_gps_lon_pos = 4113
set osd_gps_lat_pos = 4096
set osd_gps_sats_pos = 6144
set osd_home_dir_pos = 4133
set osd_home_dist_pos = 4128
set osd_compass_bar_pos = 265
set osd_altitude_pos = 6149
set osd_pid_roll_pos = 135
set osd_pid_pitch_pos = 167
set osd_pid_yaw_pos = 199
set osd_debug_pos = 0
set osd_power_pos = 320
set osd_pidrate_profile_pos = 344
set osd_warnings_pos = 2080
set osd_avg_cell_voltage_pos = 6641
set osd_pit_ang_pos = 256
set osd_rol_ang_pos = 288
set osd_battery_usage_pos = 391
set osd_disarmed_pos = 2112
set osd_nheading_pos = 310
set osd_nvario_pos = 278
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_displayport_device = MSP
set osd_canvas_width = 30
set osd_canvas_height = 16
set vcd_video_system = PAL
set gyro_1_align_yaw = 2700

profile 0

# profile 0
set vbat_sag_compensation = 100

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

### Resources configuration

```txt
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: LED 1
A03: FREE
A04: FLASH_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: PINIO 1
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: SERVO 1
C10: FREE
C11: FREE
C12: SERIAL_TX 5
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: SERIAL_TX 2
D06: SERIAL_RX 2
D07: FREE
D08: SERIAL_TX 3
D09: SERIAL_RX 3
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : DSHOT_BITBANG 2
    CH3 : DSHOT_BITBANG 5
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH4 : SERVO 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 4
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: DSHOT_BITBANG 5
DMA2 Stream 7: FREE

#
```

### Flight controller

FLYWOONANO

### Other components

DJI 03 AU, Crossfire Nano RX, RC Switch

### How are the different components wired up

I'm using an RC switch on Motor 5 to control power to the DJI VTX

### Add any other context about the problem that you think might be relevant here

Please note that all cli outputs are showing my workaround of using Classic mode.

haslinghuis, 2023-01-14:
Please add `PWM` using Custom Defines.

ramiss, 2023-01-14:
> Please add `PWM` using Custom Defines.

The context help for Custom Defines implies that this would only work if the firmware developer allowed this option.  Surely, since PWM servo output is communicated through motor pins on a quad, and because PWM is already listed in the Motor Protocol dropdown, it is much more intuitive to configure PWM from that dropdown?

ramiss, 2023-01-14:
And are you saying that blindly typing PWM in the Custom Defines would always allow PWM/Servo use regardless of the firmware?

haslinghuis, 2023-01-14:
Add `SERVOS` too (use space separator)

closed 2023-02-22 by github-actions[bot]