betaflight/betaflight #12194: on Classic mode  feature GPS is unavailable

palgy, 2023-01-14:
### Describe the bug

when updating the firmware on Classic mode the  feature GPS is unavailable and can not be configured on the configurator 
if not selecting  the  Classic mode the  feature GPS is working fine ( per my understanding the  Classic mode  should contain all the features)  

### To Reproduce

updating the firmware on Classic mode and try to enable the GPS 

### Expected behavior

classic mode  should contain all the features

### Support ID

```txt
n/a
```

### Diff configuration

```txt
# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 14 2023 / 10:58:28 (2417d50d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name SPEEDYBEEF7V2
manufacturer_id SPBE
mcu_id 0030002f3756501120383558
signature 

# name: AOS 5

# feature
feature -LED_STRIP

# serial
serial 1 131073 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1775 2100 0 0
aux 1 1 2 1800 2100 0 0
aux 2 13 4 1750 2100 0 0
aux 3 35 3 1700 2100 0 0

# vtxtable

# master
set acc_calibration = -18,10,-137,1
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1300
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set osd_cap_alarm = 1300
set osd_vbat_pos = 2433
set osd_rssi_pos = 129
set osd_link_quality_pos = 2113
set osd_rssi_dbm_pos = 2081
set osd_tim_1_pos = 2401
set osd_tim_2_pos = 2455
set osd_flymode_pos = 2381
set osd_throttle_pos = 2360
set osd_vtx_channel_pos = 2433
set osd_current_pos = 2209
set osd_mah_drawn_pos = 2433
set osd_craft_name_pos = 2081
set osd_gps_speed_pos = 2264
set osd_gps_lon_pos = 2136
set osd_gps_lat_pos = 2104
set osd_gps_sats_pos = 2167
set osd_home_dir_pos = 2190
set osd_home_dist_pos = 2200
set osd_flight_dist_pos = 2232
set osd_compass_bar_pos = 2058
set osd_altitude_pos = 2327
set osd_warnings_pos = 14633
set osd_avg_cell_voltage_pos = 2433
set osd_disarmed_pos = 2411
set osd_esc_tmp_pos = 2433
set osd_flip_arrow_pos = 2113
set osd_core_temp_pos = 2296
set osd_log_status_pos = 2145
set osd_efficiency_pos = 231
set osd_displayport_device = MSP
set debug_mode = GYRO_SCALED

set vcd_video_system = HD
set craft_name = AOS 5

profile 0

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

### Resources configuration

```txt
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: CAMERA_CONTROL
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 1
A09: MOTOR 2
A10: MOTOR 3
A11: USB
A12: USB
A13: SWD
A14: FLASH_CS
A15: PINIO 1
B00: FREE
B01: FREE
B02: BEEPER
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: SERIAL_TX 1
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: MOTOR 4
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1 : CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 1
    CH4 : DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 3
```

### Flight controller

SPEEDYBEEF7V2

haslinghuis, 2023-01-14:
Due to firmware size overflowing and ongoing development needs more space cloud build was introduced as alternative to discontinue boards with only 512K flash space.

Please use cloud build and select your radio protocol, FLASH, GPS, LED_STRIP, OSD (SD or HD),  PINIO (and VTX if needed) and you should be good to go.

REK07, 2023-01-16:
Used cloud build  Now gps works but know no gyro/acc found.  What a mess.  HAKRCF722D

haslinghuis, 2023-01-16:
@REK07 https://github.com/betaflight/unified-targets/pull/822

haslinghuis, 2023-02-20:
Fixed in https://github.com/betaflight/unified-targets/pull/822

closed 2023-02-20 by haslinghuis