betaflight/betaflight #12205: BF 4.4.0 RC4 & RC5 ELRS dynamic power always reads 0mW on OSD

aowi7280, 2023-01-17:
### Describe the bug

No dynamic power reading on the OSD (analog). Using latest ELRS and Dyn power. I selected Vx uplink power to show on the OSD, but it always shows 0mW. The correct power shows on myQX7 Tx screen and changes correctly, so the telemetry is coming back for TPW2.
Same problem in BF 4.3.2 
[image]

### To Reproduce

Flash, set OSD to show Tx uplink power. connect battery.

### Expected behavior

Correct Vtx power shows on analog OSD

### Support ID

```txt
couldn't find it
```

### Diff configuration

```txt
# Betaflight / STM32F405 (S405) 4.4.0 Jan 17 2023 / 05:46:57 (a7cf8f0c6) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id AIRB

# name: Androne

# resources
resource MOTOR 5 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 11 A01
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11

# feature
feature SOFTSERIAL
feature GPS
feature LED_STRIP

# serial
serial 0 64 115200 57600 0 115200
serial 5 2 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# led
led 0 6,0::A:0
led 1 7,0::A:0
led 2 8,0::A:0
led 3 6,1::A:0
led 4 7,1::A:0
led 5 8,1::A:0
led 6 6,2::A:0
led 7 7,2::A:0
led 8 8,2::A:0

# mode_color
mode_color 6 1 2

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1200 0 0
aux 2 2 1 1300 1700 0 0
aux 3 46 2 1700 2100 0 0
aux 4 13 2 1300 2100 0 0
aux 5 26 0 1700 2100 0 0
aux 6 35 3 1300 2100 0 0
aux 7 36 4 1700 2100 0 0

# adjrange
adjrange 0 0 5 900 2100 12 5 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FSHARK_F F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 14 26 29 32
vtxtable powerlabels 25 400 800 MAX

# vtx
vtx 0 6 0 0 1 900 1250
vtx 1 6 0 0 2 1275 1475
vtx 2 6 0 0 3 1500 1675
vtx 3 6 0 0 4 1700 2100

# rxfail
rxfail 3 s 1400
rxfail 4 s 2000
rxfail 5 s 1000
rxfail 6 s 1000
rxfail 10 s 2000

# master
set gyro_lpf1_static_hz = 325
set gyro_lpf2_static_hz = 650
set gyro_notch1_hz = 74
set gyro_notch1_cutoff = 67
set dyn_notch_count = 1
set dyn_notch_q = 500
set gyro_lpf1_dyn_min_hz = 325
set gyro_lpf1_dyn_max_hz = 650
set acc_trim_roll = 2
set acc_calibration = 51,46,-163,1
set align_mag = CUSTOM
set mag_align_roll = -200
set mag_align_pitch = 1800
set mag_align_yaw = 2700
set mag_calibration = 64,30,-292
set min_check = 1020
set max_check = 1970
set rssi_scale = 120
set fpv_mix_degrees = 10
set min_throttle = 1020
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_off_delay = 120
set failsafe_throttle = 1150
set failsafe_switch_mode = STAGE2
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1800
set vbat_min_cell_voltage = 320
set vbat_warning_cell_voltage = 340
set ibata_scale = 300
set yaw_motors_reversed = ON
set small_angle = 180
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_alt_mode = CURRENT_ALT
set gps_rescue_return_alt = 25
set gps_rescue_ground_speed = 600
set gps_rescue_pitch_angle_max = 30
set gps_rescue_throttle_hover = 1350
set gps_rescue_velocity_p = 4
set gps_rescue_velocity_i = 4
set gps_rescue_velocity_d = 4
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 2
set simplified_gyro_filter_multiplier = 130
set osd_units = IMPERIAL
set osd_warn_bitmask = 10925
set osd_cap_alarm = 3500
set osd_alt_alarm = 375
set osd_vbat_pos = 2432
set osd_rssi_pos = 6202
set osd_link_quality_pos = 6231
set osd_link_tx_power_pos = 2454
set osd_rssi_dbm_pos = 2168
set osd_tim_1_pos = 375
set osd_g_force_pos = 225
set osd_throttle_pos = 6336
set osd_vtx_channel_pos = 6517
set osd_ah_sbar_pos = 200
set osd_current_pos = 6496
set osd_mah_drawn_pos = 408
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 2100
set osd_gps_lon_pos = 18449
set osd_gps_lat_pos = 18432
set osd_gps_sats_pos = 6176
set osd_home_dir_pos = 14350
set osd_home_dist_pos = 6189
set osd_flight_dist_pos = 2087
set osd_compass_bar_pos = 2442
set osd_altitude_pos = 22744
set osd_debug_pos = 320
set osd_warnings_pos = 14697
set osd_avg_cell_voltage_pos = 14657
set osd_pit_ang_pos = 2272
set osd_disarmed_pos = 394
set osd_esc_tmp_pos = 288
set osd_esc_rpm_pos = 248
set osd_core_temp_pos = 343
set osd_stick_overlay_right_pos = 239
set osd_efficiency_pos = 320
set osd_sys_vtx_temp_pos = 311
set osd_stat_bitmask = 405272
set osd_menu_background = GRAY
set osd_canvas_height = 13
set debug_mode = GYRO_SCALED
set vtx_band = 4
set vtx_channel = 3
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5780
set vcd_video_system = NTSC
set gyro_1_align_yaw = 2700
set timezone_offset_minutes = -420
set stats_min_armed_time_s = 30
set stats_total_flights = 350
set stats_total_time_s = 78438
set stats_total_dist_m = 886844
set craft_name = Androne

profile 1

# profile 1
set profile_name = Standard
set dterm_lpf1_dyn_min_hz = 82
set dterm_lpf1_dyn_max_hz = 165
set dterm_lpf1_static_hz = 82
set dterm_lpf2_static_hz = 165
set vbat_sag_compensation = 100
set anti_gravity_gain = 100
set yaw_lowpass_hz = 0
set p_pitch = 56
set i_pitch = 70
set d_pitch = 80
set f_pitch = 149
set p_roll = 56
set i_roll = 70
set d_roll = 73
set f_roll = 151
set p_yaw = 80
set i_yaw = 50
set f_yaw = 135
set angle_level_strength = 40
set d_min_roll = 52
set d_min_pitch = 56
set thrust_linear = 20
set dyn_idle_min_rpm = 33
set simplified_pids_mode = RP
set simplified_master_multiplier = 140
set simplified_i_gain = 70
set simplified_d_gain = 125
set simplified_pi_gain = 90
set simplified_dmax_gain = 120
set simplified_feedforward_gain = 90
set simplified_pitch_d_gain = 95
set simplified_pitch_pi_gain = 95
set simplified_dterm_filter_multiplier = 110
set tpa_mode = PD

rateprofile 2

# rateprofile 2
set rateprofile_name = HiRates
set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 1
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50
set roll_srate = 73
set pitch_srate = 73
set yaw_srate = 3
```

### Resources configuration

```txt
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: SERIAL_TX 11
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: BARO_CS
B04: BEEPER
B05: LED 1
B06: LED_STRIP
B07: SDCARD_DETECT
B08: FREE
B09: FREE
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: INVERTER 6
C09: INVERTER 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH1 : LED_STRIP
TIM5:
    CH2 : SERIAL_TX 11
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
    CH3 : DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: SPI_MISO 3
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: ADC 2
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

Airbot OmnibusF4SD F405

### Other components

FC: Airbot OmnibusF4SD  F405
Tx: Frsky QX7 w_HappyModel 915MHz ES900TX ExpressLRS V3.1.2
Rx: R9MM w/ExpressLRS V3.1.2
Vtx: RDQ Mach 3 25-1000mW

TomAlperin, 2023-01-25:
Are you using wide channels on ELRS? The power is not sent to the receiver if you don’t have wide channels set.

aowi7280, 2023-01-25:
Using hybrid.

aowi7280, 2023-01-25:
But it comes to the radio. I can see it on the display, just not on the OSD.

haslinghuis, 2023-01-25:
@aowi7280 In CLI tab try `set expresslrs_switch_mode = WIDE` and `save`.

(Using CLI you can use the `Submit Support Data` button. Please insert the `ID` here).

```
/**
 * HybridWide switches decoding of over the air data
 *
 * Hybrid switches uses 10 bits for each analog channel,
 * 1 bits for the low latency switch[0]
 * 6 or 7 bits for the round-robin switch
 * 1 bit for the TelemetryStatus, which may be in every packet or just idx 7
 * depending on TelemetryRatio
 *
 * Output: crsf.PackedRCdataOut, crsf.LinkStatistics.uplink_TX_Power
 * Returns: TelemetryStatus bit
 */
static void unpackChannelDataHybridWide(uint16_t *rcData, volatile elrsOtaPacket_t const * const otaPktPtr)
{
    unpackAnalogChannelData(rcData, otaPktPtr);
    const uint8_t switchByte = otaPktPtr->rc.switches;

    // The round-robin switch, 6-7 bits with the switch index implied by the nonce. Some logic moved to processRFPacket
    if (wideSwitchIndex >= 7) {
        txPower = switchByte & 0x3F;
    } else {
        uint8_t bins;
        uint16_t switchValue;
        if (currTlmDenom < 8) {
            bins = 63;
            switchValue = switchByte & 0x3F; // 6-bit
        } else {
            bins = 127;
            switchValue = switchByte & 0x7F; // 7-bit
        }

        rcData[5 + wideSwitchIndex] = convertSwitchNb(switchValue, bins);
    }

    setRssiChannelData(rcData);
}
```

Is called from

```
void expressLrsSetRcDataFromPayload(uint16_t *rcData, const uint8_t *payload)
{
    if (rcData && payload) {
        volatile elrsOtaPacket_t * const otaPktPtr = (elrsOtaPacket_t * const) payload;
        rxExpressLrsSpiConfig()->switchMode == SM_WIDE ? unpackChannelDataHybridWide(rcData, otaPktPtr) : unpackChannelDataHybridSwitch8(rcData, otaPktPtr);
    }
}
```

In HYBRID mode we call
```
/**
 * Hybrid switches uses 10 bits for each analog channel,
 * 2 bits for the low latency switch[0]
 * 3 bits for the round-robin switch index and 2 bits for the value
 * 4 analog channels, 1 low latency switch and round robin switch data = 47 bits (1 free)
 * 
 * sets telemetry status bit
 */
static void unpackChannelDataHybridSwitch8(uint16_t *rcData, volatile elrsOtaPacket_t const * const otaPktPtr)
{
    unpackAnalogChannelData(rcData, otaPktPtr);

    const uint8_t switchByte = otaPktPtr->rc.switches;

    // The round-robin switch, switchIndex is actually index-1 
    // to leave the low bit open for switch 7 (sent as 0b11x)
    // where x is the high bit of switch 7
    uint8_t switchIndex = (switchByte & 0x38) >> 3;

    if (switchIndex >= 6) {
        // Because AUX1 (index 0) is the low latency switch, the low bit
        // of the switchIndex can be used as data, and arrives as index "6"
        rcData[11] = convertSwitchNb(switchByte & 0x0F, 15); //4-bit
    } else {
        rcData[5 + switchIndex] = convertSwitch3b(switchByte & 0x07);
    }

    setRssiChannelData(rcData);
}
```

TomAlperin, 2023-01-26:
> But it comes to the radio. I can see it on the display, just not on the OSD. Get Outlook for Android<https://aka.ms/AAb9ysg>
> […](#)

The value originates at the transmitter module, not the receiver. If you use hybrid channels, it’s in the telemetry coming from the module but it does not get sent to the receiver.

aowi7280, 2023-01-26:
@TomAlperin OK, I get that. The OSD shows the Tx power in wide mode. What is the downside of using wide versus hybrid? I am flying a quad, using standard telemetry rate. I don't see much difference (by looking at how often the * comes through in the Telemetry screen). Is it going to affect my packet rate (200Hz) or the rate telemetry comes back (1:64)?
As an add on question, If I delete unused sensors from the telemetry, will that speed anything up?
I appreciate all the work you guys do. Its pretty cool.
@haslinghuis 23f20e15-2b10-46dc-bf1b-5a3e9854bf45 The set expresslrs_switch_mode = WIDE came back with an error message. 
# set expresslrs_switch_mode = WIDE
###ERROR IN set: INVALID NAME: expresslrs_switch_mode = WIDE###

haslinghuis, 2023-01-26:
Using CRSF protocol and telemetry Seems not enough.

We need to add `RX_EXPRESSLRS RX_EXPRESSLRS_TELEMETRY` to custom defines to be able to use that.

@aowi7280 please check

haslinghuis, 2023-01-27:
@aowi7280 you now can choose ERLS (SERIAL) as Radio Protocol. Please check.

aowi7280, 2023-01-28:
OK, I finally figured it out. I set ExpressLRS as radio protocol with no custom defines. But It does not seem to behave any differently. I still have to set it to Wide in the ELRS lua. So I don't know what this does in the CLI.
# get expresslrs_switch_mode
expresslrs_switch_mode = WIDE
Allowed values: WIDE, HYBRID
[image]

BTW, I get this error when reloading my backup configuration.
[image]

aowi7280, 2023-01-28:
I am using UART ELRS and the Tx power was displayed before. I just had to set to wide in the ELRS lua. I don't see any difference in the betaflight function.

haslinghuis, 2023-01-28:
Difference is we are forced to removed features from firmware for ongoing development due to firmware overflow issues as radio protocol, options and motor protocols together with custom defines is a means to configure your build with the options you need.

closed 2023-01-28 by haslinghuis

ZzyzxTek, 2023-01-28:
> I am using UART ELRS and the Tx power was displayed before. I just had to set to wide in the ELRS lua. I don't see any difference in the betaflight function.

Setting wide mode in ELRS Lua enables ELRS to pack the TX power telemetry item into the packets being sent to the RX and on to Betaflight.

Using the `ELRS (SERIAL)` build option includes the ELRS drivers in the build, which enables Betaflight to pick the TX power telemetry out of the packets and make it available to the OSD.

The ELRS drivers used to be included by default in many targets, but has been trimmed back, that's why it used to work with just the `CRSF` build option. Now we need to use `ELRS (SERIAL)` to get the drivers, and apparently `set expresslrs_switch_mode = WIDE` to turn on wide mode parsing. This setting is default for all the targets I'm using, so I haven't needed to set it myself and haven't tried setting it to something else to verify it's needed (too lazy).

aowi7280, 2023-01-28:
Thanks, I appreciate all the info. and fixing the issues.

dmak, 2024-05-15:
> Yes thank you for reporting. This issue made us change radio protocol options to include WIDE option to be available for UART ELRS users as it's encoded in the SPI part of the code.
> 
> To enable you have to `set expresslrs_switch_mode = WIDE` and save & reboot.
> 
> OSD should now display TX Power.

Thanks for this information. I believe that `expresslrs_switch_mode` setting in BetaFlight and controller's (Lua script) "Switch Mode" value should be in sync?

Also in my case “Tx uplink power” OSD option started to work only when I set both settings to "Wide" (telemetry ratio is 1:4). "Hybrid" does not work for some reason 😪 (ELRS protocol limitations?)