betaflight/betaflight #12206: Telemetry items might be missing on 4.4 RC4 / RC5

hickscorp, 2023-01-17:
### Describe the bug

Good morning,

I've been testing 4.4 RC4 and RC5 for two days now - without flying - just making sure things seem ok.
I know this is the bug description section, but I'll start by mentioning how much I love this new release branch, it's adding things that I always wanted. Good job! And woohoo for MSP OSD!

So - it seems to me that some telemetry items are missing, or might have changed in a weird way.
The problem is that I can't really say for sure - I have set up telemetry alarms on my RX (TX16S ELRS) a long time ago, and I can't really remember how it looked in the telemetry sensors.

But it really seems some are missing - on my RX, I selected "Delete All" and "Discover New" and the amount of sensors are really low compared to before - a few critical ones that I use are missing - for example RxBatt and dynamic RX power.

Maybe it is a misconfiguration on my end, but I really think something changed.

### To Reproduce

Well - see bug description.

### Expected behavior

All sensors should be backward compatible, and use the same units - or if they changed there should be somewhere a mapping to let us know how to have equivalents.

### Support ID

```txt
`e6d5b6e4-cf62-4e0c-8a40-6bcab77d2f1e`
```

### Diff configuration

```txt
diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 17 2023 / 03:31:54 (a7cf8f0c6) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name TMOTORF7
manufacturer_id TMTR

# name: Bug

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource PINIO 1 B06

# feature
feature -TELEMETRY
feature SOFTSERIAL
feature GPS

# serial
serial 1 131073 115200 57600 0 115200
serial 2 2 115200 57600 0 115200
serial 3 1 115200 57600 0 115200
serial 4 64 115200 57600 0 115200

# beeper
beeper -ON_USB

# beacon
beacon RX_SET

# aux
aux 0 0 0 1950 2050 0 0
aux 1 1 1 950 1050 0 0
aux 2 27 2 1950 2050 0 0
aux 3 13 5 1450 2050 0 0
aux 4 40 4 1950 2050 0 0

# rxfail
rxfail 0 s 1500
rxfail 1 s 1500
rxfail 2 s 1500
rxfail 3 s 1225
rxfail 5 s 1000
rxfail 7 s 1000
rxfail 9 s 1000
rxfail 10 s 1000
rxfail 11 s 1000
rxfail 12 s 1000
rxfail 13 s 1000
rxfail 14 s 1000
rxfail 15 s 1000

# master
set acc_calibration = 14,15,-42,1
set max_check = 1950
set fpv_mix_degrees = 15
set blackbox_sample_rate = 1/8
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_delay = 40
set failsafe_switch_mode = STAGE2
set failsafe_throttle_low_delay = 10
set failsafe_procedure = GPS-RESCUE
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_descent_dist = 150
set gps_rescue_min_sats = 6
set pid_process_denom = 2
set report_cell_voltage = ON
set telemetry_disabled_fuel = ON
set telemetry_disabled_acc_x = ON
set telemetry_disabled_acc_y = ON
set telemetry_disabled_acc_z = ON
set telemetry_disabled_pitch = ON
set telemetry_disabled_roll = ON
set telemetry_disabled_heading = ON
set telemetry_disabled_vario = ON
set telemetry_disabled_distance = ON
set telemetry_disabled_esc_temperature = OFF
set osd_warn_bitmask = 15341
set osd_rssi_dbm_alarm = -90
set osd_vbat_pos = 2048
set osd_rssi_pos = 2304
set osd_link_quality_pos = 2336
set osd_link_tx_power_pos = 2368
set osd_rssi_dbm_pos = 192
set osd_tim_1_pos = 1044
set osd_flymode_pos = 2624
set osd_g_force_pos = 2208
set osd_throttle_pos = 2176
set osd_crosshairs_pos = 2396
set osd_ah_sbar_pos = 349
set osd_ah_pos = 2269
set osd_current_pos = 64
set osd_craft_name_pos = 1042
set osd_gps_speed_pos = 2152
set osd_gps_sats_pos = 2055
set osd_home_dir_pos = 2557
set osd_home_dist_pos = 2088
set osd_compass_bar_pos = 10
set osd_altitude_pos = 2120
set osd_pidrate_profile_pos = 288
set osd_warnings_pos = 14873
set osd_avg_cell_voltage_pos = 2080
set osd_disarmed_pos = 2592
set osd_stick_overlay_right_pos = 1255
set osd_rate_profile_name_pos = 2656
set osd_camera_frame_pos = 34
set osd_efficiency_pos = 301
set osd_stat_bitmask = 425788
set osd_displayport_device = MSP
set osd_canvas_width = 60
set osd_canvas_height = 22
set vcd_video_system = HD
set craft_name = Bug
set pilot_name = Doodloo

profile 0

# profile 0
set yaw_lowpass_hz = 0
set motor_output_limit = 95
set auto_profile_cell_count = 6

rateprofile 0

# rateprofile 0
set rateprofile_name = Low
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70

# end the command batch
batch end
```
```

### Resources configuration

```txt
I know that this might look weird - I've remapped long ago a motor output that I don't use so that I can control recording on my Insta360 Go 2 to `USER_1`... Apart from that, this should be pretty standard.

resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: LED 1
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: PINIO 1
B07: FREE
B08: FREE
B09: FREE
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: FLASH_CS
C09: FREE
C10: SERIAL_TX 3
C11: SERIAL_RX 3
C12: SERIAL_TX 5
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 3
    CH4 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 2
```
```

### Flight controller

TMOTOR7

### Other components

Matek GPS, Matek ELRS Diversity, Caddx Vista.

LoakAudio, 2023-01-17:
Hi @hickscorp ,
My 2 cents : in your diff you have Telemetry disabled
feature -TELEMETRY

and several specific parameters disabled too

set telemetry_disabled_fuel = ON
set telemetry_disabled_acc_x = ON
set telemetry_disabled_acc_y = ON
set telemetry_disabled_acc_z = ON
set telemetry_disabled_pitch = ON
set telemetry_disabled_roll = ON
set telemetry_disabled_heading = ON
set telemetry_disabled_vario = ON
set telemetry_disabled_distance = ON

hickscorp, 2023-01-17:
Hi @LoakAudio and thanks for the help!

Spot on. It was really just the `feature -TELEMETRY`. I issued a `feature TELEMETRY` and `save` and all my sensors started being discovered properly.

Although it fixed my issue and I will close it, I would like to mention that I never proactively disable the `TELEMETRY` feature - so something must have gone wrong (or did it without my knowledge) when I installed 4.4RC.

Thanks again!

closed 2023-01-17 by hickscorp