betaflight/betaflight #12207: GPS drift

druckgott, 2023-01-17:
### Describe the bug

Hello, I have a hugh drift on the GPS. I use BF 4.4 and if I switch to arming, the gps drifts away with around 10m.
I did a video:
https://studio.youtube.com/video/LmpTjc2sCYM/edit

### To Reproduce

1. Wait for getting GPS
2. Switch to arming
3. see how the value drifts without getting in the air

Components:
BN220 and Mamba F405 MK2

### Expected behavior

Do not drift away

### Support ID

```txt
Eindeutige Geräte ID: 0x4f0025425350032034354e (I didn´t see where to find this Button)
```

### Diff configuration

```txt
# diff

# version
# Betaflight / STM32F405 (S405) 4.4.0 Jan 14 2023 / 15:22:54 () MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id DIAT

# feature
feature GPS

# serial
serial 0 64 115200 57600 0 115200
serial 2 131073 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# map
map TAER1234

# aux
aux 0 0 2 900 1200 0 0
aux 1 1 0 900 1250 0 0
aux 2 2 0 1275 1700 0 0
aux 3 13 4 1500 2100 0 0

# rxfail
rxfail 3 h
rxfail 4 s 1000
rxfail 15 s 1000

# master
set gyro_lpf1_static_hz = 150
set gyro_lpf2_static_hz = 300
set dyn_notch_min_hz = 125
set dyn_notch_max_hz = 650
set gyro_lpf1_dyn_min_hz = 150
set gyro_lpf1_dyn_max_hz = 300
set acc_lpf_hz = 10
set acc_calibration = -78,0,7,1
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 16
set rssi_scale = 200
set fpv_mix_degrees = 20
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set motor_pwm_protocol = DSHOT300
set failsafe_switch_mode = STAGE2
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1000
set vbat_min_cell_voltage = 350
set ibata_scale = 274
set yaw_motors_reversed = ON
set small_angle = 180
set gps_rescue_min_sats = 5
set simplified_gyro_filter_multiplier = 60
set frsky_unit = IMPERIAL
set osd_rssi_alarm = 30
set osd_cap_alarm = 1000
set osd_vbat_pos = 12598
set osd_rssi_pos = 2109
set osd_flymode_pos = 3113
set osd_vtx_channel_pos = 363
set osd_crosshairs_pos = 65
set osd_ah_pos = 110
set osd_current_pos = 3208
set osd_mah_drawn_pos = 3145
set osd_gps_speed_pos = 3747
set osd_gps_lon_pos = 2121
set osd_gps_lat_pos = 2089
set osd_gps_sats_pos = 2140
set osd_home_dir_pos = 2365
set osd_home_dist_pos = 2749
set osd_home_point_pos = 2429
set osd_compass_bar_pos = 394
set osd_altitude_pos = 2741
set osd_warnings_pos = 14969
set osd_avg_cell_voltage_pos = 3177
set osd_disarmed_pos = 2714
set osd_nheading_pos = 349
set osd_stick_overlay_left_pos = 256
set osd_stick_overlay_right_pos = 256
set osd_sys_vtx_voltage_pos = 1574
set osd_sys_bitrate_pos = 1606
set osd_sys_delay_pos = 593
set osd_sys_vtx_temp_pos = 561
set osd_gps_sats_show_hdop = ON
set osd_displayport_device = MSP
set osd_canvas_width = 60
set osd_canvas_height = 22
set debug_mode = GYRO_SCALED
set vcd_video_system = HD
set gyro_1_align_yaw = 1800
set gyro_2_spibus = 1
set expresslrs_switch_mode = HYBRID

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 90
set dterm_lpf1_dyn_max_hz = 180
set dterm_lpf1_static_hz = 90
set dterm_lpf2_static_hz = 180
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set yaw_lowpass_hz = 0
set p_pitch = 69
set i_pitch = 124
set d_pitch = 45
set f_pitch = 219
set p_roll = 66
set i_roll = 118
set f_roll = 210
set p_yaw = 66
set i_yaw = 118
set f_yaw = 210
set d_min_roll = 40
set d_min_pitch = 45
set simplified_master_multiplier = 135
set simplified_pi_gain = 110
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 130
set simplified_dterm_filter_multiplier = 120

rateprofile 0

# end the command batch
batch end
```

### Resources configuration

```txt
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 1
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: BEEPER
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 2
B01: MOTOR 3
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: CAMERA_CONTROL
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 4
    CH4 : MOTOR 1
TIM3:
    CH3 : MOTOR 2
    CH4 : MOTOR 3
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11:
    CH1 : CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: MOTOR 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: MOTOR 1
DMA1 Stream 7: MOTOR 2
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

F405 MK2

### Other components

GPS: BN220

### How are the different components wired up

UART3 for GPS

KarateBrot, 2023-01-17:
In BF 4.5 we are going to fuse more sensors for positional data, so don't worry about it for now.

haslinghuis, 2023-01-17:
Need a RTK ground station :)

druckgott, 2023-01-17:
But it's really far away 🤔, I want to include the moving home point, but this this values it makes no sense 😭. Maybe I have to wait until. I implent the moving home point?

KarateBrot, 2023-01-17:
BF 4.5 development starts very soon. If you wanted to include this in 4.4 your PR came about 2.5 months too late 😀

OlegSyr, 2023-01-18:
@druckgott In your video you have only 7 sats and and very bad HDOP at 1.5.
I'd advice to wait until you have more than 10 sat and HDOP is less that 1.

Also BN220 is not the best module on the market. You can check recommended models here:
https://github.com/betaflight/betaflight/wiki/GPS-Rescue-for-4.4

I use  GPS module with bigger antennas like this one:
https://www.aliexpress.com/item/1005001892952440.html?spm=a2g0o.order_list.order_list_main.50.13dd1802cYDFoM
With TOPGNSS gps  I always have more than 20 to 30 sats and HDOP below 0.5. 

Your PR is great. Will be testing it as soon as my beater drone is fixed.

druckgott, 2023-01-18:
Ok yes its my 5 Zoll I didn't by a good one 😂,
I will test it with my 7 Zoll there I have a Matek M8Q-5883.
Then I have t switch the drone. I will come back with a new video with better gps 😉.
Thx for your information.

druckgott, 2023-01-18:
> BF 4.5 development starts very soon. If you wanted to include this in 4.4 your PR came about 2 months too late 😀

No I don't mean that this point will come to 4.5,i mean I have to wait until 4.5 is finished and then implement this point due to the point that there should be better output due to sensor fusion.

druckgott, 2023-01-19:
I now did a test with the better GPS and it looks much better, but maybe it makes sence to have not only the satellits as a point when Rescue is working. Maybe it makes also sence to add the HDOP Value as an optional criteria for people who want to have more exact values?!
https://youtu.be/znje1Wy6FoA

closed 2023-02-27 by github-actions[bot]