betaflight/betaflight #12210: GPS rescue landing mode limit to 10 seconds may cause the plane disarmed in the air during landing

SadalmelikJ, 2023-01-17:
### Is your feature request related to a problem? Please describe

Limiting the landing mode to 10s may lead to enter RESCUE_ABORT mode, disarming prematurely .
[image]

### Describe the solution you'd like

Extend the time limit for landing mode, or let the user set it.

### Describe alternatives you've considered

Extend the time limit for landing mode, or let the user set it.

robertrypula, 2023-01-19:
I think I had this "falling landing" issue:

https://youtu.be/hrfZR-N52z8?t=13 (at 0:13)

Unfortunately I don't remember exact GPS settings when it didn't land but it wasn't defaults. I reduced altitude to ~10 meters and all speeds to ~10 km/h because it was my first test of GPS rescue. I didn't want my quad to fly very fast and very high.

When I raised the speeds a bit it landed correctly. I think I still have last settings (when it was still slow but landed) in my quad - I can post that later.

andre-hilgersom, 2023-02-05:
I had 3 premature disarms yesterday. Stable altitude readings with more then enough sats.
But altitude is like 6 meters below sea level, not unusual in the Netherlands ... ;-)
Is that the cause, or what can be. Anyway; what parameter can I change to fix this ?

closed 2023-02-15 by SadalmelikJ