betaflight/betaflight #12211: HGLRCF722 mini - baro (SPI) not working with BF 4.4.0 RC5

pacze, 2023-01-17:
### Describe the bug

Barometer not working/not visible in BF Configurator (10.9 RC4 and RC5).
Confirmed with BF 4.4.0 RC4 and RC5

Same setup, baro works fine with BF 4.3.2 + config from issue: #11674
```
set baro_bustype = SPI
set baro_spi_device = 1
set baro_i2c_device = 0
```
But this config doesn't help with BF 4.4.0

### To Reproduce

Flash HGLRCF72 mini with BF 4.4.0 (RC4 or RC5).

### Expected behavior

Working barometer.

### Support ID

```txt
d8ad16d7-366a-4984-b9a0-fa68b8339662
```

### Diff configuration

```txt
# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 17 2023 / 18:16:40 (a7cf8f0c6) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name HGLRCF722
manufacturer_id HGLR

# name: Zenobiusz

# feature
feature GPS
feature LED_STRIP
feature OSD

# serial
serial 1 2 115200 57600 0 115200
serial 3 131073 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# led
led 0 4,5::CTOW:14
led 1 5,5::CTOW:14
led 2 6,5::CTOW:14
led 3 7,5::CTOW:14

# color
color 13 119,0,255
color 14 0,0,255
color 15 54,131,255

# aux
aux 0 0 0 900 1450 0 0
aux 1 1 0 1250 1350 0 0
aux 2 46 0 900 1050 0 0
aux 3 13 0 1600 1700 0 0
aux 4 15 3 1900 2100 0 0
aux 5 26 1 1900 2100 0 0
aux 6 31 2 1900 2100 0 0
aux 7 35 0 1075 1200 0 0

# rxfail
rxfail 3 h
rxfail 11 s 1000

# master
set gyro_lpf1_static_hz = 150
set gyro_lpf2_static_hz = 300
set dyn_notch_count = 2
set gyro_lpf1_dyn_min_hz = 150
set gyro_lpf1_dyn_max_hz = 300
set acc_lpf_hz = 10
set acc_calibration = -4,49,-47,1
set baro_bustype = SPI
set baro_spi_device = 1
set baro_i2c_device = 0
set min_check = 1010
set max_check = 1990
set rssi_channel = 12
set blackbox_sample_rate = 1/1
set dshot_bidir = ON
set failsafe_delay = 5
set failsafe_procedure = GPS-RESCUE
set vbat_min_cell_voltage = 300
set vbat_warning_cell_voltage = 300
set vbat_scale = 111
set ibata_scale = 400
set ibatv_scale = 27
set yaw_motors_reversed = ON
set small_angle = 180
set gps_sbas_mode = AUTO
set gps_set_home_point_once = ON
set gps_rescue_return_alt = 50
set gps_rescue_ground_speed = 1388
set gps_rescue_descent_dist = 150
set gps_rescue_descend_rate = 250
set gps_rescue_throttle_max = 1800
set gps_rescue_min_sats = 6
set yaw_deadband = 5
set pid_process_denom = 2
set simplified_gyro_filter_multiplier = 60
set osd_vbat_pos = 257
set osd_rssi_pos = 3129
set osd_link_quality_pos = 1092
set osd_rssi_dbm_pos = 161
set osd_tim_1_pos = 353
set osd_tim_2_pos = 321
set osd_flymode_pos = 2716
set osd_throttle_pos = 2677
set osd_vtx_channel_pos = 193
set osd_crosshairs_pos = 348
set osd_ah_sbar_pos = 2397
set osd_ah_pos = 2269
set osd_current_pos = 3681
set osd_mah_drawn_pos = 3715
set osd_motor_diag_pos = 2709
set osd_craft_name_pos = 2048
set osd_gps_speed_pos = 2608
set osd_gps_lon_pos = 3114
set osd_gps_lat_pos = 3082
set osd_gps_sats_pos = 3096
set osd_home_dir_pos = 2651
set osd_home_dist_pos = 2655
set osd_flight_dist_pos = 184
set osd_compass_bar_pos = 2617
set osd_altitude_pos = 18927
set osd_pid_roll_pos = 64
set osd_pid_pitch_pos = 96
set osd_pid_yaw_pos = 128
set osd_power_pos = 371
set osd_warnings_pos = 2232
set osd_avg_cell_voltage_pos = 3722
set osd_battery_usage_pos = 36516
set osd_disarmed_pos = 2490
set osd_esc_tmp_pos = 408
set osd_rtc_date_time_pos = 2068
set osd_flip_arrow_pos = 65
set osd_core_temp_pos = 248
set osd_log_status_pos = 507
set osd_stat_bitmask = 2111276
set osd_displayport_device = MSP
set osd_canvas_width = 60
set osd_canvas_height = 22
set debug_mode = GYRO_SCALED
set vcd_video_system = HD
set gyro_1_sensor_align = DEFAULT
set rpm_filter_harmonics = 2
set craft_name = Zenobiusz

profile 1

# profile 1
set profile_name = UAV TECH
set dterm_lpf1_dyn_min_hz = 90
set dterm_lpf1_dyn_max_hz = 180
set dterm_lpf1_static_hz = 90
set dterm_lpf2_static_hz = 180
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set p_pitch = 59
set i_pitch = 69
set d_pitch = 72
set f_pitch = 215
set p_roll = 57
set i_roll = 66
set d_roll = 58
set f_roll = 206
set p_yaw = 57
set i_yaw = 66
set f_yaw = 206
set d_min_roll = 58
set d_min_pitch = 72
set dyn_idle_min_rpm = 40
set simplified_master_multiplier = 150
set simplified_i_gain = 65
set simplified_d_gain = 130
set simplified_pi_gain = 85
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 115
set simplified_pitch_d_gain = 110
set simplified_dterm_filter_multiplier = 120

rateprofile 2

# rateprofile 2
set rateprofile_name = CINE + T
set thr_mid = 33
set thr_expo = 50
set roll_rc_rate = 2
set pitch_rc_rate = 2
set yaw_rc_rate = 2
set roll_expo = 60
set pitch_expo = 60
set yaw_expo = 60
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 63

# end the command batch
batch end

#
```

### Resources configuration

```txt
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: PREINIT 2
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED_STRIP
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: LED_STRIP
DMA2 Stream 4: ADC
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

HGLRC Zeus F730 STACK FPV Racing Drone 3-6S F722 Flight Controller

### Add any other context about the problem that you think might be relevant here

Working correctly BF 4.3.2 says:
```
# resource show all
...
A04: BARO_CS
...
B12: PREINIT 4
```
For BF 4.4.0 I've tried:
```
resource BARO_CS A04
save
```
commands succeed but it didn't help.

haslinghuis, 2023-01-17:
Try flashing `HGLRC722E` target.

closed 2023-01-17 by haslinghuis

pacze, 2023-01-17:
Thanks @haslinghuis!
With `HGLRC722E` works out of the box :-)