betaflight/betaflight #12214: Deadlock with SITL target

ykeuter, 2023-01-18:
### Describe the bug

When trying to use the SITL target with Gazebo (https://github.com/betaflight/betaflight/tree/master/src/main/target/SITL), there seems to be a deadlock between the two: both Gazebo ArduCopter plugin and BF SITL target are waiting for message to arrive.

### To Reproduce

Follow the instructions on https://github.com/betaflight/betaflight/tree/master/src/main/target/SITL

### Expected behavior

Messaging over ports 9002 and 9003 between BF SITL and Gazebo's ArduCopter plugin.

### Support ID

```txt
f1186672-167e-4a05-8685-fa22fc56b451
```

### Diff configuration

```txt
# version
# Betaflight / SITL (SITL) 4.4.0 Jan 18 2023 / 11:31:55 (d96586d50) MSP API: 1.45

# start the command batch
batch start

# feature
feature -GPS
feature -TELEMETRY

# aux
aux 0 0 0 1300 1700 0 0

# master
set acc_hardware = NONE
set motor_pwm_protocol = PWM
set pid_process_denom = 16

profile 0

rateprofile 0

# end the command batch
batch end

#
```

### Resources configuration

```txt
N/A
```

### Flight controller

SITL

### Other components

Gazebo 9

### How are the different components wired up

N/A

ykeuter, 2023-01-18:
I finally got it working with this fork/branch: https://github.com/nikhil-sethi/betaflight/tree/airsim-sitl Just merging this branch does not do the trick though, so I suspect something broke it down the road. Having a quick look at the code, my guess would be `#undef USE_PWM_OUTPUT` in the header file of the SITL target is the culprit. It is blocking motor updates in `motor.c`.

haslinghuis, 2023-01-18:
@ykeuter please test :point_up:

ykeuter, 2023-01-19:
> @ykeuter please test point_up

I will have a look now. Thanks!

ykeuter, 2023-01-19:
#12217 is not sufficient, as we also need to define `USE_PWM_OUTPUT`. I opened a new PR :point_up:

PR #12217 (2023-01-19): Fix SITL
Fixes: #12214
PR #12220 (2023-01-23): fix SITL
Fixes #12214

Further improvements are in #11303 , but this PR makes SITL at least run for now.
closed 2023-01-23 by blckmn