betaflight/betaflight #12216: Matek405STD flight controller BF4.4.0-RC5 - flash wipes the accelerometer

billjow, 2023-01-18:
### Describe the bug

Trying to upgrade my Mermaid X220 to BF4.4.0. The flight controller appears to be a Matek405STD. When I flash the RC5 firmware, it removes the accelerometer in the Setup screen and shows no gyro in the Configuration tab.  I attach the diff all from 4.3.0 and the diff all from after the flash and applying the custom defaults.
[BTFL_cli_backup_Mermaid220_20230118_173506 BTF 4.3.0.txt](https://github.com/betaflight/betaflight/files/10449227/BTFL_cli_backup_Mermaid220_20230118_173506.BTF.4.3.0.txt)
[BTFL_cli_20230118_174048 X220 BTF4.4.0 backup basic defaults.txt](https://github.com/betaflight/betaflight/files/10449235/BTFL_cli_20230118_174048.X220.BTF4.4.0.backup.basic.defaults.txt)

### To Reproduce

Have repeated the attempt a number of times - happens every time.

### Expected behavior

A straightforward flash finding the expected hardware.

### Support ID

```txt
15edeae0-0906-478e-aa99-db6abbbda851
```

### Diff configuration

```txt
# diff

# version
# Betaflight / STM32F405 (S405) 4.3.0 Jun 11 2022 / 08:23:42 (229ac6675) MSP API: 1.44

# config: YES

# start the command batch
batch start

board_name MATEKF405STD
manufacturer_id MTKS

# name: Mermaid220

# feature
feature -RX_PARALLEL_PWM

# serial
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 3 1 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1875 2100 0 0
aux 1 1 1 900 1175 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1875 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY    0    0    0    0    0    0    0    0
vtxtable band 2 BOSCAM_B B FACTORY    0    0    0    0    0    0    0    0
vtxtable band 3 BOSCAM_E E FACTORY    0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F FACTORY    0 5760    0 5800    0    0    0    0
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I FACTORY    0    0    0    0    0    0    0    0
vtxtable powerlevels 3
vtxtable powervalues 14 23 0
vtxtable powerlabels 25 200 0

# master
set acc_calibration = -52,-32,47,1
set max_check = 2000
set serialrx_provider = CRSF
set crsf_use_rx_snr = ON
set motor_pwm_protocol = DSHOT600
set ibata_scale = 1875
set yaw_motors_reversed = ON
set small_angle = 180
set osd_warn_rc_smoothing = OFF
set osd_warn_link_quality = ON
set osd_warn_rssi_dbm = ON
set osd_rssi_dbm_alarm = -98
set osd_vbat_pos = 2534
set osd_rssi_pos = 87
set osd_link_quality_pos = 2072
set osd_rssi_dbm_pos = 2104
set osd_tim_1_pos = 467
set osd_tim_2_pos = 2520
set osd_flymode_pos = 2496
set osd_g_force_pos = 2541
set osd_throttle_pos = 2528
set osd_crosshairs_pos = 2350
set osd_current_pos = 2547
set osd_mah_drawn_pos = 2501
set osd_craft_name_pos = 2049
set osd_gps_speed_pos = 242
set osd_gps_lon_pos = 170
set osd_gps_lat_pos = 137
set osd_altitude_pos = 18898
set osd_warnings_pos = 14348
set osd_avg_cell_voltage_pos = 2507
set osd_disarmed_pos = 2379
set osd_rtc_date_time_pos = 480
set osd_core_temp_pos = 162
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_min_link_quality = ON
set osd_displayport_device = MSP
set vtx_band = 5
set vtx_channel = 1
set vtx_freq = 5658
set displayport_msp_serial = 3
set gyro_1_sensor_align = CW270
set gyro_1_align_yaw = 2700
set name = Mermaid220

profile 0

rateprofile 0

# end the command batch
batch end
```

### Resources configuration

```txt
# version
# Betaflight / STM32F405 (S405) 4.3.0 Jun 11 2022 / 08:23:42 (229ac6675) MSP API: 1.44

# config: YES

# start the command batch
batch start

board_name MATEKF405STD
manufacturer_id MTKS

# name: Mermaid220

# feature
feature -RX_PARALLEL_PWM

# serial
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 3 1 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1875 2100 0 0
aux 1 1 1 900 1175 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1875 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY    0    0    0    0    0    0    0    0
vtxtable band 2 BOSCAM_B B FACTORY    0    0    0    0    0    0    0    0
vtxtable band 3 BOSCAM_E E FACTORY    0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F FACTORY    0 5760    0 5800    0    0    0    0
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I FACTORY    0    0    0    0    0    0    0    0
vtxtable powerlevels 3
vtxtable powervalues 14 23 0
vtxtable powerlabels 25 200 0

# master
set acc_calibration = -52,-32,47,1
set max_check = 2000
set serialrx_provider = CRSF
set crsf_use_rx_snr = ON
set motor_pwm_protocol = DSHOT600
set ibata_scale = 1875
set yaw_motors_reversed = ON
set small_angle = 180
set osd_warn_rc_smoothing = OFF
set osd_warn_link_quality = ON
set osd_warn_rssi_dbm = ON
set osd_rssi_dbm_alarm = -98
set osd_vbat_pos = 2534
set osd_rssi_pos = 87
set osd_link_quality_pos = 2072
set osd_rssi_dbm_pos = 2104
set osd_tim_1_pos = 467
set osd_tim_2_pos = 2520
set osd_flymode_pos = 2496
set osd_g_force_pos = 2541
set osd_throttle_pos = 2528
set osd_crosshairs_pos = 2350
set osd_current_pos = 2547
set osd_mah_drawn_pos = 2501
set osd_craft_name_pos = 2049
set osd_gps_speed_pos = 242
set osd_gps_lon_pos = 170
set osd_gps_lat_pos = 137
set osd_altitude_pos = 18898
set osd_warnings_pos = 14348
set osd_avg_cell_voltage_pos = 2507
set osd_disarmed_pos = 2379
set osd_rtc_date_time_pos = 480
set osd_core_temp_pos = 162
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_min_link_quality = ON
set osd_displayport_device = MSP
set vtx_band = 5
set vtx_channel = 1
set vtx_freq = 5658
set displayport_msp_serial = 3
set gyro_1_sensor_align = CW270
set gyro_1_align_yaw = 2700
set name = Mermaid220

profile 0

rateprofile 0

# end the command batch
batch end
```

### Flight controller

Matek405STD

### Other components

Happymodel EP1 ELRS Rx
HDZero RaceV2 VTx

### How are the different components wired up

As normal - works and flys fine in 4.3.0

haslinghuis, 2023-01-18:
Please try flashing `MATEK405STD-CLONE` board configuration instead.

billjow, 2023-01-18:
Sadly, it did not work – still no accelerometer. Diff attached.

 

Kind Regards,

 

Bill

 

Bill Jowitt

Tel: 07790 491385

 

# Betaflight / STM32F405 (S405) 4.4.0 Jan 18 2023 / 19:04:15 (a7cf8f0c6) MSP API: 1.45

board_name MATEKF405STD_CLONE
manufacturer_id CLNE

# master
set osd_canvas_width = 30
set osd_canvas_height = 13
set gyro_1_align_yaw = 1800

haslinghuis, 2023-01-18:
Please flash 4.3.2 and report the output of the status command in CLI tab.

billjow, 2023-01-18:
Successfully flashed, and accelerometer stayed. Diff attached.

 

Kind Regards,

 

Bill

 

# Betaflight / STM32F405 (S405) 4.3.2 Nov 28 2022 / 07:26:30 (60c9521) MSP API: 1.44

# master
set acc_calibration = 21,18,50,1

haslinghuis, 2023-01-18:
Please go to the cli tab. Type `status` and paste the output here.

billjow, 2023-01-18:
From 4.3.2:

 

status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.22V, Core temp=27degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 4100, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked
GYRO=BMI270, ACC=BMI270
OSD: MSP
System Uptime: 27 seconds, Current Time: 2023-01-18T23:08:59.394+00:00
CPU:33%, cycle time: 308, GYRO rate: 3246, RX rate: 15, System rate: 9
Voltage: 158 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Startup failed
Arming disable flags: RXLOSS CLI MSP RPMFILTER

 

From 4.4.0_RC5:

 

STATUS
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.22V, Core temp=30degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CUSTOM DEFAULTS, size: 3386, max available: 16384
Devices detected: SPI:0, I2C:1
Gyros detected: locked
, BARO=BMP280
OSD: MSP (30 x 13)
BUILD KEY: ac8c96ceb8950749ebd1d68e5fd19f4c (4.4.0-RC5)
System Uptime: 16 seconds, Current Time: 2023-01-18T23:12:12.129+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 0, System rate: 9
Voltage: 167 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Startup failed
Arming disable flags: NOGYRO BOOTGRACE CLI MSP

 

Kind Regards,

 

Bill

haslinghuis, 2023-01-18:
Thanks. If above PR is merged GYRO and ACC should work.

billjow, 2023-01-19:
Many thanks - I look forward to trying it out. Appreciate all your efforts.

closed 2023-01-19 by blckmn

vaska1992, 2023-01-20:
I have same issue with my flight controller: Speedybee F7 mini. 
Flashing 4.4 RC4/5 leaves me without acc/gyro.

Flashing back the 4.3.2 - everything is there.

haslinghuis, 2023-01-20:
@vaska1992 Please provide required information:

- [ ] Flash 4.3.2
- [ ] Click the CLI tab.
- [ ] Type version and paste the output here.
- [ ] Type status and paste the output here.

SgtStank, 2023-01-27:
Same issue with AxisFlying F411 using MatekF411 target. Although, it seems to work fine when selecting "Core Only" for 4.4 RC5.

haslinghuis, 2023-01-27:
Please in Core Only mode go to CLI and press the Submit Support Data button and provide the ID here.