betaflight/betaflight #12223: SPI ELRS not working on G4

David-OConnor, 2023-01-21:
### Describe the bug

G4 SPI ELRS not working. STM32G473 MCU. No sign of life on the LED pin. Have tried with Reset pin wired, and not
I can confirm the SPI connection to the SX1280 works, due to successful test register reads on custom firmware. I can confirm the Busy and DIO pins are wired correctly from a high/low toggle program. (Also have applied this to SCK, MOSI, MISO, and CS)

Relevant BF config:
```
resource RX_SPI_EXTI 1 C13  # ie DIO1 on the radio.
resource RX_SPI_EXPRESSLRS_BUSY 1 C14
resource RX_SPI_CS 1 C15
resource RX_SPI_LED 1 C10 # todo temp troubleshooting
# resource RX_SPI_EXPRESSLRS_RESET 1 B06 # todo: Temp TS

set rx_spi_protocol = EXPRESSLRS
set rx_spi_bus = 3
```

Of note, the LED pin showing no signs of life is interesting to me. For example, if BF couldn't find the radio, you'd expect it to beep in a certain way, like the official ELRS does in that case.

From the clock (SCK) line. On power-up, then stay slow:
[image]

### To Reproduce

Connect a G4 FC with a SX1280 radio. Compile with these flags:
```
make clean ; make TARGET=STM32G47X EXTRA_FLAGS="-DCLOUD_BUILD -DUSE_DSHOT -DUSE_RX_EXPRESSLRS -DUSE_FLASH -DUSE_GPS -DUSE_OSD -DUSE_VTX -DUSE_RX_SX1280 -DUSE_RX_SPI -DUSE_GYRO_SPI_ICM42688P -DUSE_ACC_SPI_ICM42688P -DUSE_BARO -DUSE_BARO_DPS310 -DUSE_RX_EXPRESSLRS_TELEMETRY -DRX_CHANNELS_AETR -DUSE_GYRO -DUSE_ACC -DUSE_SERIALRX_CRSF -DUSE_SERIALRX -DUSE_TELEMETRY_CRSF -DUSE_TELEMETRY -DUSE_FLASH_M25P16"
```

Attempt to bind.

### Expected behavior

Rx binds. Good value on clock (and other lines) on LA.

### Support ID

```txt
I do not see this button.
```

### Diff configuration

```txt
# diff

# version
# Betaflight / STM32G47X (SG47) 4.4.0 Jan 17 2023 / 21:57:34 (norevision) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name MERCURYG4
manufacturer_id ANYL

# master
set osd_canvas_width = 30
set osd_canvas_height = 13
set gyro_1_align_yaw = 900

profile 0

rateprofile 0

# end the command batch
batch end
```

### Resources configuration

```txt
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FLASH_CS
A01: ADC_BATT
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: MOTOR 2
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: BEEPER
A09: I2C_SCL 2
A10: I2C_SDA 2
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: I2C_SCL 1
B00: MOTOR 3
B01: MOTOR 4
B02: ADC_CURR
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: FREE
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: GYRO_CS 1
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 1
C07: FREE
C08: FREE
C09: FREE
C10: LED
C11: FREE
C12: FREE
C13: RX_SPI_EXTI
C14: RX_SPI_EXPRESSLRS_BUSY
C15: RX_SPI_CS
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1 : MOTOR 1
    CH2 : MOTOR 2
    CH3 : MOTOR 3
    CH4 : MOTOR 4
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM15: FREE
TIM16: FREE
TIM17: FREE
TIM20: FREE

Currently active DMA:
--------------------
DMA1 Channel 1: TIMUP 3
DMA1 Channel 2: FREE
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: ADC 1
DMA1 Channel 8: ADC 2
DMA2 Channel 1: SPI_MOSI 1
DMA2 Channel 2: SPI_MISO 1
DMA2 Channel 3: SPI_MOSI 2
DMA2 Channel 4: SPI_MISO 2
DMA2 Channel 5: SPI_MOSI 3
DMA2 Channel 6: SPI_MISO 3
DMA2 Channel 7: FREE
DMA2 Channel 8: FREE
```

### Flight controller

AnyLeaf G4

### Other components

Sx1281 radio.

### How are the different components wired up

[image]

[image]

haslinghuis, 2023-02-21:
You need a lot more defines. But development on SPI based receivers is not recommended as the SPI part of the firmware is not official supported.
```
#define RX_SPI_DEFAULT_PROTOCOL         RX_SPI_EXPRESSLRS
#define RX_EXPRESSLRS_TIMER_INSTANCE    TIM5
#define RX_EXPRESSLRS_SPI_RESET_PIN     PA8
#define RX_EXPRESSLRS_SPI_BUSY_PIN      PA13
#define RX_SPI_CS                       PA15
#define RX_SPI_EXTI                     PC14
#define RX_SPI_BIND                     PB10
#define RX_SPI_LED                      PA14
```

David-OConnor, 2023-02-21:
Thanks. We can close this if you want as I've given up and put an ELRS MCU on the board.

haslinghuis, 2023-02-21:
Agree, good decision.

closed 2023-02-21 by haslinghuis