betaflight/betaflight #12225: Launch Control missing from 4.4RC5

noahliketheark, 2023-01-22:
### Describe the bug

After flashing 4.4RC5, launch control is missing from the CLI.

# get launch
###ERROR IN get: INVALID NAME: launch###

### To Reproduce

Flash 4.4RC5, go to CLI, type "get launch"

### Expected behavior

4.4RC4:

# get launch
launch_control_mode = NORMAL
profile 0
Allowed values: NORMAL, PITCHONLY, FULL

launch_trigger_allow_reset = ON
profile 0
Allowed values: OFF, ON

launch_trigger_throttle_percent = 20
profile 0
Allowed range: 0 - 90

launch_angle_limit = 0
profile 0
Allowed range: 0 - 80

launch_control_gain = 40
profile 0
Allowed range: 0 - 200

### Support ID

```txt
Id: "34cff405-27db-4543-8b6e-39b39c8a3529"
```

### Diff configuration

```txt
diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 21 2023 / 06:26:30 (a7cf8f0c6) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name PYRODRONEF7
manufacturer_id PYDR

# name: Switchback HDZ

# feature
feature -TELEMETRY

# serial
serial 0 64 115200 57600 0 115200
serial 3 131073 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# beeper
beeper -DISARM_REPEAT
beeper -ON_USB

# beacon
beacon RX_SET

# aux
aux 0 0 0 1975 2025 0 0
aux 1 1 1 1975 2025 0 0
aux 2 13 1 1400 1450 0 0
aux 3 28 1 1700 2025 0 0
aux 4 35 1 1250 1300 0 0
aux 5 40 0 900 2100 0 0
aux 6 49 1 975 1025 0 0

# rxrange
rxrange 0 989 2012
rxrange 1 989 2012
rxrange 2 989 2012
rxrange 3 989 2012

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY    0    0    0    0    0    0    0    0
vtxtable band 2 BOSCAM_B B FACTORY    0    0    0    0    0    0    0    0
vtxtable band 3 BOSCAM_E E FACTORY    0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F FACTORY    0 5760    0 5800    0    0    0    0
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I FACTORY    0    0    0    0    0    0    0    0
vtxtable powerlevels 3
vtxtable powervalues 14 23 0
vtxtable powerlabels 25 200 0

# master
set gyro_lpf1_static_hz = 0
set dyn_notch_count = 2
set dyn_notch_q = 500
set dyn_notch_min_hz = 200
set gyro_lpf1_dyn_min_hz = 0
set acc_lpf_hz = 10
set acc_calibration = 10,25,-17,1
set rc_smoothing_auto_factor = 25
set rc_smoothing_auto_factor_throttle = 25
set blackbox_device = NONE
set dshot_bidir = ON
set bat_capacity = 1200
set vbat_scale = 111
set ibata_scale = 180
set beeper_dshot_beacon_tone = 3
set small_angle = 180
set simplified_gyro_filter = OFF
set report_cell_voltage = ON
set osd_link_quality_pos = 13323
set osd_rssi_dbm_pos = 13355
set osd_tim_2_pos = 14880
set osd_throttle_pos = 14336
set osd_vtx_channel_pos = 9769
set osd_current_pos = 14368
set osd_mah_drawn_pos = 12768
set osd_pilot_name_pos = 8756
set osd_warnings_pos = 14837
set osd_avg_cell_voltage_pos = 14400
set osd_battery_usage_pos = 47616
set osd_disarmed_pos = 14806
set osd_ready_mode_pos = 472
set osd_flip_arrow_pos = 14650
set osd_profile = 3
set osd_displayport_device = MSP
set osd_canvas_width = 50
set osd_canvas_height = 18
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 1
set vtx_freq = 5658
set vcd_video_system = HD
set rpm_filter_harmonics = 2
set rpm_filter_min_hz = 150
set rpm_filter_fade_range_hz = 100
set craft_name = Switchback HDZ
set pilot_name = NoahLikeTheArk

profile 1

# profile 1
set profile_name = old tune
set dterm_lpf1_dyn_expo = 7
set vbat_sag_compensation = 100
set anti_gravity_gain = 50
set p_pitch = 51
set i_pitch = 96
set d_pitch = 36
set f_pitch = 130
set p_roll = 51
set i_roll = 96
set d_roll = 35
set f_roll = 132
set p_yaw = 75
set i_yaw = 130
set f_yaw = 180
set d_min_roll = 35
set d_min_pitch = 36
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 65
set feedforward_jitter_factor = 3
set feedforward_boost = 18
set feedforward_max_rate_limit = 100
set dyn_idle_min_rpm = 30
set simplified_pids_mode = RP
set simplified_i_gain = 105
set simplified_d_gain = 120
set simplified_pi_gain = 115
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 110
set simplified_pitch_d_gain = 90
set simplified_pitch_pi_gain = 95
set tpa_rate = 60

rateprofile 0

# rateprofile 0
set roll_rc_rate = 18
set pitch_rc_rate = 18
set yaw_rc_rate = 18
set roll_expo = 54
set pitch_expo = 54
set yaw_expo = 54

# end the command batch
batch end

#
```

### Resources configuration

```txt
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: LED 1
B01: FREE
B02: FREE
B03: FREE
B04: BEEPER
B05: SPI_MOSI 3
B06: MOTOR 1
B07: MOTOR 2
B08: MOTOR 3
B09: FREE
B10: FREE
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 4
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: PINIO 1
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
    CH4 : DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 3

#
```

### Flight controller

Pyrodrone F7 20x20

McGiverGim, 2023-01-22:
You must add LAUNCH_CONTROL to the custom defines in the cloud build. More info here https://github.com/betaflight/betaflight/wiki/Flashing-firmware

closed 2023-01-22 by KarateBrot

noahliketheark, 2023-01-23:
Whenever I try to add LAUNCH_CONTROL to the "Custom Defines" box and load firmware online, the log says nothing about Launch Control:

git fetch origin 4.4.0-RC5 && git checkout -f FETCH_HEAD && make TARGET=STM32F7X2 EXTRA_FLAGS="-D'BUILD_KEY=f2c76c72f9c72a53d1ba6570ac21aab7' -D'RELEASE_NAME=4.4.0-RC5' -D'BOARD_NAME=PYRODRONEF7' -D'MANUFACTURER_ID=PYDR' -DCLOUD_BUILD -DUSE_ACC -DUSE_ACC_SPI_MPU6000 -DUSE_DSHOT -DUSE_FLASH -DUSE_GPS -DUSE_GYRO -DUSE_GYRO_SPI_MPU6000 -DUSE_LED_STRIP -DUSE_MAX7456 -DUSE_OSD -DUSE_OSD_HD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX"

noahliketheark, 2023-01-23:
I figured it out. One must have Expert Mode enabled for the Custom Definitions to be honored. The field should be hidden if Expert Mode isn't enabled IMO.