betaflight/betaflight #12226: walksnail avatar OSD

wizzzz0, 2023-01-22:
### Describe the bug

walksnail avatar elements do not line up with the osd image on betaflight they are down and to the right , also the clicking the hd doses not make a diffrence  

### To Reproduce

[image]
[image]

### Expected behavior

Should work like the dji msp osd 

### Support ID

```txt
Id: "aceafddc-2265-499c-a625-803b5def4a8a"
```

### Diff configuration

```txt
# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 17 2023 / 04:30:33 (a7cf8f0c6) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name SPEEDYBEEF7V3
manufacturer_id SPBE

# feature
feature -TELEMETRY
feature ESC_SENSOR

# serial
serial 0 131073 115200 57600 0 115200

# aux
aux 0 0 0 1900 2100 0 0
aux 1 1 1 1900 2100 0 0
aux 2 13 2 1700 2100 0 0

# master
set gyro_lpf2_static_hz = 1000
set dyn_notch_count = 2
set dyn_notch_q = 400
set dyn_notch_max_hz = 800
set gyro_lpf1_dyn_min_hz = 300
set gyro_lpf1_dyn_max_hz = 600
set gyro_lpf1_dyn_expo = 8
set acc_lpf_hz = 10
set acc_calibration = -32,-58,0,1
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set deadband = 2
set yaw_deadband = 2
set simplified_gyro_filter = OFF
set osd_vbat_pos = 14818
set osd_flymode_pos = 6151
set osd_ah_pos = 2159
set osd_warnings_pos = 14699
set osd_avg_cell_voltage_pos = 6632
set osd_battery_usage_pos = 14347
set osd_disarmed_pos = 14833
set osd_efficiency_pos = 14844
set osd_displayport_device = MSP
set vcd_video_system = HD
set rpm_filter_harmonics = 1

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 0
set dterm_lpf1_dyn_max_hz = 0
set dterm_lpf1_dyn_expo = 0
set dterm_lpf1_static_hz = 80
set dterm_lpf2_static_hz = 140
set p_pitch = 54
set i_pitch = 72
set d_pitch = 35
set f_pitch = 112
set p_roll = 51
set i_roll = 69
set d_roll = 33
set f_roll = 108
set p_yaw = 51
set i_yaw = 69
set f_yaw = 108
set d_min_roll = 33
set d_min_pitch = 35
set feedforward_smooth_factor = 30
set feedforward_jitter_factor = 9
set simplified_i_gain = 75
set simplified_d_gain = 110
set simplified_pi_gain = 115
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 90
set simplified_pitch_d_gain = 95
set simplified_dterm_filter = OFF

rateprofile 0

# rateprofile 0
set throttle_limit_percent = 95

# end the command batch
batch end
```

### Resources configuration

```txt
A00: FREE
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: MOTOR 1
B00: FREE
B01: FREE
B02: PREINIT 1
B03: MOTOR 2
B04: MOTOR 3
B05: FREE
B06: MOTOR 4
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: PREINIT 4
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: GYRO_EXTI
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: PINIO 1
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: SDCARD_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 1
    CH4 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 2
```

### Flight controller

speedy bee v3

SteveCEvans, 2023-01-22:
Correct. I’m working with Walksnail, but please don’t expect much due to the New Year. Currently WTFOS and HDZero support the declaration of canvas size, with position-able system OSD elements being worked on I believe. Colour text for warnings etc. is still in the pipeline.

wizzzz0, 2023-01-22:
Perfect thanks for the update

howels, 2023-01-23:
@wizzzz0 you may wish to try out a more recent firmware. HD OSD is available on 29.33.16 I believe which is now the Caddx page:  https://caddxfpv.com/pages/download
You may need to align the OSD area with the goggles screen using the goggles settings as the HD OSD is initially offset too far right & down. There's a OSD position setting in the goggles to adjust.

wizzzz0, 2023-01-28:
@howels  thanks for the info all sorted now cheers

howels, 2023-02-08:
Please close if this is sorted out, thanks

closed 2023-03-18 by github-actions[bot]