betaflight/betaflight #12228: Betaflight 4.4 RC5 cannot arm due to CALIB error

mcaraballo, 2023-01-23:
### Describe the bug

In Betaflight 4.4 RC5 you cannot arm the quad which results in a “CALIB” error using an iFlight SucceX-D F7 TwinG F722 Flight Controller - target firmware "EXUA/EXF722DUAL(STM32F7X2)".  This occurs in a fresh install of the firmware only enabling an arm switch, even with a still quad with a calibrated gyro and accelerometer.  Flashing back to release 4.3.2 with the same basic configuration removes the issue.  It only occurs on 4.4.

### To Reproduce

Install 4.4 RC5 to a iFlight SucceX-D F7 TwinG F722 Flight Controller and apply the default parameters. Configure a switch to arm the quad and try arming.  The OSD will display “CALIB”.

### Expected behavior

Quad should arm.

### Support ID

```txt
Typing Support ID in the CLI in 4.4 per instructions yields:

# Support ID
###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###
```

### Diff configuration

```txt
# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 23 2023 / 04:51:29 (a7cf8f0c6) MSP API: 1.45

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name EXF722DUAL
manufacturer_id EXUA
mcu_id 004600353238511236353235
signature 

# feature
feature GPS
feature ESC_SENSOR

# serial
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 4 1024 115200 57600 0 115200

# map
map TAER1234

# aux
aux 0 0 0 1300 1700 0 0

# master
set acc_calibration = 68,43,-5,1
set rc_smoothing_auto_factor = 60
set serialrx_provider = CRSF
set vcd_video_system = HD
set gyro_2_align_yaw = 900

profile 0

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

### Resources configuration

```txt
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: GYRO_EXTI
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: GYRO_CS 1
B00: FREE
B01: FREE
B02: GYRO_EXTI
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: PINIO 3
B09: FLASH_CS
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: GYRO_CS 2
C04: LED 1
C05: FREE
C06: MOTOR 2
C07: MOTOR 4
C08: MOTOR 1
C09: MOTOR 3
C10: SERIAL_TX 4
C11: SERIAL_RX 4
C12: FREE
C13: PINIO 1
C14: PINIO 2
C15: BEEPER
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

iFlight SucceX-D F7 TwinG V2.2 F722 Flight Controller 

### Other components

Catalyst Machineworks Banggod Frame
DJI O3 Air Unit
Matek SAM-M8Q GPS
TBS Nano Diversity Receiver
SucceX 50A 2-6S BLHeli_32 4 in 1 ESC
iFlight Xing 2208 1300KV Motors

### How are the different components wired up

UART2 - DJI O3 Air Unit
UART4 - Matek SAM-M8Q GPS
UART3 - TBS Nano Diversity Receiver

### Add any other context about the problem that you think might be relevant here

I tried setting the moron_threshold now known as gyro_calib_noise_limit = 90 and it didn’t make a difference. Gyro does not seem defective.  This assumption is based on the fact that reverting to 4.3.2 works with no CALIB error

4stah, 2023-01-29:
Maybe I'm wrong, but with I have probably the same board. I'm using IFLIGHT_F722_TWING target and it works.

FreeBirdBrain, 2023-02-07:
My Darwin 129 also produces the gyro "Calib" on OSD and will not arm. Was running 4.1.1 previously perfectly well. Thanks if you can fix this!

DroneRob, 2023-02-07:
> My Darwin 129 also produces the gyro "Calib" on OSD and will not arm. Was running 4.1.1 previously perfectly well. Thanks if you can fix this!

Same for me Exept on my darwin129- MATEKF405STD board I had it working on 4.3 .

DroneRob, 2023-02-07:
I found that if you dont load ANY backup , there is no CALIB error . But Now Even after verifing Motor Directions and all other Prechecks, It will not lift off the Ground. Reverting back to 4.3 makes it fly Normal again

FreeBirdBrain, 2023-02-07:
Good to know I'm not the only one with the Darwin 129 issue on v4.4. Glad you found a fix, Rob. 
I thought my FC was done for, but I finally managed to fix it. What worked / long story short: Configurator 10.6 + 4.1.1 (my last version of Betaflight.)
The quad wouldn't arm with v4.4 with the gyro "Calib" error despite moving the "moron threshold" to 200. 
When I flashed 4.1.1 using 10.7.2, which I think was the last version of Configurator I was running, when I armed, quad would try to fly to the moon/freak out! (instead of giving the gyro calibration error.)
The quad seems to have more trouble being recognized by PC when I plug in USB, but if I re-plug USB about 5 times, it seems to work. Fingers crossed this is the worst of it. I'm definitely gonna be hesitant to flash newest firmware on this quad going forward. I wanted to try the new GPS Rescue, but it ain't worth it after going through this, although I think this sort of troubleshooting is part of the hobby and you have to expect it to happen and learn to be patient :-/

SolkGit, 2023-02-23:
Also have Darwin 129 and "Calib" flag with official 4.4 release. Solved it by choosing MATEKF405STD_CLONE as target. I have not applied my old backup, made a fresh setup step-by-step. Worked fine for me.

FreeBirdBrain, 2023-02-23:
That's interesting, SolkGit.  I tried the STD_CLONE and had the gyro Calib error, but I think I had loaded presets with help from the 4.3 migrator (https://bf43-safe-migration.robertodedomenico.it/)

I may have tried with stock settings and still had the Calib-error. I can't remember. 

So you didn't migrate your backup-settings? (ie: you didn't "load presets") I guess it all makes sense now. I wonder what's the root-cause...

SolkGit, 2023-02-23:
Correct, I haven't used any presets or backup migration. Just started from stock settings.

DroneRob, 2023-02-23:
I concur that  Adding any Preset or Back Up Brings the CALIB Error .

vzhabko, 2023-03-02:
Hi All, Here is some more information on this issue, which allow sort of workaround.

I have DarwinFPV Darwin129 and Darwin240, they have the same FC clone MATEKF405, target for 4.3 is either MATEKF405STR (suggested by DarwinFPV support) or MATEKF405STD(-CLONE) they work with no issues on this version firmware. Upgrading them to the 4.4 may require custom MPU/GYRO GYRO_SPI_MPU6500 ACC_SPI_MPU6500 options. 

Now about issue itself. It looks like it is a resource conflict between BARO and LED_STRIP. After bringing config from 4.3 just turn off either one or both (could be done through UI or CLI). CALIB issue will go away. At least in my case. However either BARO or LED_STRIP functionality will be lost, depends on what was turned off.

I did quick compare of resource mapping/timers between 4.3 and 4.4, at first glance they look the same.

mcaraballo, 2023-04-17:
> just turn off either one or both (could be done through UI or CLI). CALIB issue will go away. At least in my case. However either BARO or LED_STRIP functionality will be lost, depends on what was [...]

I tried turning off the BARO through the CLI and it fixes the issue but of course, as you said the BARO is now not available.  This bug still seems present and only resolved with workarounds like this.  May I ask how you "turned off" LED_STRIP?  If my Baro is left turned on in the UI with the LED_STRIP turned off in the UI, I still get the calib error.  I found that I must completely leave both off in the UI.

How are you turning the LED_STRIP off in the CLI?

This bug is still present it seems.

Thanks for any help from the community and/or Betaflight team.

vzhabko, 2023-04-18:
@mcaraballo Last time I was playing with changing both settings for BARO and LED_STRIP I have used MATEK405CTR (recommended by DarwinFPV support) firmware 4.4.0 with custom defaults: 

GYRO_SPI_MPU6500 ACC_SPI_MPU6500

I tried latest 4.4.1 only to check if bug fixed, which is not. Rolled back to 4.3.x.

To handle LED_STRIP with CLI use the following:

Turn **on**:

**feature LED_STRIP**
save

Turn **off**:

**feature -LED_STRIP**
save

mcaraballo, 2023-04-22:
Thank you for the answer.  I will give this a try.

0crap, 2023-06-17:
@haslinghuis Can you please reopen this issue?
It still happens on BF v4.4.2

Original MATEKF405-STD board with MATEKF405STD target gives CALIB message on OSD and can't arm.
Tried with and without bidir dshot, keeps giving CALIB.

Only workaround is disable LED_STRIP.
Flashed back to 4.3.2 solves the issue completely.

haslinghuis, 2023-06-17:
Hi @0crap please leave a support ID. Did you try with dshot_bitbang = ON/OFF?

Note: To be certain use full chip erase and test before restoring settings (only configuring bare minimum).

0crap, 2023-06-18:
> Hi @0crap [LTNS](https://7esl.com/long-time-no-see/) please leave a support ID. What is the receiver type and rate?
> 
> Did you try:
> 
> * set dshot_bitbang = ON/OFF
> * 4.5.0 pre-release
> 
> Note: To be certain use full chip erase and test before restoring settings (only configuring bare minimum).

Thx! :-)

*** Support data submitted *** 
Id: "30b3ba76-3358-4b82-9426-c7be098cc5e1"

bitbang ON/OFF didn't make a difference.
Nope, didn't try 4.5.0 pre-release.

Will test with bare minimum config right now.

haslinghuis, 2023-06-18:
> Remove ledstrip and it all works.

For further investigation you can:
1. debug calibration with https://github.com/betaflight/betaflight/pull/12241
2. increase `gyro_calib_duration` https://github.com/betaflight/betaflight/pull/5932

0crap, 2023-06-18:
I tried the moron stuff, no change.
I tried 4.5.0, no change. (remove LED_STRIP and all is fine.

Debug gave this:

[image]

haslinghuis, 2023-06-18:
Only can think of comparing dma / timers on 4.3.2 and 4.4.2 (with and without ledstrip).
Also might check flashing different 4.4 Release Candidates to find out where the issue started.

0crap, 2023-06-18:
Interesting find, if I disable the BARO it works again!
When BARO is disabled I can enable the LED_STRIP without any issue.
So yeah, dma / timer related stuff I guess.

The MATEK [F405-STD](http://www.mateksys.com/?portfolio=f405-std) does have a BARO.

I don't need the BARO so that fixes it for me to run BF v4.4.2

dzurdo, 2023-09-11:
Hello i am suffering same issue with a Flycolor F7, in 4.3 it was working perfect, once upgraded to 4.4.2 imposible due calib error, i have change the gyro gyro_calib_noise_limit to 150 and removed LED_STRIP from CLI and seems it works, but i am suffering a metallic sound on the motors once armed, I have changed motors for new brand and sounds remains the same.... really frustrated.

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.2 Jul  9 2023 / 10:30:10 (23d066d08) MSP API: 1.45

# config: YES
# board: manufacturer_id: FLCO, board_name: FLYCOLORF

closed 2023-12-17 by haslinghuis

mirik123, 2024-05-31:
I had this problem and it was fixed when I loaded to the BetaFlight the default BetaFPV CLI from their site:
https://support.betafpv.com/hc/en-us/articles/4412322118681-CLI-for-Meteor75-2022