betaflight/betaflight #12230: Acro Trainer mode missing betaflight 4.4, configurator 10.9.0-rc5 acro trainer mode missing

jawz101, 2023-01-24:
### Describe the bug

When I go into the modes tab, unhide all modes, I do not see the option to enable Acro Trainer mode

### To Reproduce

When I go into the modes tab, unhide all modes, I do not see the option to enable Acro Trainer mode

### Expected behavior

Acro Trainer mode able to use

### Support ID

```txt
3fc5fa44-262b-4986-a55c-2746709a75b3
```

### Diff configuration

```txt
# diff

# version
# Betaflight / STM32F411 (S411) 4.4.0 Jan 17 2023 / 01:58:32 (a7cf8f0c6) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name CRAZYBEEF4SX1280
manufacturer_id HAMO

# name: Mobula7 1S

# feature
feature -RX_SERIAL
feature -SOFTSERIAL
feature -AIRMODE
feature INFLIGHT_ACC_CAL
feature RX_SPI

# serial
serial 1 2048 115200 57600 0 115200

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 13 2 1700 2100 0 0
aux 3 28 1 1300 2100 0 0
aux 4 35 5 1700 2100 0 0
aux 5 47 1 1300 1700 0 0

# adjrange
adjrange 0 0 1 900 1200 12 1 0 0
adjrange 1 0 1 1200 1800 12 1 0 0
adjrange 2 0 1 1800 2100 12 1 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWRACE  L FACTORY 5333 5373 5413 5453 5493 5533 5573 5613
vtxtable powerlevels 5
vtxtable powervalues 10 2 14 20 26
vtxtable powerlabels 0 RCE 25 100 400

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 750
set dyn_notch_count = 2
set dyn_notch_max_hz = 1000
set gyro_lpf1_dyn_min_hz = 375
set gyro_lpf1_dyn_max_hz = 750
set acc_lpf_hz = 10
set acc_calibration = -72,-37,27,1
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set vbat_min_cell_voltage = 290
set ibata_scale = 470
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 1
set yaw_deadband = 1
set pid_process_denom = 1
set simplified_gyro_filter_multiplier = 150
set report_cell_voltage = ON
set osd_units = IMPERIAL
set osd_warn_bitmask = 8063
set osd_link_quality_pos = 2135
set osd_link_tx_power_pos = 2103
set osd_tim_1_pos = 2455
set osd_tim_2_pos = 2455
set osd_flymode_pos = 2081
set osd_warnings_pos = 14729
set osd_avg_cell_voltage_pos = 2433
set osd_flip_arrow_pos = 2446
set osd_canvas_height = 13
set debug_mode = RX_EXPRESSLRS_SPI
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 5
set vtx_low_power_disarm = ON
set vtx_freq = 5917
set gyro_1_sensor_align = CW0
set gyro_1_align_yaw = 0
set expresslrs_uid = 172,123,235,247,1,122
set expresslrs_rate_index = 2
set rpm_filter_harmonics = 2
set craft_name = Mobula7 1S
set pilot_name = jawz101

profile 0

# profile 0
set vbat_sag_compensation = 100
set iterm_relax_cutoff = 10
set p_pitch = 75
set i_pitch = 134
set d_pitch = 76
set f_pitch = 199
set p_roll = 71
set i_roll = 127
set d_roll = 67
set f_roll = 191
set p_yaw = 71
set i_yaw = 127
set f_yaw = 191
set d_min_roll = 67
set d_min_pitch = 76
set thrust_linear = 20
set feedforward_smooth_factor = 30
set feedforward_jitter_factor = 9
set dyn_idle_min_rpm = 35
set simplified_master_multiplier = 160
set simplified_d_gain = 140
set simplified_dmax_gain = 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 1
set roll_expo = 55
set pitch_expo = 55
set yaw_expo = 55
set roll_srate = 100
set pitch_srate = 100
set yaw_srate = 70

# end the command batch
batch end

#
```

### Resources configuration

```txt
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: RX_SPI_EXPRESSLRS_RESET
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: RX_SPI_EXPRESSLRS_BUSY
A14: SWD
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 1
TIM3: FREE
TIM4:
    CH1 : MOTOR 2
    CH2 : MOTOR 3
    CH3 : MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 2
DMA1 Stream 1: MOTOR 1
DMA1 Stream 2: SPI_MISO 3
DMA1 Stream 3: MOTOR 3
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 4
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

CRAZYBEEF4SX1280

haslinghuis, 2023-01-24:
Please add `ACRO_TRAINER` as custom defines while flashing to add required features.

jawz101, 2023-01-24:
> Please add `ACRO_TRAINER` as custom defines while flashing to add required features.

Thank you.  Is there a document listing the possible custom defines?  It makes me wonder what I will miss out on without knowing what is stripped out.
Or have a checkbox list in the configurator listing all of the defines to pick which to include.

haslinghuis, 2023-01-25:
Jump to https://betaflight.com/docs/configurator/firmware-flasher-tab#custom-defines

jawz101, 2023-01-25:
I tried both `ACRO_TRAINER` as you mentioned above and `ACRO TRAINER` as mentioned in the documentation and neither made Acro Trainer available in Betaflight Configurator.

haslinghuis, 2023-01-25:
Adding `ACRO_TRAINER` with `_` (documentation is missing that) works for me on new build:

[image]

haslinghuis, 2023-01-25:
[image]

https://build.betaflight.com/api/builds/3fc5546fcbd6bacb6798b23158b888a4/log

closed 2023-01-25 by haslinghuis

jawz101, 2023-01-25:
ok it's working now.  Thanks.  BTW, what is DASHBOARD AND PIN IO?  I don't understand what those might be.

haslinghuis, 2023-01-25:
Check our new documentation platform on betaflight.com - there is a search option explaining in detail - at least for PINIO :grin: