betaflight/betaflight #12239: custom GPS commands

nightflasher, 2023-01-26:
### Is your feature request related to a problem? Please describe

when using older GPS modules or non-ublox (GK or MT) which only speak NMEA, and don't save settings, you have to configure the GPS every time you power up the miniKwad. Most of those GPS modules start with 9600 baud and have only 1 Hz update/refresh rate. 

in my case, betaflight detects 57600 baud, which isn't working even you've tried to configure the module. so i've configured it to 38400 baud and set up 5 Hz, which is working. But after a while without power, the settings in the module are reset to 9600 baud &1 Hz.

### Describe the solution you'd like

two input fields for manual expert configuration for the GPS that is sent to the GPS if the check for the set baudrate (e.g. 38400) in ports tab failed to get valid input.

baudrate: 3800 ($PMTK251,38400*27\r\n)
refresh: 5 ($PMTK220,200*2C\r\n)

### Describe alternatives you've considered

as you already search for the correct settings in gps.c from line 120 and follows (and set those again?) you could fallback to 9600 and set the "saver" and most commonly available baudrate which supports 5 Hz.

haslinghuis, 2023-01-26:
https://cdn-shop.adafruit.com/datasheets/PMTK_A08.pdf

fabi9163, 2023-02-04:
Great idea, it could not only be used for baud-rates but also to set manual configs like GNSS eg. for modules without flash-memory like the BN-180. [Such tutorials](https://oscarliang.com/gps-settings-u-center/) could be replaced with a single CLI-Command.

My problem now 'using Beidou instead of Glonass' which seems to be perform lot better here (higher hdop) could be easily and permanently solved

PaulFPV, 2023-02-06:
I agree that we need the possibility to setup a couple more options about gps directly in betaflight. I setup everything in u-center and it works perfectly, but after some hours betaflight cannot detect it at all... then I enable auto setup and it starts working again... but then the changes made in u-center are lost (as example.. Beidou instead of Glonass, exactly like fabi9163).

Receiver settings have been moved to the receiver tab from "configure", what do you thing about doing the same for GPS?

So, in configuration leave just "enable GPS" and the other settings could go in the GPS tab (so when you change something you see what happens in realtime).

As said in Oscar's GPS tutorial, it's useful to change port settings and speed, refresh rate, sats constellations (Beidou, glonass,..), power (not shure it makes a difference), dynamic platform (portable), maybe even assistnow...

An alternative could be a simpler setup by country (around europe Beidou seems to be better than glonass), but let choose a couple more details could be better.

nightflasher, 2023-02-06:
Moving expert settings for gps to the gps tab makes totally sense. But why do I need to enable gps in configuration, when I also need to setup a port? 
If port is gps - gps is enabled!?

haslinghuis, 2023-02-07:
@nightflasher - with the change I made if you enable GPS in ports tab it will enable GPS feature.

This change makes moving GPS configuration to GPS tab possible as this is on my todo.

- Adding custom GPS commands needs firmware changes, as if we would send a GPS command sequence using MSP, retrieving the response might be tricky. - To have a fresh look deliberately did not read gps.c yet but guess we could implement custom commands there to solve the response issue.
- Changing GPS baudrate is not that easy according the tutorial - as it requires baud rate change on parts tab plus save and reboot. Likewise Ublox UCenter seems to have the same constraint - autobauding.

haslinghuis, 2023-02-07:
Suggestions are welcome. on UI :)

[image]

haslinghuis, 2023-02-08:
Please test https://github.com/betaflight/betaflight-configurator-nightlies/releases/tag/v20230208.381

Did not add custom command yet.

nightflasher, 2023-03-28:
For now about three days purely getting my gps sets to work (didn't expect to take so much time) as they should, i was reading the betaflight gps c/h and sensors sources, to find the NMEA thing is quite simple to setup:
Situation/Problem = S | Fix/Solution = F
S: my module defaults always to 9600
F: betaflight can and does set it to 57600
S: some modules default to 1Hz
F: betaflight CAN change it to 10Hz the same way it does the baud rate change, without user knowledge
    if the 10Hz setting fails (sometimes it does!), i found to have 5Hz working instead and than should be 
    possible to change again to 10Hz
S: one module (GTPA010) defaults to RMC/GGA, i don't know if that is sufficient for betaflight gps functions (eg RTH)
F: single line can change the NMEA outputs as well
S: i found DGPS is some kind of date/time correction, do we need it? Yes? SBAS or RTCM
F: another simple line

I've now a list of testet commands [GK9501 (GY-NEO6MV2), GTPA010, MT3337], which i tested using arduino terminal (send command) and MiniGPS1.7.1 (verify ettings):
//GPS+GLONASS
$PMTK353,1,1,0,0,0*2B\r\n
//GPS only
$PMTK353,1,0,0,0,0*2A\r\n
//BAUD RATE 57600
$PMTK251,57600*2C\r\n
//10Hz REFRESH
$PMTK220,100*2F\r\n
//NMEA OUTPUTS
$PMTK314,0,1,0,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0*28\r\n
//disable DGPS
$PMTK301,0*2C\r\n

https://nmeachecksum.eqth.net/
[MiniGPS_v1.7.1.zip](https://github.com/betaflight/betaflight/files/11088079/MiniGPS_v1.7.1.zip)

ASDosjani, 2023-03-28:
> Your PR isn't related to this issue, but I like your idea and smart changes. Does it improve something, or is it just "the correct form of configuration"?

I can't test now but based on Oscar Liang's tutorial it has lower deviation.

ASDosjani, 2023-03-28:
@nightflasher can you test this PR? I don't have NMEA gps.

nightflasher, 2023-03-31:
> https://cdn-shop.adafruit.com/datasheets/PMTK_A08.pdf

for the B&T BG01 from AliExpress and similar GK chipsets 
https://www.cdebyte.com/Uploadfiles/Files/2022-7-6/2022761056555773.pdf

I've also updated a the commands list for users to test:
//MTK GPS+GLONASS (doesn't work for me)
$PMTK353,1,1,0,0,0*2B\r\n
//MTK GPS only (works fine)
$PMTK353,1,0,0,0,0*2A\r\n
//MTK BAUD RATE 57600
$PMTK251,57600*2C\r\n
//MTK 5Hz REFRESH
$PMTK220,200*2C\r\n
//MTK 10Hz REFRESH
$PMTK220,100*2F\r\n
//MTK NMEA OUTPUTS
$PMTK314,0,1,0,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0*28
//MTK disable DGPS
$PMTK301,0*2C
//Firmware Version & Model
$PMTK605*31\r\n

//GK NMEA OUTPUT
$PGKC146,3,3,57600*0F
//GK BAUD RATE 57600
$PGKC147,57600*35
//GK 10Hz REFRESH 
$PGKC101,100*32
//GK GPS+GLONASS (doesn't work for me)
$PGKC115,1,1,0,0*2A\r\n
//GK GPS only (works fine)
$PGKC115,1,0,0,0*2B\r\n
//GK NMEA OUTPUTS
$PGKC242,0,1,0,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0*37\r\n

https://nmeachecksum.eqth.net/

PR #12591 (2023-05-19): Add NMEA Custom commands
Resolves: #12239 
You can add a custom NMEA commands with `set gps_nmea_custom_commands =` and the commands you want.
You can add multiple commands with a space ' ' between them.
Commands will be sent to the gps after the initialization.

I can't test because I only have UBLOX gps.
closed 2023-05-19 by haslinghuis