betaflight/betaflight #12245: Question about cloud build

TheIsotopes, 2023-01-28:
### Describe the bug

I'm testing latest master with core and cloud build on a MATEKF22MINI but i can not working with betaflight tx lua scripts.
the core version hangs on `Waiting for API version` and the cloud build version hangs on `Downloading VTX tables`.

here the build configuration for cloud build:
[image]

which custom defines do I have to use for this to work.

### To Reproduce

use MATEKF22MINI as described above

### Expected behavior

working betaflight tx lua scripts

### Diff configuration

```txt
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 25 2023 / 09:26:25 (0151c3006) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name MATEKF722MINI
manufacturer_id MTKS

# name: F7-Sbus

# feature
feature LED_STRIP

# serial
serial 0 32 115200 57600 0 115200

# led
led 0 6,1::C:0
led 1 6,2::C:0
led 2 6,3::C:0
led 3 6,4::C:0
led 4 9,4::C:0
led 5 9,3::C:0

# aux
aux 0 0 0 1900 2100 0 0
aux 1 13 1 900 1100 0 0
aux 2 36 1 1900 2100 0 0

# master
set acc_hardware = NONE
set rc_smoothing_auto_factor = 25
set serialrx_provider = SBUS
set blackbox_sample_rate = 1/8
set motor_pwm_protocol = DSHOT600
set ibata_scale = 125
set telemetry_disabled_mode = ON
set telemetry_disabled_acc_x = ON
set telemetry_disabled_acc_y = ON
set telemetry_disabled_acc_z = ON
set telemetry_disabled_pitch = ON
set telemetry_disabled_roll = ON
set telemetry_disabled_heading = ON
set telemetry_disabled_altitude = ON
set telemetry_disabled_vario = ON
set telemetry_disabled_lat_long = ON
set telemetry_disabled_ground_speed = ON
set telemetry_disabled_distance = ON
set ledstrip_visual_beeper = ON
set ledstrip_visual_beeper_color = RED
set ledstrip_profile = RACE
set craft_name = F7-Sbus

profile 0

# profile 0
set feedforward_smooth_factor = 20
set feedforward_jitter_factor = 6

rateprofile 0

# rateprofile 0
set roll_rc_rate = 20
set pitch_rc_rate = 20
set yaw_rc_rate = 20
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 50
set roll_srate = 99
set pitch_srate = 99
set yaw_srate = 72

# end the command batch
batch end

#
```

### Resources configuration

```txt
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED_STRIP
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: PREINIT 2
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: LED_STRIP
DMA2 Stream 4: ADC
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

### Flight controller

MATEKF722MINI

klutvott123, 2023-01-28:
add VTX under other options and it should work

haslinghuis, 2023-01-28:
See: https://github.com/betaflight/betaflight/commit/dbc647d4648d133357c7cea7b6eb862fc9f9eb65

TheIsotopes, 2023-01-28:
omg, i probably didn't see the forest for the trees.

working now. thx @klutvott123

closed 2023-01-28 by TheIsotopes