betaflight/betaflight #12246: The Mag (SPI AK8963) on AFNG controllers is only working if MAG option is configured during build.

haslinghuis, 2023-01-28:


haslinghuis, 2023-06-28:
@MJ666 please test issue with latest zulu (4.4.3 and 4.5.0).

Please use nightly configurator to fix issues with custom defines etc.

MJ666, 2023-06-30:
With Bf 4.5.0 some setting for the mag are not properly applied even if the are in the configuration? Apply the following settings manually will enable the mag (SPI AK8963 and the internal mag of a SPI MPU9250). They also show up properly in the CLI status command.

resource COMPASS_CS 1 C15
set mag_spi_device = 3

They also show up properly in the CLI status command. Anyhow the internal mag of a SPI MPU9250 is not showing any sensor data. The SPI AK8963 is working.

haslinghuis, 2023-06-30:
Configuration definite needs updating. For instance I think:

- #define `MAG_SPI_DEVICE` should be `MAG_SPI_INSTANCE`
- #define `COMPASS_CS_PIN` should be `MAG_CS_PIN`
- We might need to add `USE_MAG` like we do for ACC, BARO, FLASH, GYRO, MAX7456
- Need to add USE_SPI_MAG if applicable.

haslinghuis, 2023-06-30:
@MJ666 updated the ALIENFLIGHTNGF7. Please check this (might to need to force new build by adding new option or custom define - otherwise we get a cached version).

closed 2023-06-30 by haslinghuis

MJ666, 2023-07-01:
All looks to be working well now for 4.5.0. Many thanks for your help.