betaflight/betaflight #12249: Missing Acro Trainer feature

thegettys, 2023-01-29:
### Describe the bug

Acro trainer option is missing in 4.4. From what I can see `#define USE_ACRO_TRAINER` can no longer be found in source

### To Reproduce

Go to modes and look for acro trainer even with an accelerometer present

### Expected behavior

Acro trainer option should be available to FCs with accelerometer

### Support ID

```txt
86bdff37-e913-443c-936f-e88d34104a27
```

### Diff configuration

```txt
# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 29 2023 / 04:59:00 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name CLRACINGF7
manufacturer_id CLRA

# name: x1

# feature
feature ESC_SENSOR

# serial
serial 3 64 115200 57600 0 115200
serial 4 1024 115200 57600 0 115200

# beacon
beacon RX_SET

# aux
aux 0 0 0 1900 2100 0 0
aux 1 1 1 1275 1400 0 0
aux 2 2 1 1400 1600 0 0
aux 3 13 1 1100 1225 0 0
aux 4 35 1 1775 1900 0 0

# rxfail
rxfail 4 s 1000
rxfail 5 s 1200

# master
set gyro_to_use = FIRST
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = -183,-57,17,1
set fpv_mix_degrees = 20
set dshot_bidir = ON
set current_meter = ESC
set battery_meter = ESC
set osd_vbat_pos = 14688
set osd_link_quality_pos = 14368
set osd_flymode_pos = 14393
set osd_vtx_channel_pos = 14356
set osd_mah_drawn_pos = 14744
set osd_craft_name_pos = 39
set osd_pilot_name_pos = 14336
set osd_warnings_pos = 14601
set osd_avg_cell_voltage_pos = 14720
set osd_battery_usage_pos = 14729
set osd_disarmed_pos = 14538
set osd_canvas_height = 13
set vcd_video_system = NTSC
set craft_name = x1
set pilot_name = z

profile 0

rateprofile 0

# end the command batch
batch end
```

### Resources configuration

```txt
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: PINIO 1
A15: OSD_CS
B00: LED 1
B01: FREE
B02: FREE
B03: FREE
B04: BEEPER
B05: SPI_MOSI 3
B06: MOTOR 1
B07: MOTOR 2
B08: MOTOR 3
B09: MOTOR 4
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: MCO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: PREINIT 2
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1 : DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

### Flight controller

Talon F7 as well as Kakute H7

Denethor85, 2023-01-29:
"Please add ACRO_TRAINER as custom defines while flashing to add required features."....

SentinelFour, 2023-01-29:
Running into the same issue. What confuses me is that the setting "Acro Trainer Angle Limit" is still available in the PID Tuning tab.

thegettys, 2023-01-29:
I have added ACRO_TRAINER to custom defines in the firmware flash section of the configurator, but without any luck. Here is the output from that build, but ACRO_TRAINER does not appear to get passed to the make command. In fact nothing I put into the custom defines seems to alter the output of the build though changing radio, telemetry, and other options do affect build output 
```
***** Commencing Build *****

VERSION: 4.4.0
COMMIT:  4.4.0
BOARD:   CLRACINGF7/STM32F7X2
STARTED: 2023-01-29 04:58:55Z

Want to run this build locally (note "origin" needs to be the Betaflight Repo, not your fork)?

git fetch origin 4.4.0 && git checkout -f FETCH_HEAD && make TARGET=STM32F7X2 EXTRA_FLAGS="-D'BUILD_KEY=e0f33c3ba4cd26e531dfba3d586f5c96' -D'RELEASE_NAME=4.4.0' -D'BOARD_NAME=CLRACINGF7' -D'MANUFACTURER_ID=CLRA' -DCLOUD_BUILD -DUSE_ACC -DUSE_ACC_SPI_MPU6000 -DUSE_ACC_SPI_MPU6500 -DUSE_DSHOT -DUSE_FLASH -DUSE_FLASH_W25N01G -DUSE_FLASH_W25Q128FV -DUSE_GYRO -DUSE_GYRO_SPI_MPU6000 -DUSE_GYRO_SPI_MPU6500 -DUSE_MAX7456 -DUSE_OSD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX"

Docker log: 

Pulling from Github (4.4.0):
From https://github.com/betaflight/betaflight
 * [new tag]             4.4.0      -> CLOUD_HEAD

Checking out the branch -> CLOUD_HEAD (for 4.4.0)
Switched to branch 'CLOUD_HEAD'

Commencing the build (make STM32F7X2):
make -j ./obj/betaflight_4.4.0_STM32F7X2.hex
make[1]: Entering directory '/betaflight'
rm -f ./obj/main/STM32F7X2/.efhash_*
EF HASH -> ./obj/main/STM32F7X2/.efhash_e161434e3ddf90c117102f3361adcd3b
%% startup_stm32f722xx.s 
%% (optimised) ./src/main/startup/system_stm32f7xx.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu.c 
%% (optimised) ./src/main/drivers/adc_stm32f7xx.c 
%% (optimised) ./src/main/drivers/audio_stm32f7xx.c 
%% (speed optimised) ./src/main/drivers/bus_i2c_hal.c 
%% (size optimised) ./src/main/drivers/bus_i2c_hal_init.c 
%% (size optimised) ./src/main/drivers/bus_i2c_timing.c 
%% (optimised) ./src/main/drivers/dma_stm32f7xx.c 
%% (size optimised) ./src/main/drivers/light_ws2811strip_hal.c 
%% (size optimised) ./src/main/drivers/transponder_ir_io_hal.c 
%% (speed optimised) ./src/main/drivers/bus_spi_ll.c 
%% (optimised) ./src/main/drivers/persistent.c 
%% (optimised) ./src/main/drivers/dshot_bitbang.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_decode.c 
%% (optimised) ./src/main/drivers/dshot_bitbang_ll.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_hal.c 
%% (speed optimised) ./src/main/drivers/pwm_output_dshot_shared.c 
%% (optimised) ./src/main/drivers/timer_hal.c 
%% (optimised) ./src/main/drivers/timer_stm32f7xx.c 
%% (optimised) ./src/main/drivers/system_stm32f7xx.c 
%% (optimised) ./src/main/drivers/serial_uart_hal.c 
%% (optimised) ./src/main/drivers/serial_uart_stm32f7xx.c 
%% (optimised) ./src/main/drivers/sdio_f7xx.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_fake.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_mpu3050.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_mpu6050.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_mpu6500.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_spi_bmi160.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_spi_bmi270.c 
%% (optimised) ./src/main/drivers/accgyro/accgyro_spi_icm20649.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_spi_icm20689.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_spi_icm426xx.c 
%% (optimised) ./src/main/drivers/accgyro/accgyro_spi_l3gd20.c 
%% (speed optimised) ./src/main/drivers/accgyro/accgyro_spi_lsm6dso.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_spi_lsm6dso_init.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_spi_mpu6000.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_spi_mpu6500.c 
%% (size optimised) ./src/main/drivers/accgyro/accgyro_spi_mpu9250.c 
%% (optimised) ./src/main/drivers/accgyro/gyro_sync.c 
%% (optimised) lib/main/BoschSensortec/BMI270-Sensor-API/bmi270_maximum_fifo.c 
%% (size optimised) ./src/main/drivers/barometer/barometer_2smpb_02b.c 
%% (size optimised) ./src/main/drivers/barometer/barometer_bmp085.c 
[... 374 lines cut ...]
ITCM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     ITCM_FLASH1:          0 GB       480 KB      0.00%
      AXIM_FLASH:        2364 B        10 KB     23.09%
AXIM_FLASH_CUSTOM_DEFAULTS:           8 B         6 KB      0.13%
AXIM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     AXIM_FLASH1:      329890 B       480 KB     67.12%
AXIM_FLASH_CUSTOM_DEFAULTS_EXTENDED:          0 GB         0 GB
        DTCM_RAM:       34172 B        64 KB     52.14%
           SRAM1:       49232 B       176 KB     27.32%
           SRAM2:          0 GB        16 KB      0.00%
       MEMORY_B1:          0 GB         0 GB
   text	   data	    bss	    dec	    hex	filename
 326058	   6204	  77204	 409466	  63f7a	./obj/main/betaflight_STM32F7X2.elf
Creating HEX ./obj/betaflight_4.4.0_STM32F7X2.hex 
make[1]: Leaving directory '/betaflight'

Copying hex files (to output):
'/betaflight/obj/betaflight_4.4.0_STM32F7X2.hex' -> '/output/betaflight_4.4.0_STM32F7X2.hex'
 -> completed

 ***** Completed Build *****
BUILD TIME: 2023-01-29 04:59:16Z --- 21.73

Submitted: 1/29/2023 4:58:53 AM, wait time: 23.73
```

haslinghuis, 2023-01-29:
Your log does not show inclusion. Please show a full screenshot of the firmware flasher tab.

[image]

When I add `ACRO_TRAINER` I get:

[image]

thegettys, 2023-01-29:
[image]
 
I do see launch control but not Acro Trainer

[image]

haslinghuis, 2023-01-29:
You might need to `enable expert mode` - becomes available after enabling `show release candidates`.

thegettys, 2023-01-29:
That got it working. I though `Pertmanently enable Expert Mode` and the presence of the custom defines would have been enough to pass them along. Thanks for the help.

closed 2023-01-29 by thegettys