betaflight/betaflight #12250: CAMERA_CONTROL is unavailable in 4.4 in F411

vodka-bears, 2023-01-29:
### Describe the bug

CAMERA_CONTROL stuff isn't being built in even if I specify CAMERA_CONTROL in custom defines.

### To Reproduce

1. Build and flash Betaflight 4.4.0 with or without CAMERA_CONTROL custom define
2. Open CLI and type `resource CAMERA_CONTROL 1 [some pin]`

### Expected behavior

Camera control feature is expected to be available

### Support ID

```txt
*** Support data submitted *** 
Id: "23e5f3fd-7ea4-44ed-97b8-237257f6c0bc"
```

### Diff configuration

```txt
diff

# version
# Betaflight / STM32F411 (S411) 4.4.0 Jan 29 2023 / 09:07:27 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name CRAZYBEEF4SX1280
manufacturer_id HAMO

# name: 65BLACK

# resources
resource SERIAL_RX 2 NONE

# feature
feature -RX_SERIAL
feature -SOFTSERIAL
feature RX_SPI

# serial
serial 1 2048 115200 57600 0 115200

# beacon
beacon RX_SET

# aux
aux 0 0 0 1500 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1300 1700 0 0
aux 4 39 3 1700 2100 0 0
aux 5 35 2 1700 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWRACE  L FACTORY 5333 5373 5413 5453 5493 5533 5573 5613
vtxtable powerlevels 5
vtxtable powervalues 10 2 14 16 26
vtxtable powerlabels 0 RCE 25 100 400

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set acc_calibration = 42,-150,24,1
set rc_smoothing_auto_factor = 25
set rc_smoothing_auto_factor_throttle = 25
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_output_reordering = 3,2,1,0,4,5,6,7
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 310
set vbat_warning_cell_voltage = 330
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warn_bitmask = 33791
set osd_rssi_dbm_alarm = -99
set osd_tim1 = 2579
set osd_tim2 = 2577
set osd_vbat_pos = 2434
set osd_rssi_pos = 377
set osd_link_quality_pos = 2421
set osd_rssi_dbm_pos = 2392
set osd_tim_2_pos = 2452
set osd_throttle_pos = 2402
set osd_crosshairs_pos = 2253
set osd_current_pos = 2133
set osd_mah_drawn_pos = 2114
set osd_craft_name_pos = 393
set osd_pilot_name_pos = 2441
set osd_warnings_pos = 14696
set osd_disarmed_pos = 2315
set osd_stat_bitmask = 8603372
set osd_canvas_height = 13
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 3
set vtx_freq = 5732
set gyro_1_sensor_align = CW180
set gyro_1_align_yaw = 1800
set expresslrs_uid = 0,0,53,102,161,123
set expresslrs_switch_mode = HYBRID
set craft_name = 65BLACK
set pilot_name = VODKABEARS

profile 0

# profile 0
set vbat_sag_compensation = 100
set angle_level_strength = 70
set level_limit = 85
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 65
set feedforward_jitter_factor = 3
set feedforward_boost = 18

rateprofile 0

# rateprofile 0
set roll_rc_rate = 33
set pitch_rc_rate = 33
set yaw_rc_rate = 33
set roll_expo = 54
set pitch_expo = 54
set yaw_expo = 54
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50

# end the command batch
batch end
```

### Resources configuration

```txt
# resource
resource BEEPER 1 C15
resource MOTOR 1 B10
resource MOTOR 2 B06
resource MOTOR 3 B07
resource MOTOR 4 B08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_RX 1 A10
resource LED 1 C13
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource ADC_BATT 1 B00
resource ADC_CURR 1 B01
resource OSD_CS 1 B12
resource RX_SPI_CS 1 A15
resource RX_SPI_EXTI 1 C14
resource RX_SPI_BIND 1 B02
resource RX_SPI_LED 1 B09
resource RX_SPI_EXPRESSLRS_RESET 1 A08
resource RX_SPI_EXPRESSLRS_BUSY 1 A13
resource GYRO_EXTI 1 A01
resource GYRO_CS 1 A04
```

### Flight controller

Happymodel DiamondF4 ELRS

### Other components

Caddx Ant Lite

### How are the different components wired up

Camera control wire is soldered at RX2

haslinghuis, 2023-01-29:
It is available. You have to find a pin to use for the CAMERA_CONTROL resource.

[image]

vodka-bears, 2023-01-29:
@haslinghuis It's not.
```
# resource CAMERA_CONTROL 1 A03
###ERROR IN resource: INVALID RESOURCE NAME: 'CAMERA_CONTROL'###
````

```
# get camera
osd_camera_frame_pos = 35
Allowed range: 0 - 65535

osd_camera_frame_width = 24
Allowed range: 2 - 30

osd_camera_frame_height = 11
Allowed range: 2 - 16
````

haslinghuis, 2023-01-29:
Please add `CAMERA_CONTROL` to Custom Defines and reflash the firmware

[image]

```
# resource CAMERA_CONTROL 1 C15

NOTE: C15 already assigned to BEEPER.

Resource is set to C15
```

[image]

alexdibi, 2023-01-29:
same problem on f745...

ggoraa, 2023-01-29:
Same thing on F722

haslinghuis, 2023-01-29:
If adding the custom define `CAMERA_CONTROL` does not work while flashing please turn on `show release candidates` and enable `expert mode`.

ggoraa, 2023-01-29:
That worked, nice!

NameOfTheDragon, 2023-01-30:
FWIW this did work for me on my SpeedyBee F7 Mini (F722). After flashing the cloud build, the camera control pin is mapped as part of the custom defaults for the board and appears in the list of mapped resources as exp[ected.

I do think its a bit odd hiding this away under developer/debug options. Why not make it appear in the drop-down list?

ggoraa, 2023-01-31:
> Why not make it appear in the drop-down list?

And also fix `defaults` not working properly, which results in backups from `diff all` not working properly because of an error while running `defaults` (it complains about CAMERA_CONTROL`)

vodka-bears, 2023-02-08:
Seems fixed now, Cam. Control is now included in Other Options.

closed 2023-02-08 by vodka-bears

shrung, 2023-03-01:
Hi, I'm not seeing camera control at the moment either, using 4.4.0 from 1/29/23. Flashing on a JB AIO v2.1, which has a DAC pin for cam. control. I added camera control when flashing, and also set the camera_control_mode = DAC, how can I enable this ability in the OSD? Thanks for any help.