betaflight/betaflight #12254: 4.4.0 official FW FOXEERF722V2 freeze trying open any tab and fail to open serial port

AndroidGX, 2023-01-29:
### Describe the bug

I have two FOXEERF722V2 FC's, the first didn't has any issue at all to migrate from 4.4 RC5 to official 4.4, but the second one yes.
The main issue is that after flashing properly the 4.4. FW and after importing my full CLI (via preset save from RC5) right after that, the communication seems to be stopped and I can't go on any tab at all, and e message "**Failed to open serial port**" is displayed ((please check the joined video to show).
I precise I use (of course) latest 10.9 configurator for 4.4 FW.

So I made some test trying to flash back RC5 (still via 10.9 configurator) and re-import my CLI : **works fine back !** (please check the joined video to show).

So definitely, there is somethings wrong with latest official 4.4.0 FW.
Please let me know if you guys needs anything/tests.

### To Reproduce

1. Flash FOXEERF722V2 officiel 4.4.0 firmware
2. Import my CLI (avoid the two minus errors)
3. Navigate in any tab from the configurator

### Expected behavior

Should navigate after that in any tab from the configurator (like with RC5 FW).

### Support ID

```txt
Id: "4b8c9727-e5f5-42ea-9061-8966e23e4c31"
```

### Diff configuration

```txt
diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 29 2023 / 11:20:04 (a7cf8f0c6) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name FOXEERF722V2
manufacturer_id FOXE

# name: Rival HD 5p

# resources
resource MOTOR 1 C08
resource MOTOR 2 C09
resource MOTOR 3 A08
resource MOTOR 4 A09

# serial
serial 1 64 115200 57600 0 115200
serial 2 2 115200 115200 0 115200
serial 4 131073 115200 57600 0 115200

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# map
map TAER1234

# aux
aux 0 0 0 1975 2050 0 0
aux 1 1 2 1475 1525 0 0
aux 2 13 1 950 1025 0 0
aux 3 35 1 1975 2050 0 0
aux 4 36 3 1975 2050 0 0

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_static_hz = 210
set dyn_notch_count = 1
set dyn_notch_q = 500
set dyn_notch_max_hz = 320
set gyro_lpf1_dyn_min_hz = 0
set acc_calibration = -76,-1,204,1
set min_check = 1015
set max_check = 2000
set rssi_channel = 15
set rc_smoothing_auto_factor = 45
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 180
set vbat_warning_cell_voltage = 365
set vbat_scale = 113
set ibata_scale = 127
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 3
set yaw_deadband = 3
set simplified_gyro_filter = OFF
set osd_warn_bitmask = 639
set osd_rssi_alarm = 90
set osd_cap_alarm = 0
set osd_tim1 = 769
set osd_tim2 = 1280
set osd_vbat_pos = 1
set osd_rssi_pos = 2070
set osd_link_quality_pos = 390
set osd_tim_1_pos = 404
set osd_tim_2_pos = 404
set osd_flymode_pos = 54
set osd_throttle_pos = 356
set osd_current_pos = 0
set osd_mah_drawn_pos = 25
set osd_craft_name_pos = 15
set osd_gps_speed_pos = 395
set osd_gps_lon_pos = 0
set osd_gps_lat_pos = 32
set osd_gps_sats_pos = 64
set osd_home_dir_pos = 160
set osd_home_dist_pos = 161
set osd_altitude_pos = 128
set osd_power_pos = 388
set osd_warnings_pos = 2393
set osd_avg_cell_voltage_pos = 3076
set osd_disarmed_pos = 300
set osd_stat_bitmask = 1632
set osd_displayport_device = MSP
set osd_canvas_width = 60
set osd_canvas_height = 22
set debug_mode = GYRO_SCALED
set vcd_video_system = HD
set gyro_1_align_yaw = 2700
set craft_name = Rival HD 5p
set pilot_name = Feisar

profile 0

# profile 0
set profile_name = Tuned
set dterm_lpf1_dyn_min_hz = 105
set dterm_lpf1_dyn_max_hz = 210
set dterm_lpf1_dyn_expo = 7
set dterm_lpf1_static_hz = 105
set dterm_lpf2_static_hz = 210
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set yaw_lowpass_hz = 0
set throttle_boost = 0
set p_pitch = 67
set i_pitch = 120
set d_pitch = 38
set f_pitch = 29
set p_roll = 53
set i_roll = 95
set d_roll = 28
set f_roll = 23
set p_yaw = 53
set i_yaw = 95
set f_yaw = 23
set d_min_roll = 28
set d_min_pitch = 38
set thrust_linear = 25
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 65
set dyn_idle_min_rpm = 30
set simplified_d_gain = 95
set simplified_pi_gain = 120
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 20
set simplified_pitch_d_gain = 120
set simplified_pitch_pi_gain = 120
set simplified_dterm_filter_multiplier = 140

rateprofile 0

# rateprofile 0
set rateprofile_name = Feisar
set rates_type = BETAFLIGHT
set roll_rc_rate = 110
set pitch_rc_rate = 110
set yaw_rc_rate = 110
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 60

# end the command batch
batch end

#
```

### Resources configuration

```txt
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: BEEPER
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: FREE
B01: FREE
B02: GYRO_CS 1
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: FREE
C02: ADC_CURR
C03: PREINIT 3
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 1
C09: MOTOR 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: FREE
C14: FREE
C15: LED 1
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 3
    CH4 : DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 1

#
```

### Flight controller

FOXEERF722V2 

### Other components

Caddx Vista
DIATONE MAMBA F50A PRO ESC 4IN1
ELRS Happymodel EP1 (2.4Ghz)

### How are the different components wired up

Very clean build (I have now very good experience building quads and make perfect Betaflight tunings in all my quads).

haslinghuis, 2023-01-29:
@AndroidGX to rule out production vs debug build could you also test with latest development version

AndroidGX, 2023-01-29:
> @AndroidGX to rule out production vs debug build could you also test with latest development version

Sure, I can do that 👍 
Can you please point me where I can grab it to flash it ?

AndroidGX, 2023-01-31:
I tried the latest zulu 4.5.0 FW as from yesterday and same issue : 

[image]
[image]

ggoraa, 2023-01-31:
I've had the same issue before even on BF 4.3, but that was fixed by doing these steps:

- Boot FC in DFU mode
- Open firmware flasher
- Flash with "full erase" enabled

AndroidGX, 2023-01-31:
> I've had the same issue before even on BF 4.3, but that was fixed by doing these steps:
> 
> * Boot FC in DFU mode
> * Open firmware flasher
> * Flash with "full erase" enabled

I always flash with "full erase"
Unfortunately, I already tried this too, without any success

AndroidGX, 2023-01-31:
So, I investigate a bit more tonight and I figured out something interesting.
After many attempts to flash and add portions of CLI to update little by little, I now truely thinks that the issue is what following the « **set gyro_lpf1_static_hz = 0** » line from my whole full CLI.
The order of the commands line right after this one seems to have issues to brings back connexion (but whithout beeing sure it’s really this CLI line that is causing the fail).

haslinghuis, 2023-02-01:
Build picture please. What is the distance between ESC and FC as too less filtering on gyro noise seems to be the issue here?

AndroidGX, 2023-02-01:
> Build picture please. What is the distance between ESC and FC as too less filtering on gyro noise seems to be the issue here?

Definitely not, since I always flew like this since 4.3 without any issue, I use well blackbox and know how to tune properly (no need to worry).
But I don't know why this case should require pictures of my build since this this has nothing to do with a story of tuning.
But here you go, here is a photo as requested : 

[image]

I use almost the same way of tuning in both my 6" and 5" rigs in filters part. The 6" (which also has FOXEERF722V2 on it) don't have any issue at all while restoring the full CLI (as explained in my first post).
I can't really explain why the full CLI in my 6" is working but not in 5". It's an interesting case

haslinghuis, 2023-02-01:
Please provide the support ID from the 6"  build as we only have ID 4b8c9727-e5f5-42ea-9061-8966e23e4c31 to be able to compare.

Does the 6" use the exact same components (ESC, Vista, ELRS)?

AndroidGX, 2023-02-01:
> Please provide the support ID from the 6" build as we only have ID 4b8c9727-e5f5-42ea-9061-8966e23e4c31 to be able to compare.
> 
> Does the 6" use the exact same components (ESC, Vista, ELRS)?

here is the support id from my 6" : Id: "e7b9eb59-6ff0-444c-b7fe-ffcb01a3e125"
Yes, exact same FC, but not the same ESC (Tekko32 F3 45A 4IN1). Vista and ELRS (but different manufacturer) also.

AndroidGX, 2023-02-01:
ok I figured out and solved the issue by making a compare between my previous full CLI preset saved from 4.4.0 RC5, and a new one that I did (after setting everything almost manually back, with the portions that were working) from official 4.4.0, and here are the only diff that has been remarked : 

[image]

So I made further tests and looks like I can't save until I want my value (dyn_notch_max_hz) back to 320 via the configurator.
If I try, it will not save to EEPROM and get stuck again.
I tried with 600 to 500 = OK!
I tried with 500 to 400 = OK!
I tried with 400 to 300 = OK!
but if I try to set 320 = NOK!

Now I'm not sure if it's an issue from the configurator or the FW itself.
But I'm glad I finally discovered the main issue here

What do you think @haslinghuis ?

haslinghuis, 2023-02-01:
@AndroidGX :tada: you have spotted a bug.

# get dyn_notch_max_hz
dyn_notch_max_hz = 600
Allowed range: 200 - 1000

Yes can confirm trying to save 320 FC communication is lost. Have to replug USB to regain access. Settings are not saved.
CLI will save the setting and although can connect initially communication is cut off while retaining serial connection.

| OK | FAILS |
| --: | --: |
| 300 | |
| 319 | |
| | 320 |
| | 321 |
| | 330 |
| | 340 |
| | 350 |
| | 360 |
| 370 | |

nerdCopter, 2023-02-01:
i don't understand why it even accepted `0`
```
# set dyn_notch_max_hz = 199
###ERROR IN set: INVALID VALUE###
Allowed range: 200 - 1000

# set dyn_notch_max_hz = 0
###ERROR IN set: INVALID VALUE###
Allowed range: 200 - 1000
```

closed 2023-02-09 by haslinghuis