betaflight/betaflight #12255: no blackbox found 4.4.0 MAMBAF722

favets, 2023-01-29:
### Describe the bug

No blackbox detected with Mamba FC and BF 4.4.0  --> DIAT/MAMBAF722(STM32F7X2) <--

### To Reproduce

flash from 4.3.2 to 4.4.0 erasing chip
[image]

### Expected behavior

found an usable blackbox

### Configurator version

10.9.0 (697943b)

favets, 2023-01-29:
with Core Only it work. Sorry thanks

KAJE1977, 2023-01-30:
I flashed core only and gps stopped working. same board. had flash chip not detected first and gps working.

haslinghuis, 2023-01-30:
See https://github.com/betaflight/betaflight/issues/12255#issuecomment-1407742684

haslinghuis, 2023-01-30:
Or put `FLASH_W25Q128FV` in custom defines before flashing.

KAJE1977, 2023-01-30:
It works perfectly. Thank you so much!!!!

closed 2023-01-30 by blckmn