betaflight/betaflight #12265: Issue Motor: Flywoo Explorer LR4 HD (but with HEX FC) Betaflight 4.4

mmedinasantacruz, 2023-01-30:
### Describe the bug

Hello! 
I have updated the Betaflight from version 4.2.11 to the 4.4 version of my Quad BF Flywoo Explorer LR4 which comes with the "F745 Hex" flight controller.
At the end of the installation I go to the motor section and when trying to do a test it only turns on 3 of 4 of them.
I actually compared the settings between them and I really don't find much difference.
The motor that won't start is number 3.
Your help would be greatly appreciated, I have "broken my head" trying to solve it but as you can see, without success.

Thanks

### To Reproduce

I go to the motor section and when trying to do a test it only turns on 3 of 4 of them.

[image]

### Expected behavior

All four motors start.

### Support ID

```txt
512828f8-bc7e-4854-a09a-67ab949cbc18
```

### Diff configuration

```txt
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff all

# version
# Betaflight / STM32F745 (S745) 4.4.0 Jan 30 2023 / 13:33:39 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name FLYWOOF745
manufacturer_id FLWO
signature 

# master
set dyn_notch_count = 1
set dyn_notch_q = 500
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set osd_canvas_width = 30
set osd_canvas_height = 13
set gyro_1_align_yaw = 2700

profile 0

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

### Resources configuration

```txt
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: LED 1
A03: FREE
A04: FLASH_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: PINIO 1
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: LED_STRIP
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH1 : LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH2 : DSHOT_BITBANG 2
    CH4 : DSHOT_BITBANG 5
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: SPI_MISO 4
DMA2 Stream 1: SPI_MOSI 4
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 5
```

### Flight controller

GOKU HEX F745 FC (16*16)

### Other components

ELRS, DJI Vista, GPS

### Add any other context about the problem that you think might be relevant here

**Version 4.2.11 who WORKS**

# version
# Betaflight / STM32F745 (S745) 4.2.11 Nov  9 2021 / 20:29:04 (948ba6339) MSP API: 1.43
# config: manufacturer_id: FLWO, board_name: FLYWOOF745, version: c35e171e, date: 2023-01-03T14:15:03Z

# start the command batch
batch start

board_name FLYWOOF745
manufacturer_id FLWO
mcu_id 003c00433538511039303632
signature 

# name: MAD FlyLR4

# resources
resource BEEPER 1 D15
resource MOTOR 1 B01
resource MOTOR 2 C09
resource MOTOR 3 B00
resource MOTOR 4 E11
resource MOTOR 5 E09
resource MOTOR 6 A03
resource MOTOR 7 B04
resource MOTOR 8 B05
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 E13
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 D12
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 E08
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 D06
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 E07
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 B06
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 B07
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 A02
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource TRANSPONDER 1 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 NONE
resource SPI_SCK 4 E02
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 NONE
resource SPI_MISO 4 E05
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 NONE
resource SPI_MOSI 4 E06
resource CAMERA_CONTROL 1 NONE
resource ADC_BATT 1 C03
resource ADC_RSSI 1 C05
resource ADC_CURR 1 C02
resource ADC_EXT 1 NONE
resource BARO_CS 1 NONE
resource BARO_EOC 1 NONE
resource BARO_XCLR 1 NONE
resource COMPASS_CS 1 NONE
resource COMPASS_EXTI 1 NONE
resource SDCARD_CS 1 NONE
resource SDCARD_DETECT 1 NONE
resource PINIO 1 C00
resource PINIO 2 NONE
resource PINIO 3 NONE
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 A04
resource OSD_CS 1 B12
resource RX_SPI_CS 1 NONE
resource RX_SPI_EXTI 1 NONE
resource RX_SPI_BIND 1 NONE
resource RX_SPI_LED 1 NONE
resource RX_SPI_CC2500_TX_EN 1 NONE
resource RX_SPI_CC2500_LNA_EN 1 NONE
resource RX_SPI_CC2500_ANT_SEL 1 NONE
resource GYRO_EXTI 1 E01
resource GYRO_EXTI 2 NONE
resource GYRO_CS 1 E04
resource GYRO_CS 2 NONE
resource USB_DETECT 1 A08
resource VTX_POWER 1 NONE
resource VTX_CS 1 NONE
resource VTX_DATA 1 NONE
resource VTX_CLK 1 NONE
resource PULLUP 1 NONE
resource PULLUP 2 NONE
resource PULLUP 3 NONE
resource PULLUP 4 NONE
resource PULLDOWN 1 NONE
resource PULLDOWN 2 NONE
resource PULLDOWN 3 NONE
resource PULLDOWN 4 NONE

# timer
timer E13 AF1
# pin E13: TIM1 CH3 (AF1)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer E09 AF1
# pin E09: TIM1 CH1 (AF1)
timer E11 AF1
# pin E11: TIM1 CH2 (AF1)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer A03 AF2
# pin A03: TIM5 CH4 (AF2)
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)
timer B04 AF2
# pin B04: TIM3 CH1 (AF2)
timer B05 AF2
# pin B05: TIM3 CH2 (AF2)
timer D12 AF2
# pin D12: TIM4 CH1 (AF2)

# dma
dma SPI_TX 1 NONE
dma SPI_TX 2 NONE
dma SPI_TX 3 NONE
dma SPI_TX 4 NONE
dma SPI_RX 1 NONE
dma SPI_RX 2 NONE
dma SPI_RX 3 NONE
dma SPI_RX 4 NONE
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
dma ADC 2 NONE
dma ADC 3 NONE
dma UART_TX 1 NONE
dma UART_TX 2 NONE
dma UART_TX 3 NONE
dma UART_TX 4 NONE
dma UART_TX 5 NONE
dma UART_TX 6 NONE
dma UART_TX 7 NONE
dma UART_TX 8 NONE
dma UART_RX 1 NONE
dma UART_RX 2 NONE
dma UART_RX 3 NONE
dma UART_RX 4 NONE
dma UART_RX 5 NONE
dma UART_RX 6 NONE
dma UART_RX 7 NONE
dma UART_RX 8 NONE
dma pin E13 1
# pin E13: DMA2 Stream 6 Channel 6
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin E09 2
# pin E09: DMA2 Stream 3 Channel 6
dma pin E11 1
# pin E11: DMA2 Stream 2 Channel 6
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin A03 0
# pin A03: DMA1 Stream 1 Channel 6
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3
dma pin B04 0
# pin B04: DMA1 Stream 4 Channel 5
dma pin B05 0
# pin B05: DMA1 Stream 5 Channel 5
dma pin D12 0
# pin D12: DMA1 Stream 0 Channel 2

# mixer
mixer QUADX

mmix reset

mmix 0  1.000 -0.809  0.659 -1.000
mmix 1  1.000 -1.000 -0.659  1.000
mmix 2  1.000  0.809  0.659  1.000
mmix 3  1.000  1.000 -0.659 -1.000

mmedinasantacruz, 2023-02-06:
For your information and anyone who may be interested: the friends of Flywoo recommend the following mapping of the Motors. With that setup, the drone works properly :). Thank you very much friends of Flywoo and Betaflight.

resource MOTOR 1 B01
resource MOTOR 2 C09
resource MOTOR 3 B00
resource MOTOR 4 E11
resource MOTOR 5 E09
resource MOTOR 6 A03
resource MOTOR 7 B04
resource MOTOR 8 B05
save

closed 2023-02-06 by mmedinasantacruz

woefe1985, 2023-02-08:
Hey, if i change to the motor mapping:
resource MOTOR 1 B01
resource MOTOR 2 C09
resource MOTOR 3 B00
resource MOTOR 4 E11
resource MOTOR 5 E09
resource MOTOR 6 A03
resource MOTOR 7 B04
resource MOTOR 8 B05

i get an error message: "you need 4 motor resources only 3 are avaleble with this firmware.
do you have any idea? could you please poast a complete dump backup??
Thanks a lot!!!

mmedinasantacruz, 2023-02-08:
> 

Hello, yes here it's :) , hope that helps,
Have a nice day.
[BTFL_cli_FLR4_dump.txt](https://github.com/betaflight/betaflight/files/10688995/BTFL_cli_FLR4_dump.txt)

woefe1985, 2023-02-08:
wow your are fast,

do you use the hex version ore the quad? in your backup you have an F745 FC this i think is for the hexacopter correct?
The quad version uses the F745 Nano or am i wrong?
Best regards Wolfgang

mmedinasantacruz, 2023-02-08:
> 

Hello, yes, my Quad has the FC Hex version and the firmware version available for it is the FLYWOOF745. I've been using that version since version 4.2.11.
Best regards :)

woefe1985, 2023-02-09:
Hey okay in my case it doesn’t work i have FLYWOOF745Nano for 4.4. With 4.2.11 your solution works. I don’t know maybe they changed something in the firmware. But anyway thanks a lot for supporting!!!

mmedinasantacruz, 2023-02-09:
> Hey okay in my case it doesn’t work i have FLYWOOF745Nano. With 4.2.11 your solution works. I don’t know maybe they changed something in the firmware. But anyway thanks a lot for supporting!!!

Hi, it's sad that it didn't work for you. :( Have a happy day.

woefe1985, 2023-02-09:
hey, okay now it works with your mapping and this additional setting:
set dshot_bitbang = OFF
save
regards Wolfgang

haslinghuis, 2023-02-09:
@mmedinasantacruz Have updated FLYWOOF745 target with motor resources and bitbang setting.

We have no info regarding the FLYWOOF745NANO.

woefe1985, 2023-02-12:
Hey, maybe it’s my fault. With the firmware of the nano, I cannot use OSD standard or WTFOS. If I switch to F745 (not nano) Firmware. OSD and WTFOS hack working fine. So now I am confused. In the manual there is described that in my Explorer is a nano implemented. But I get the system only running with F745 FW.

mmedinasantacruz, 2023-02-14:
> Hey, maybe it’s my fault. With the firmware of the nano, I cannot use OSD standard or WTFOS. If I switch to F745 (not nano) Firmware. OSD and WTFOS hack working fine. So now I am confused. In the [...]

Hi, I'm not sure if this can help since in my case I haven't had any problems, but use the WTFOS presets in Betaflight Configurator, that might help you. Best regards :)

santicolomo, 2023-08-18:
> Hey, maybe it’s my fault. With the firmware of the nano, I cannot use OSD standard or WTFOS. If I switch to F745 (not nano) Firmware. OSD and WTFOS hack working fine. So now I am confused. In the [...]

Hey @woefe1985 . I'm having exactly the same problems than you. Motor 4 not working and WTFOS not working either. 
I've got y Explorer set with the FLYWOOF745NANO. 
Are you saying that if I use the target FLYWOOF745 instead of the nano when updating the betaflight firmware it should work? ... both motor problem and the OSD?

woefe1985, 2023-08-18:
Hey,

open the drone and take a picture of the FC (Not ESC!!) then google FLYWOO F745 and FLYWOO F745NANO. With the white labels I could confirm that in my drone there is an FLYWOO F745 installed and not a NANO. This sh***** took me ours. IF you then know which you have you can be sure to install the correct firmware. For OSD use the WTF hack. It works fine ==>https://fpv.wtf/. Good luck my friend this issue took me 3 full days to solve. 
If you need help you can ask any time!

woefe1985, 2023-08-18:
i forgot this is the current motor mapping which works fine:
# resources
resource BEEPER 1 D15
resource MOTOR 1 B01
resource MOTOR 2 C09
resource MOTOR 3 B00
resource MOTOR 4 E11
resource MOTOR 5 E09
resource MOTOR 6 A03
resource MOTOR 7 B04
resource MOTOR 8 B05

santicolomo, 2023-08-19:
Hey @woefe1985 , I certainly had the same problem, my FC is tge FLYWOO F745, not the NANO... and yes, I wasted hours trying to short this out... now everything worked perfectily fine after flashing the FC using the new target.

Thanks for your help!!