betaflight/betaflight #12266: Missing barometer in target GEPRCF722BT

ilcato, 2023-01-30:
### Describe the bug

This target use a BMP280 barometer. 
After flashing betaflight 4.4 it is no more possible to activate the barometer switch in configuration tab of Betaflight configurator.

### To Reproduce

Flash betaflight 4.4 on a GEPRCF722BT flight controller,
Select barometer switch in betaflight configurator tab.
After saving and rebooting the switch is reset to false..

### Expected behavior

To be able to enable the barometer switch.

### Support ID

```txt
Id: "75f72ff6-5ab6-4c6f-b636-f00f36696c56"
```

### Diff configuration

```txt
diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 30 2023 / 21:33:35 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name GEPRCF722BT
manufacturer_id GEPR

# name: Mark 5

# feature
feature -TELEMETRY
feature GPS

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# led
led 0 5,7::A:0
led 1 6,7::A:0
led 2 7,7::A:0
led 3 8,7::A:0
led 4 9,7::A:0
led 5 10,7::A:0

# mode_color
mode_color 6 1 2

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 46 4 1700 2100 0 0
aux 4 13 2 1700 2100 0 0
aux 5 35 3 1700 2100 0 0

# rxfail
rxfail 3 s 1500
rxfail 5 s 1000

# master
set gyro_lpf1_static_hz = 0
set gyro_lpf2_type = BIQUAD
set gyro_lpf2_static_hz = 300
set dyn_notch_q = 700
set gyro_lpf1_dyn_min_hz = 0
set gyro_lpf1_dyn_max_hz = 300
set acc_calibration = -25,35,-260,1
set align_mag = CW90FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 900
set mag_calibration = -747,-369,-132
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 40
set failsafe_procedure = GPS-RESCUE
set beeper_dshot_beacon_tone = 3
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_rescue_min_start_dist = 20
set gps_rescue_alt_mode = FIXED_ALT
set gps_rescue_ground_speed = 1000
set gps_rescue_pitch_angle_max = 32
set gps_rescue_throttle_max = 1800
set gps_rescue_throttle_hover = 1500
set gps_rescue_sanity_checks = RESCUE_SANITY_OFF
set gps_rescue_min_sats = 6
set simplified_gyro_filter_multiplier = 60
set osd_vbat_pos = 14570
set osd_link_quality_pos = 101
set osd_rssi_dbm_pos = 133
set osd_craft_name_pos = 14816
set osd_gps_speed_pos = 14394
set osd_gps_lon_pos = 128
set osd_gps_lat_pos = 8288
set osd_gps_sats_pos = 14362
set osd_home_dir_pos = 14352
set osd_home_dist_pos = 14400
set osd_flight_dist_pos = 14336
set osd_altitude_pos = 14368
set osd_warnings_pos = 14763
set osd_avg_cell_voltage_pos = 14592
set osd_canvas_width = 30
set osd_canvas_height = 16
set debug_mode = GYRO_SCALED
set vcd_video_system = PAL
set gyro_1_sensor_align = CW180
set gyro_1_align_yaw = 1800
set rpm_filter_harmonics = 2
set craft_name = Mark 5
set pilot_name = ilcato

profile 0

# profile 0
set dterm_lpf1_dyn_min_hz = 63
set dterm_lpf1_dyn_max_hz = 127
set dterm_lpf1_dyn_expo = 3
set dterm_lpf1_static_hz = 63
set dterm_lpf2_static_hz = 127
set p_pitch = 65
set i_pitch = 135
set d_pitch = 49
set f_pitch = 341
set p_roll = 63
set i_roll = 128
set d_roll = 44
set f_roll = 327
set p_yaw = 63
set i_yaw = 128
set f_yaw = 327
set d_min_roll = 44
set d_min_pitch = 49
set motor_output_limit = 90
set feedforward_smooth_factor = 30
set feedforward_jitter_factor = 9
set dyn_idle_min_rpm = 20
set simplified_master_multiplier = 140
set simplified_i_gain = 115
set simplified_d_gain = 105
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 195
set simplified_dterm_filter_multiplier = 85

rateprofile 1

# rateprofile 1
set thr_mid = 40
set thr_expo = 50
set roll_rc_rate = 10
set pitch_rc_rate = 10
set yaw_rc_rate = 10
set roll_expo = 90
set pitch_expo = 90
set yaw_expo = 50
set roll_srate = 90
set pitch_srate = 90
set yaw_srate = 72
set throttle_limit_type = SCALE
set throttle_limit_percent = 85

# end the command batch
batch end

#
```

### Resources configuration

```txt
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 2
A09: MOTOR 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: LED_STRIP
B04: FREE
B05: SPI_MOSI 3
B06: SERIAL_TX 1
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: MOTOR 4
C09: MOTOR 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2 : LED_STRIP
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 1
    CH4 : DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: LED_STRIP
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 3

#
```

### Flight controller

SPAN F722BT on version 1 mark 5

haslinghuis, 2023-01-30:
Please flash with `BARO_BMP280` in custom defines field and report back.

closed 2023-01-30 by blckmn

ilcato, 2023-01-31:
@haslinghuis @blckmn the change in the unified-target profile is wrong: it must be USE_BARO_BMP80 and not USE_BMP80.
Tested with my mark and it works. Thanks.