betaflight/betaflight #12267: Failsafe Stage 2 Landing mode: replace timer with newer landing detection

ggoraa, 2023-01-31:
### Is your feature request related to a problem? Please describe

In BF 4.4, I see that you guys vastly improved GPS Return to Home (btw why is it still called GPS Rescue in the Configurator even though you renamed it to Return to Home in the release notes?), like how instead of just going to the pilot it would do a whole mission to return home(climb, start going home, descend when near, stop when close, descend until a landing was detected), but you have left out the normal "Landing" mode, bc it still has that "timer" landing method instead of the newer "landing detection" method used in the new GPS Return to Home. And that can result in situations like this:
- If the timer is too small, it doesn't have time to descent properly, making it fall out of the sky anyway, just from a smaller height
- If the timer is too large, it doesn't stop motors when it lands for some time, which can be a real pain in the ass when it lands in tall grass or even a tree
- 

### Describe the solution you'd like

What about bringing that same landing detection feature to a "dumb" Stage 2 Landing mode("dumb" because it just lands when GPS Return to Home literately brings your drone back to you)? That will fix both issues, the drone will not fall out of the sky, and still won't nuke itself when it senses that it touched something (basically landed)! 

### Describe alternatives you've considered

Just use GPS Return to Home, but for some people it may not be an option (maybe they don't have GPS on their quad or smth)

ctzsnooze, 2023-08-14:
This is something we I would like to do, but it takes quite a lot of work to do properly. 

In descending order of functionality:

- position hold with a controlled descent rate and automatic disarm on impact; requires GPS, Mag, and would be best with baro as well
- a modified GPS rescue where the nose points to home, the quad maintains distance to home and use GPS+/-Baro to control the descent, with automatic disarm on impact.  The quad could drift sideways due to crosswind or acc cal not being perfect.
- controlled vertical descent velocity and with impact detection, but no attempt to maintain position.  This is very possible with Baro only or GPS only  Potentially there could be a lot of drift during descent.  Sanity check would time out if descent stopped.
- Fixed throttle 'landing mode' for an indefinite period with impact detection.  This is hazardous because we don't have any way to know if we are achieving any particular descent velocity, and if the impact detection fails, it will never disarm.

Any decent controlled descent solution really needs a GPS, and in that case, using GPS rescue is probably the best option.

ggoraa, 2023-08-24:
> This is hazardous because we don't have any way to know if we are achieving any particular descent velocity, and if the impact detection fails, it will never disarm.

What about a combination of the two? I mean primarily detecting landing using landing detection, but also use a timer, for cases if landing detection fails

closed 2026-09-26 by blckmn