betaflight/betaflight #12271: No Gyro BF4.4

siamskoi, 2023-01-31:
### Describe the bug

After updating firmware FC GEPRCF722_BT_HD to 4.4.0 version I got NO_GYRO error. If restore firmware 4.3.2 - all good working.
 

### To Reproduce

Update firmware to 4.4.0

### Expected behavior

No error about gyro.

### Support ID

```txt
version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 30 2023 / 18:20:45 (4605309d8) MSP API: 1.45

# config: YES
# board: manufacturer_id: GEPR, board_name: GEPRCF722_BT_HD

# status
MCU F722 Clock=216MHz, Vref=3.28V, Core temp=56degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CUSTOM DEFAULTS, size: 3627, max available: 16384
Devices detected: SPI:0, I2C:0
Gyros detected: locked shared

OSD: MAX7456 (30 x 13)
BUILD KEY: 8fba42d9e3c583daf21384a175c4dc09 (4.4.0)
System Uptime: 11 seconds, Current Time: 2023-01-31T16:21:08.337+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 0, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
Arming disable flags: NOGYRO BOOTGRACE CLI MSP

# flash_info
Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=0, totalSize=0 JEDEC ID=0x00000000
Partitions:
FlashFS size=0, usedSize=0

# dump hardware

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 30 2023 / 18:20:45 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name GEPRCF722_BT_HD
manufacturer_id GEPR

# resources
resource BEEPER 1 C15
resource MOTOR 1 C08
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C07
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 B01
resource MOTOR 8 B00
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 C10
[... 433 lines cut ...]
# Betaflight / STM32F7X2 (S7X2) 4.3.2 Nov 28 2022 / 07:30:19 (60c9521) MSP API: 1.44

# config: YES

# start the command batch
batch start

board_name GEPRCF722_BT_HD
manufacturer_id GEPR

# feature
feature -RX_PARALLEL_PWM

# master
set gyro_to_use = FIRST

profile 0

rateprofile 0

# end the command batch
batch end

# 
```

haslinghuis, 2023-01-31:
Please flash Core Only build and provide the support ID

siamskoi, 2023-01-31:
Wow. If flash just core - gyro is working! Wtf?! 

Core only support id:
```
version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 29 2023 / 02:32:03 (4605309d8) MSP API: 1.45

# config: YES
# board: manufacturer_id: GEPR, board_name: GEPRCF722_BT_HD

# status
MCU F722 Clock=216MHz, Vref=3.28V, Core temp=57degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3791, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma shared
GYRO=ICM42688P, ACC=ICM42688P
OSD: MAX7456 (30 x 13)
BUILD KEY: d171e91b63de268fe5acf30bc7664f6c (4.4.0)
System Uptime: 52 seconds, Current Time: 2023-01-31T16:49:26.335+00:00
CPU:50%, cycle time: 124, GYRO rate: 8064, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 22
SD card: Manufacturer 0x0, 491520kB, 07/2021, v0.0, 'APPSD'
Filesystem: Ready
Arming disable flags: RXLOSS CLI MSP RPMFILTER NO_ACC_CAL

# flash_info
Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=0, totalSize=0 JEDEC ID=0x00000000
Partitions:
FlashFS size=0, usedSize=0

# dump hardware

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 29 2023 / 02:32:03 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name GEPRCF722_BT_HD
manufacturer_id GEPR

# resources
resource BEEPER 1 C15
resource MOTOR 1 C08
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C07
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 B01
resource MOTOR 8 B00
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
[... 266 lines cut ...]
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

# 
```

haslinghuis, 2023-01-31:
https://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines

siamskoi, 2023-02-01:
For gyro in this FC used ICM42688P - I found this info in official site.
I think, need replace in config to something like this:
```
#define USE_GYRO_SPI_ICM42605
#define USE_GYRO_SPI_ICM42688P
#define USE_ACC_SPI_ICM42605
#define USE_ACC_SPI_ICM42688P
```

nightflasher, 2023-02-01:
my JHEMCUF7DUAL also boots with NO_GYRO sometimes, but i only had this on USB power only.

Kpiiica4580, 2023-02-04:
I have same problem on my GEP-F722 HD Flight Controller
this is a stats, when flashing core only:
# 
# Building AutoComplete Cache ... Done!
# 
# status
MCU F722 Clock=216MHz, Vref=3.27V, Core temp=49degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3791, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma shared
GYRO=ICM42688P, ACC=ICM42688P
OSD: MAX7456 (30 x 13)
BUILD KEY: d171e91b63de268fe5acf30bc7664f6c (4.4.0)
System Uptime: 16 seconds, Current Time: 2023-02-04T14:31:31.720+00:00
CPU:49%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 22
SD card: Manufacturer 0x0, 491520kB, 12/2021, v0.0, 'APPSD'
Filesystem: Ready
Arming disable flags: RXLOSS CLI MSP RPMFILTER NO_ACC_CAL

closed 2023-02-08 by KarateBrot