betaflight/betaflight #12274: Mamba F722_i2C GPS not recognised in 4.4.0

riddles55, 2023-01-31:
### Describe the bug

GPS not recognised in 4.4.0, but is recognised in 4,4 RC3 with a Mamba F722_i2C (Mamba F722 mk3 fc)

### To Reproduce

Just flash with the standard 4,4,0 official release for this FC, applied standard defaults and set the ports correctly and GPS baud 115200

### Expected behavior

GPS to show in the list of sensors at the top menu of betaflight

### Diff configuration

```txt
# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 16 2023 / 13:12:30 (2c34b3949) MSP API: 1.45

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAF722_I2C
manufacturer_id DIAT
mcu_id 004300453331510c33323639
signature 

# feature
feature -TELEMETRY
feature GPS

# serial
serial 1 2 115200 115200 0 115200
serial 3 0 115200 57600 0 115200
serial 4 131073 115200 57600 0 115200

# master
set gyro_1_align_yaw = 1800

profile 0

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

### Resources configuration

```txt
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 1
C09: MOTOR 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3 : DSHOT_BITBANG 3
    CH4 : DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: DSHOT_BITBANG 3
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: DSHOT_BITBANG 1

#
```

### Flight controller

Mamba F722 Basic MK3

fabi9163, 2023-02-01:
Which GPS do you have? Got the same problem with the Flywoo GM10 Pro v3.  The GPS-Icon only appears, if I disable auto-config (not flown yet). Maybe the autoconfig for the new m10-chip must be different than m8...

riddles55, 2023-02-01:
Sam m8q. It may be because I flashed it using "classic" on the firmware flasher page which I now hear from bardwell may exclude some functionality so perhaps the GPS stuff is excluded. Going to try again tonight using the cloud build option

riddles55, 2023-02-01:
This fixed this. Don't use "core" (or "classic" as it used to be called) mode when flashing

closed 2023-02-01 by riddles55