betaflight/betaflight #12275: Blackbox problem on OMNIBUSF4SD

ilpino8, 2023-01-31:
### Describe the bug

After the update to 4.4 the SD card used for the blackbox doesn't read anymore. Tried with two different FCs, one of which was new. Also tried with multiple SD cards and it still doesn't read to me. What can I do?

### To Reproduce

After the update to 4.4 the SD card used for the blackbox doesn't read anymore. Tried with two different FCs, one of which was new. Also tried with multiple SD cards and it still doesn't read to me. What can I do?

### Expected behavior

I was just expecting the SD card to be read and I could record the blackbox

### Support ID

```txt
3b20e889-cd49-4f39-9a68-5d657e6ed091
```

### Flight controller

F4 Hobbywing (OMNIBUSF4SD)

### Other components

I own walksnail and the ExpressRLS RX

### How are the different components wired up (including port information)

The only free port that the configurator tells me is 6. 1 and 3 are occupied by RX and Walksnail

haslinghuis, 2023-01-31:
Please try formatting with the tool mentioned in https://betaflight.com/docs/development/Blackbox#microsdhc

OMNIBUSF4SD:
```
SD card: Manufacturer 0xad, 61798400kB, 11/2020, v1.0, 'USD00'
Filesystem: Fatal - no FAT MBR partitions
```

MATEKH743:
```
SD card: Manufacturer 0x3, 31166976kB, 01/2007, v8.5, 'SD32G'
Filesystem: Ready
```

ilpino8, 2023-01-31:
Now it detects it for me, but every 2/3 times I restart / disconnect and reconnect the FC it always tells me an error. What could it be?

closed 2023-03-11 by github-actions[bot]