betaflight/betaflight #12279: Flash memory chip does not exist when any test version of BF4.4 firmware appears.

ACC-DBL, 2023-02-01:
### Describe the bug

When writing 4.4 test firmware from version 4.3.1, the flash memory chip can not be recognized. When I swipe back the 4.3.1 firmware, the flash memory chip will work properly.

### To Reproduce

None

### Expected behavior

None

### Support ID

```txt
# resources
resource BEEPER 1 C13
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 A15
resource MOTOR 6 B03
resource MOTOR 7 B06
resource MOTOR 8 B07
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A02
resource PWM 2 A01
resource PWM 3 A00
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 B08
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
[... 1061 lines cut ...]
set quickrates_rc_expo = OFF
set roll_rc_rate = 7
set pitch_rc_rate = 7
set yaw_rc_rate = 7
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# end the command batch
batch end
```

### Flight controller

MTKS/MATEKF722SE(STM32F7X2)

robert-b, 2023-02-01:
skystarsf77hdpro - the same - no flash chip

haslinghuis, 2023-02-01:
MATEKF722SE uses SDCARD

http://www.mateksys.com/?portfolio=f722-se

haslinghuis, 2023-02-01:
@robert-b please open your own issue report. See https://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines

ACC-DBL, 2023-02-02:
> MATEKF722SE使用SD卡
> 
> http://www.mateksys.com/?portfolio=f722-se

FC is HAKRC-F722 SE, and its FC firmware name and version is MTKS/MATEKF722SE (STM32F7X2), not MATEKSYS-F722 SE flight control.

haslinghuis, 2023-02-02:
This needs a new target as we can't pollute MATEK official targets with CLONE hardware variations.

ACC-DBL, 2023-02-02:
I don't know how to do it, but the FC firmware name and file officially given by HAKRC is only MTKS/MATEKF722SE (STM32F7X2), and I don't want to damage the official target file of matek.But there has been a problem since 4.4FC firmware was written.There has never been a problem with writing 4.3 firmware from 4.2 before.

haslinghuis, 2023-02-03:
We need some information here:

1. Flash your board with the Core Only switch enabled
2. Go to the CLI tab and click the Submit Support Data button.
3. With this generated support ID we have all required information to update board configuration files.

https://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines

Nanopilotz, 2023-02-17:
HAKRC F722 DM Flight Controller use target: MATEK722SE. Previously in BF4.2, the Blackbox can be enabled by this method:

resource FLASH_CS 1 D02
set flash_spi_bus = 3
set blackbox_device = SPIFLASH

Unfortunately, the "FLASH_CS 1" and "flash_spi_bus" are missing from BF4.4 Target: MATEKF722SE. 

I have checked the resource via Dump All, and it is confirmed missing although i have selected the Flash Storage during Cloud Build.

However, the FLASH_CS 1 is available in the Classic Mode Build Configuration, but after the BF4.4 update, suddenly GPS function is not working. All the Failsafe GPS parameters turn into zero "0".

Appreciate your help. Thanks.

closed 2023-03-28 by github-actions[bot]