betaflight/betaflight #12280: ADC VBAT values freeze once armed on GH743

howels, 2023-02-01:
### Describe the bug

ADC values for VBAT and current appear to freeze once the quad is armed, updating maybe once or twice in 120 seconds of flight.

This is visible on the OSD and via radio telemetry.

Worked normally in the December 5th build of 4.4, stopped working after that.

### To Reproduce

Arm quad, voltage values no longer update

### Expected behavior

ADC should be read whether armed or disarmed.

### Support ID

```txt
5b26d636-f75a-48da-91f8-7ab0c19ee191
```

### Flight controller

GH743

### Other components

DJI Vista with WTFOS (HD OSD)
eLRS RX 500Hz

### How are the different components wired up (including port information)

RX - UART2
VTX - UART6

### Add any other context about the problem that you think might be relevant here

Was working in 4.3, stopped after December 5th build.

haslinghuis, 2023-02-01:
@howels any particular PR to point at ?

https://github.com/betaflight/betaflight/commit/23602e6a895f489a3d5407849efd26657c7a8f5d

hydra, 2023-02-01:
Probably related to https://github.com/betaflight/betaflight/issues/11333

hydra, 2023-02-01:
@howels Did you try using git bisect to identify any commit when when it worked to when it didn't, or did you just test nightly builds between the two days?  the nightly builds usually differ by a few commits.

SteveCEvans, 2023-02-01:
Do other OSD elements update whilst the ADC reading is frozen?

hydra, 2023-02-02:
> Do other OSD elements update whilst the ADC reading is frozen?

I was wondering the same thing, I suspect they do, but perhaps there's an issue with the scheduling of the battery tasks (BATTERY_VOLTAGE/BATTERY_CURRENT/BATTERY_ALERTS/ADCINTERNAL).

hydra, 2023-02-02:
@howels please provide a complete cli 'dump' so that your config can be replicated.

The GPS elements in the OSD, enabled,inactive or otherwise, should certainly not prevent VBAT TLM output, so there is a bug somewhere for sure.

howels, 2023-02-02:
> @howels please provide a complete cli 'dump' so that your config can be replicated.
> 
> The GPS elements in the OSD, enabled,inactive or otherwise, should certainly not prevent VBAT TLM output, so there is a bug somewhere for sure.

I flashed a couple of new versions and after flashing the PR for the HD CMS scrolling it seemed to suddenly start working. I had master installed from a post 4.4 nightly already but suddenly it started working. Can't do much currently due to bad back pain.

hydra, 2023-02-03:
when things randomly start working due to an unrelated code change that usually signifies a bug - usually either a memory corruption issue or a side effect of new coding taking more or less time to run than previously.

@howels We need a complete cli dump from when it was NOT working to investigate the issue as we need to re-create your runtime environment.

hydra, 2023-02-06:
A few comments on discord suspect a cache issue, If a cache-coherency issue is suspected, then check the cache invalidation code around the ADC buffer and test with CPU cache disabled (comment out the cache enable line in the MCU startup code and re-test).

However, a blackbox log will tell us more.

Please attach us a cli dump and blackbox logs as requested.

howels, 2023-02-06:
CLI dump from 4.4 with the issue present attached. Getting BB logs now.

[AOS35v2-44_ADC_issue_BTFL_cli_AOS-35v2_20230206_124824.txt](https://github.com/betaflight/betaflight/files/10663159/AOS35v2-44_ADC_issue_BTFL_cli_AOS-35v2_20230206_124824.txt)

MacDaddyFPV, 2023-02-07:
I am seeing a similar issue. Voltage is frozen from plugin and incrementally changes every minute or so. I found that if I added acro trainer to the default options for cloud build that it then works. If helpful I can provide a dump from both working and not working.

howels, 2023-02-07:
> I am seeing a similar issue. Voltage is frozen from plugin and incrementally changes every minute or so. I found that if I added acro trainer to the default options for cloud build that it then works. If helpful I can provide a dump from both working and not working.

Dumps would be helpful, yes please.

hydra, 2023-02-07:
@MacDaddyFPV are you using the same hardware as @howels?

@klutvott123 Are you sure the `adcConversionBuffer` is in the dmaram_start/end region?  Did you check the corresponding .map file?

MacDaddyFPV, 2023-02-08:
> @MacDaddyFPV are you using the same hardware as @howels?
> 
> @klutvott123 Are you sure the `adcConversionBuffer` is in the dmaram_start/end region? Did you check the corresponding .map file?

Yes using the jhemcu h743 AIO. 
I was able to get it working by leaving the default options and adding the acro trainer option. If I do the default without acro trainer then voltage readings are frozen.

trxrich, 2023-02-13:
I've had the same problem with vbat and current not updating on the osd using 4.4.0 with a Mamba H743 using analog vtx. I flashed it back to 4.3.2 and all is working normally. Dump file attached.
[BTFL_cli_XL8_H743_20230129_105655_dump all.txt](https://github.com/betaflight/betaflight/files/10717749/BTFL_cli_XL8_H743_20230129_105655_dump.all.txt)

closed 2023-03-03 by blckmn