betaflight/betaflight #12283: Dyn notch hangs up FC

mituritsyn, 2023-02-01:
### Describe the bug

Applying next settings will hang FC forever so only reflashing will help:
set dyn_notch_min_hz = 120
set dyn_notch_max_hz = 500
probably there are other min max combinations that will lead to such behaviour

### To Reproduce

flash PR 12282 or later(reproduced on release 4.4)
apply cli:
set dyn_notch_min_hz = 120
set dyn_notch_max_hz = 500
or set those values manually

### Expected behavior

FC works well, connects to configurator, user can switch tabs or change settings
Actual:
Seems you can connect, but you can't change any settings, switching tab will lead to infinite tab loading

### Support ID

```txt
Id: "6c3382bc-ac68-4fa7-b8ee-bb67d8f9fe86"
```

### Flight controller

matek f405std, iflight blitz f722 mini

mituritsyn, 2023-02-01:
@KarateBrot can you plese take a look?

haslinghuis, 2023-02-01:
@KarateBrot Duplicate of #12254 - please see last comment(s)

closed 2023-02-09 by haslinghuis