betaflight/betaflight #12292: HGLRCF722 target, cannot connect after flashing

antonsn, 2023-02-03:
### Describe the bug

after flashing HGLRCF722 target to HGLRCF722  board cannot connect after flashing, FC is not booting up, long 2 beeps

### To Reproduce

Flash `core only` or any other configuration to HGLRC F722 board with HGLRCF722 target

### Expected behavior

after flashing I can connect to FC, FC seems to nott to be booted correctly

### Support ID

```txt
cannot access CLI
```

### Flight controller

HGLRC F722

haslinghuis, 2023-02-03:
Is there any particular reason flashing core only build?

Please try HGLRCF722E target if BARO is not working.

antonsn, 2023-02-14:
> Is there any particular reason flashing core only build?
> 
> Please try HGLRCF722E target if BARO is not working.

I tried different kinds of builds, including with baro.  I ll try HGLRCF722E this I did not tried

alienwaredream, 2023-03-12:
> > Is there any particular reason flashing core only build?
> > Please try HGLRCF722E target if BARO is not working.
> 
> I tried different kinds of builds, including with baro. I ll try HGLRCF722E this I did not tried

It works!

closed 2023-03-31 by antonsn