betaflight/betaflight #12296: FlywooF745Hex Says 6 motors not supported in 4.4

DivingFalconFPV, 2023-02-04:
### Describe the bug

Motor tab says 6 motors not supported in this firmware only 4. I tried setting up myself. I also tried diff and dump from previous 
Wondering if I need to enter a Hex command on the firmware compile before flashing. 
4.2.8 it came with. 
I go back to 4.2.8 it works. I am back on 4.2.8 
I believe 4.3 didnt work either. I couldnt find a fix on here. 
[BTFL_cli_FLYWOO_20230204_004737.txt](https://github.com/betaflight/betaflight/files/10607894/BTFL_cli_FLYWOO_20230204_004737.txt)

### To Reproduce

Just goto motor tab it will not support 6 motors. Wont spin any motors but resources are set. 

### Expected behavior

Has resources for all 6 motors. 

### Support ID

```txt
back on 4.2.8 though to make it work 
*** Support data submitted *** 
Id: "ebe04ffa-ab8a-49c9-b829-9e3dd83ec73d"

# copy ID and provide to the betaflight team.
```

### Flight controller

FlywooF745Hex

butcher51, 2023-02-10:
It should work, mine is working on 4.4. Are you sure that the mixer setup is correct?

haslinghuis, 2023-02-10:
Please see: https://github.com/betaflight/betaflight/issues/11205 for explanation

Duplicate of: https://github.com/betaflight/betaflight/issues/12265

Fixed in: https://github.com/betaflight/unified-targets/pull/891

closed 2023-02-10 by haslinghuis