betaflight/betaflight #12297: SPI speed on H7 for divider and clock calcs incorrect

David-OConnor, 2023-02-04:
### Describe the bug

https://github.com/betaflight/betaflight/blob/master/src/main/drivers/bus_spi.c#L353
https://github.com/betaflight/betaflight/blob/master/src/main/drivers/bus_spi.c#L378

Currently, in `spiCalculateDivider` and `spiCalculateClock` in `drivers/bus_spi`, SPI clock is hard-coded to 100Mhz.

We can see from [the startup code](https://github.com/betaflight/betaflight/blob/master/src/main/startup/system_stm32h7xx.c#L608) That all H7 variants are set to PLL1Q as SPI clock.

Looking through the H7 variants, the only one where this calculation comes out correct is on the H743RevY. I think the most common one is actually the Rev V one. 

What it actually is, at the current config from the startup file:

  -100Mhz for H743Y
  - 120Mhz for H743V
  - 70Mhz for H7A3
  - 275Mhz for H72x (This exceeds the hardware limit of 200Mhz SPI bus; addressed in my HSE-customization PR by increasing DIVQ)
  - 130Mhz for H73x

Bottom line: This can, in some cases, cause the SPI slave device to be oversped. I think the fix is to add methods the `pllConfig_t` struct for various clock speeds, werein it calculates from the scalers.

This is likely to not cause errors in practice unless those algorithms at the top of this post put the SPI clock over the devices limits. Its likely this hasn't come up because eg if the limit of a device is 24Mhz, the divider calc likely didn't set to a point where a 120% speed increase (eg from H743V) would cause a problem, since it can only set SPI speed in crude increments. So, could cause trouble in the future.

### To Reproduce

N/A

### Expected behavior

N/A

### Support ID

```txt
N/A
```

### Flight controller

N/A

SteveCEvans, 2023-03-15:
@David-OConnor Did you have a PR on the way for this?

David-OConnor, 2023-03-15:
No. I'm happy to help, but would need someone to talk me through the relevant sections of the codebase. Would probably be easiest for a BF dev who's familiar with the code base.

An approach I like for this sort of thing: Use methods (functions here?) on the clock-config struct that outputs speed for each peripheral. In the case of some MCUs, this may just be something that gets the `apb1` or `apb2` speed, depending on which SPI periph it is. For example:

```rust
/// Get the APB1 peripheral clock frequency frequency, in hz
pub fn apb1(&self) -> u32 {
        self.hclk() / self.apb1_prescaler.value() as u32
}
```
(Which code that determines which `apb` to use based on bus).

On H7, the idea is similar, but you would also take into account which SPI source is used, (PLL1Q, PLL2P etc), if that's configurable in BF. Or if it's hard-coded, use `get_pll1q()` speed as above.

SteveCEvans, 2023-03-15:
I’m happy to do the coding. You may have given enough info already. Let me review and come back with any questions.

klutvott123, 2023-09-05:
is this fixed in https://github.com/betaflight/betaflight/pull/12291 ?