betaflight/betaflight #12304: Upgrading Eachine F405 clone FC results in no OSD

CurryKitten, 2023-02-04:
### Describe the bug

I've got a few quads with this FC https://www.banggood.com/Eachine-Tyro129-Tyro119-Spare-Part-GPS-F4-Flight-Controller-AIO-OSD-BEC-for-RC-Drone-FPV-Racing-p-1580405.html  They were used in some of the Banggood Tyro quads, and are an Eachine FC which seems to be a clone of the Matek F405.

When supplied, they were running Betaflight 4.0.3, where the version command would give -

`# Betaflight / MATEKF405 (MKF4) 4.0.3 Jun  1 2019 / 11:36:38 (094cfc956) MSP API: 1.41`

I went ahead and attempted to upgrade this FC to 4.4.  With the unified targets, I wasn't completely sure if this should have been on the MATEKF405CTR or MATEKF405_CLONE.  I actually ended up trying both and they both came up with the same issue in which no OSD is shown, and when in the configurator OSD tab, the "Manage fonts" option is greyed out - leading me to think it can't detect the OSD chip.

Wondering if the OSD chip was damaged and didn't work, I grabbed a betaflight_4.0.6_MATEKF405.hex file and reflashed that, and the OSD was back again.  Back on 4.4, I tried setting osd_displayport_device to MAX7456 manually, but that didn't help.  I also flashed the core only build, but no difference there.

### To Reproduce

Upgrade the aforementioned Eachine F405 board to 4.4

### Expected behavior

Obviously, we'd expect the OSD to be there

### Support ID

```txt
Id: "dde872fa-459c-4ab6-9f7f-0a04e68af59e"
```

### Flight controller

Eachine F405 board as used in the Tyro119/129 (from Banggood)

### Other components

I've actually had everything disconnected from the board whilst upgrading (it doesn't want to go into DFU mode with things connected), but I'm running a R9mm converted to run ELRS 3.2.0 as an RX.  The VTX is a RushFPV Max Solo, it also has a Beitian BN-220 GPS 

### How are the different components wired up (including port information)

GPS in UART 1, Smart Audio on UART 3, RX on UART 4

howels, 2023-02-04:
Please post the full cli dumps from 4.0.3 and 4.4.0, thanks

CurryKitten, 2023-02-04:
Full dumps from Betaflight 4.4 and 4.0.3

[Tyro_129_44.txt](https://github.com/betaflight/betaflight/files/10609826/Tyro_129_44.txt)
[Tyro_129_403.txt](https://github.com/betaflight/betaflight/files/10609827/Tyro_129_403.txt)

haslinghuis, 2023-02-04:
403 is not a dump ???

CurryKitten, 2023-02-04:
> 403 is not a dump ???

Oops, my apologies.  I'd done a save via presets and a dump and got the files mixed up.  I've corrected the file upload now so the 403 is the proper dump

haslinghuis, 2023-02-04:
Please try
```
set max7456_clock = DEFAULT
save
```

CurryKitten, 2023-02-04:
> set osd_displayport_device = MAX7456
> set max7456_clock = DEFAULT

It didn't seem to like the clock being set to DEFAULT

```
# set max7456_clock = DEFAULT
###ERROR IN set: INVALID VALUE###
Allowed values: HALF, NOMINAL, DOUBLE
```

haslinghuis, 2023-02-04:
Yes update comment, as I was flashing to see the value had changed (but that's the first difference I'd spotted)

CurryKitten, 2023-02-04:
Ok, well this is embarrassing.  During the testing of those cli commands, I learnt that there is quite a difference in the OSD if we have the USB cable connected to the FC and then plug the battery in (no OSD chip detected) and when we plug the battery in and THEN the USB cable - in which case OSB chip is connected and we see the appropriate stuff on a VRX.  I was completely thrown off by the fact that in 4.0.3, I never needed to add power via a lipo to have the OSD chip get detected.

I went back and flashed the regular 4.4 version (rather than core only) and on it's defaults - as long as lipo is connected and then USB, it's all good.  So not actually a problem at all.  I'll close this off and pretend it never happened :)

closed 2023-02-04 by CurryKitten