betaflight/betaflight #12312: Blackbox device missing

TheIsotopes, 2023-02-05:
### Describe the bug

testing latest master on MatekF405TEmini and the flash chip was not found.

imo #12221 make the problem .... @haslinghuis pls can you check this

### To Reproduce

use latest firmware

### Expected behavior

working blackbox

### Support ID

```txt
*** Support data submitted *** 
Id: "ad8aade8-7dff-4a8d-a54c-2ded97a17d1d"
```

### Flight controller

MATEKF405TEMINI

haslinghuis, 2023-02-05:
Tested, for me it's found? Did you remove Flash from cloud build as the build key is missing from the status command?

> flash_info
> ###ERROR IN input: UNKNOWN COMMAND, TRY 'HELP'###

Test results:
```
ID 8075880d-3a3b-45c1-9164-e4a9f9dba33e

# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.29V, Core temp=41degC
Stack size: 2048, Stack address: 0x1000fff0
Configuration: CONFIGURED, size: 3239, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=ICM42605, ACC=ICM42605, BARO=DPS310
OSD: NONE (162 x 141)
BUILD KEY: 4ae0d85020dee98b6c4cec691f26e9e6 (4.5.0-zulu)
System Uptime: 13 seconds, Current Time: 2023-02-05T21:18:05.821+00:00
CPU:33%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 2 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4018 16M
Arming disable flags: RXLOSS CLI MSP MOTOR_PROTO
```

TheIsotopes, 2023-02-05:
using firmware hex from nightly asset.zip.

haslinghuis, 2023-02-05:
Confirm core only build does not detect the chip as normal build does.

TheIsotopes, 2023-02-05:
you right, only cloud build with default configuration is working

what is the problem?

haslinghuis, 2023-02-05:
Might be a problem with this define:

https://github.com/betaflight/betaflight/pull/12247/files#diff-f47f5a32f399bf68d833e35182cbc20004c8f3b97309d40f10c8f4956035eb63R317

TheIsotopes, 2023-02-05:
this is where the problem begins. https://github.com/betaflight/betaflight/actions/runs/4034319431

haslinghuis, 2023-02-06:
@TheIsotopes after some testing (and backtracking all changes in `#12221` to verify) - the change pointed out above prevents the detection with local build.

mikkokam, 2023-02-10:
I can confirm the same problem with BetaFPV F722 35A AIO Brushless Flight Controller V2.
V2 has the blackbox.

4.3.X – the integrated blackbox works.
4.4 – not visible.
4.5 – not visible.

Tried all – fad to downgrade to back to 4.3.2 – visible again.

closed 2023-02-10 by blckmn