betaflight/betaflight #12313: MATEKF722HD Baro DPS310 not recognized

simoccc, 2023-02-05:
### Describe the bug

MATEKF722HD board with DPS310 .

Betaflight 4.3 recognize the Baro without issue , Betaflight 4.4 latest official release does not.

BF 4.4 Core firmware detect Baro and it works.
Adding BARO_DPS310 or BARO_SPI_DPS310 at Custom defines doesn't resolve the issue , the status command does not recognise anything ( any Baro ) while with core it's ok.

I have submitted firmware core firmware data if needed  and Firmware not core with custom defines : BARO_DPS310 

Id: "158cc99a-793f-4138-8a3b-7159d13360f6"

This is the git fetch cloud build output :

git fetch origin 4.4.0 && git checkout -f FETCH_HEAD && make TARGET=STM32F7X2 EXTRA_FLAGS="-D'BUILD_KEY=96b855444897259d4e1bb11d83c59537' -D'RELEASE_NAME=4.4.0' -D'BOARD_NAME=MATEKF722HD' -D'MANUFACTURER_ID=MTKS' -DCLOUD_BUILD -DUSE_ACC -DUSE_ACC_SPI_MPU6000 -DUSE_BARO -DUSE_BARO_BMP280 -DUSE_DSHOT -DUSE_FLASH -DUSE_FLASH_W25Q128FV -DUSE_GPS -DUSE_GPS_PLUS_CODES -DUSE_GYRO -DUSE_GYRO_SPI_MPU6000 -DUSE_LED_STRIP -DUSE_OSD -DUSE_OSD_HD -DUSE_OSD_SD -DUSE_PINIO -DUSE_SERIALRX -DUSE_SERIALRX_CRSF -DUSE_TELEMETRY -DUSE_TELEMETRY_CRSF -DUSE_VTX"

### To Reproduce

install Firmware not core for MATEKF722HD target . Enable Barometer ( is enabled by default ) .

or

install Firmware core for MATEKF722HD target. Everything works

### Expected behavior

In custom defines adding BARO_DPS310 should work at least. 
 

### Support ID

```txt
b157e2ce-3392-4fcb-a1ac-6ccb9eed0414
```

### Flight controller

Matek F722HD

### Other components

Walksnail VTX and Matel 5883 GPS

simoccc, 2023-02-05:
Guys you are super fast ! now it's working and after a cloud build the firmware is ok , if I add Custom defines DPS310 . git fetch it's now with DUSE DPS310

Thank you very much

Lo90renzo, 2023-02-07:
Hey, I have the same issue! I tried to flash with custom defines DPS310 but still does not work.
Could you please help me?

haslinghuis, 2023-02-07:
Try `BARO_DPS310`

Do you have a genuine Matek board?

Lo90renzo, 2023-02-07:
I just tried with BARO_DPS310 and it worked! Btw it's a genuine matek I think 😁 thank you so much!

closed 2023-02-08 by haslinghuis