betaflight/betaflight #12318: Matek F722miniSE Baro not working

miralex773, 2023-02-06:
### Describe the bug

When using 4.4 Matek F722mini target on Matek F722miniSE board the baro support is absent, even if Baro is added to Custom Defines.
When flashing Core only baro is supported and operating normally.

### To Reproduce

Flash BF 4.4 Matek F722mini target to Matek F722miniSE board, add BARO_DPS310 to Custom Defines box.

### Expected behavior

Barometer DPS310 recognized and supported

### Support ID

```txt
*** Support data submitted *** 
Id: "0245dcb6-a12a-4cf9-947b-964a19c4be36"
```

### Flight controller

Matek F722miniSE

haslinghuis, 2023-02-06:
MATEKF722MINI has no baro. Please try MATEKF722SE

http://www.mateksys.com/?portfolio=f722-mini
http://www.mateksys.com/?portfolio=f722-minise

haslinghuis, 2023-02-06:
Please provide a link to the product page of the flight controller.

miralex773, 2023-02-06:
http://www.mateksys.com/?portfolio=f722-minise

However the current existing target MatekF722SE is for this controller:
http://www.mateksys.com/?portfolio=f722-se

closed 2023-02-06 by haslinghuis

miralex773, 2023-02-07:
All good, thank you!