betaflight/betaflight #12320: Barometer not recognized on CineApe25

Denethor85, 2023-02-06:
### Describe the bug

With BF 4.4 Cloud Building the barometer is not recognize on CineApe25. On Core Only it work

### To Reproduce

Flash 4.4 Cloud Building

### Expected behavior

The barometer working

### Support ID

```txt
7ec25c51-c75f-4ea3-bb87-0d93b4cb6de5
```

### Flight controller

Stock FC

Denethor85, 2023-02-07:
Meanwhile flashed with the following define?

BARO_BMP280

closed 2023-02-08 by haslinghuis

MR444, 2023-09-14:
Having the same issue. Any news after 7 months? Was is solved in a newer version and I just missed it?

haslinghuis, 2023-09-15:
Did you flash DARWINF411 ⁉️

MR444, 2023-09-15:
Yes

haslinghuis, 2023-09-15:
Please follow https://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines

jonathanoeijoeng, 2023-10-21:
I have same issue using 4.4.2. Already added BMP_280 in additional setting but barometer icon still not lit. 
Reading to this thread, actually I only need to use DARWINF411 and barometer should be already included, right?

Thanks,
Jonathan 

> Did you flash DARWINF411 ⁉️