betaflight/betaflight #12345: Gazebo SITL support

ahcorde, 2023-02-10:
### Is your feature request related to a problem? Please describe

I have seen two open PRs which are related to Gazebo and SITL simulation

https://github.com/betaflight/betaflight/pull/11303 and https://github.com/betaflight/betaflight/pull/11431

I'm interested in the Gazebo support but the PR is stuck in an old Betaflight version.

### Describe the solution you'd like

 I'm open to create a PR with a more update version for the Gazebo STIL support

I have some code working [here](https://github.com/osrf/vehicle_gateway/tree/main/betaflight_gazebo), but I would love to rely on the `main` branch instead of draft PRs

[image]

### Describe alternatives you've considered

Ping the other contributors @xuhao1 @nikhil-sethi

haslinghuis, 2023-02-10:
Please feel free to raise a PR for improving Gazebo SITL support.

See also: https://betaflight.com/docs/development

haslinghuis, 2023-02-20:
Fixed in https://github.com/betaflight/betaflight/pull/12346
Documentation update: https://github.com/betaflight/betaflight.com/pull/156

closed 2023-02-20 by haslinghuis