betaflight/betaflight #12347: GEPRC-F722 AIO not acting accurate

Niki0813, 2023-02-10:
### Describe the bug

After flashing with BF 4.4 the drone stop acting like it should. No issues with giro and Accel is online but when setup everything and arm the quad pops up and flip! Can’t make it fly normally! 
Also when plug in battery start up sound is playing twice!

### To Reproduce

Plug in battery, startup sound is playing twice, I can arm as usual but when push little any sticks very low quad pop up and flip. 

### Expected behavior

Can’t make it to hover or fly at all!

### Support ID

```txt
b4d7b705-3979-4d79-88d1-469d0f15c69a
```

### Flight controller

Geprc F722-AIO bought from Geprc online store

bramschats, 2023-02-13:
I have the GEPRCF722BT v1 stack and does also have problems. I did put on the stable 4.4 fw but gyro stopped working so i changed it back to 4.3.1. 

Previous i installed the beta 4.4 firmware and gyro was working but my buzzer, leds and OSD stopped working. For now i am not going to update to 4.4 and i'm not sure these problems are going to be solved in the future. I am also not really convinced on the quality on the GEPRC FC stack.

Niki0813, 2023-02-13:
Thanks for replaying, I think when  firmware 4.4 for specific board is listed it should be working fine -especially when signed "stable"! I recently updated Nazgul 5 v2 with F7 stack and everything is fine! I wish/hope 4.4 firm to work fine for my GPRC Vrocodile baby 5 long range to!
Hope so somebody to look for those problems or just remove firmware from BF server - because it is useless.

x4code, 2023-02-14:
damm, think im going to revert to 4.3.1 also.
sent email to GEPRC and they said to not use 4.4 for now, since is unstable with this drone (CineLog30)
will be watching this ticket. 
Thank you :)

mycosd, 2023-02-21:
I'm using F722 35A version in two of my Cinelog35 and I can't get them to work with BF 4.3.2 either (haven't tested it with 4.4 because I just want to get them to fly with the FW given by the manufacturer first). 
They fly normal until I just push the throttle quickly, then they roll over and crash. Both have the same hardware and are built/setup like a dozen other before, I've never seen this on a fresh build before. It's not like there is a power spike or something either, as the ESC don't restart, also the Blackbox continues til I disarm.

bramschats, 2023-03-11:
Any news on this topic?

Niki0813, 2023-03-12:
> Does anyone have a good working GEPRC FC running on BF4.4?

Not for now!

mbze430, 2023-04-17:
I have a GEPRC CineBot30 which has the F722 AIO.  But I am running 4.4.1, it seems to be a lot better.  But I am still getting accelerometer drift.  toward end of a flight with lots of stick movement in Angle mode the quad when center it will continue to drift it's last position

psionic001, 2023-06-12:
GEPRC F722-45A V2 AIO
Upgraded a Cinelog30 with 4.4 default and had a fly away. 
the LED strip is flickering, and it wobbles on takeoff, then full power until we disarm it and it drops to the ground. This is dangerous and must be resolved.

haslinghuis, 2023-06-12:
Please try 4.5.0 build.

haslinghuis, 2023-06-12:
> startup sound is playing twice

Can you provide ESC hardware & firmware version info and a blackbox log?

x4code, 2023-06-12:
i tried today with 4.4.2 and seems acceptable with default settings :)

HappyPrius, 2023-08-04:
I have a cinebot30 running BF 4.4.2 and the drone just flips over when I try to fly. I've been trying to configure it so that doesn't happen but now when I try to roll my drone to the right it pitches down, however the pitch works fine as well as Throttle, and Yaw. Anyone know how to solve the issue? 
P.S I've made sure that the channel mapping is correct currently setup as AETR1234

eXhausted, 2024-06-18:
I have the same problem on F722@Cinebot35 after update to 4.5.0. I tried not to apply any configs, but it still does **double sound** on startup. It doesn't react to arm anyhow, `status` returns only `CLI MSP`.

What does double sound mean?

closed 2026-09-26 by blckmn