betaflight/betaflight #12358: A way to show which antenna is used with a diversity rx, in real time in the OSD.

LoakAudio, 2023-02-12:
### Is your feature request related to a problem? Please describe

During a fly, I'm always asking myself if the rx's antennas are well positioned. There is a way to log that into the radio but it's hard to remember the position of the quad when looking at the logs after the fly.

### Describe the solution you'd like

As the frame rate used during a fly is fixed, the frame rate number in the LQ part of  the OSD could be a choice : rate or antenna used. Having it close to the link quality would give an easy read of the efficiency of the antennas.

### Describe alternatives you've considered

The "Antenna used" could have its own entry in the OSD, but it will take more place in the screen, I'm  not sure it could be the best way.

haslinghuis, 2023-02-12:
src/main/rx/expresslrs.c:409:        otaPkt.tlm_dl.ul_link_stats.uplink_RSSI_2 = 0; //diversity not supported

haslinghuis, 2023-02-12:
@JyeSmith @phobos- do you know if diversity is supported on ELRS ?

ZzyzxTek, 2023-02-12:
> As the frame rate used during a fly is fixed, the frame rate number in the LQ part of the OSD could be a choice : rate or antenna used. Having it close to the link quality would give an easy read of the efficiency of the antennas.

I would prefer not to subvert the frame rate in the LQ element.

The RSSI dBm reported is for the active antenna (pretty sure that is the case looking at the code). Add an option to append the active antenna number to the RSSI dBm element, for example have it display `-72 1` (similar to the way DOP is appended to the GPS number of satellites element).

Looking at the code, I think the active antenna number is passed down in telemetry from ELRS into BF, but I'll leave it up to the experts to confirm.

LoakAudio, 2023-02-12:
> The RSSI dBm reported is for the active antenna (pretty sure that is the case looking at the code). Add an option to append the active antenna number to the RSSI dBm element, for example have it display `-72 1` (similar to the way DOP is appended to the GPS number of satellites element).

Very good idea. Keeping it as an option ? SNR or SNR + antenna

JyeSmith, 2023-02-12:
> @JyeSmith @phobos- do you know if diversity is supported on ELRS ?

Antenna and True Diversity are both available in ELRS.  The active antenna is already passed to BF in the link stats and it sprobably a good choice to have it as a separate OSD element with the range 0-255.  While 0/1 will be the normal use case, its unit8 and useful for devs to bastardise the field for other things.

haslinghuis, 2023-02-13:
PR can be tested with development mode enabled and putting `#12359` in the commit field.

haslinghuis, 2023-02-13:
@JyeSmith

Don't think we even need a new element. If the active antenna is the second one we display (2) on the RSNR element.

LoakAudio, 2023-02-13:
> PR can be tested with development mode enabled and putting `#12359` in the commit field.

Is it ok to flash with BF configurator 10.9 (stable) ?
Does a diff all from 4.4 is ok or manual config ?

PR #12359 (2023-03-27): Update RSSI_DBM to include diversity antenna in OSD element (CRSF)
Fixes: #12358
closed 2023-03-27 by haslinghuis