betaflight/betaflight #12369: Enable OSD feature for OSD HD build option

mituritsyn, 2023-02-15:
### Describe the bug

Minor confusing issue:
With OSD SD build option OSD feature will be enabled by default, but with OSD HD only it won't be enabled

### To Reproduce

Flash via cloud build with OSD HD only

### Expected behavior

OSD feature enabled by default

### Support ID

```txt
osd hd: 8f29abeb-c891-4e7d-9116-65df20c80148
osd sd: 106e3f30-a857-4c7f-b64e-4a55ddea2f95
```

### Flight controller

matek f405std

haslinghuis, 2023-02-15:
```
diff 106e3f30-a857-4c7f-b64e-4a55ddea2f95 8f29abeb-c891-4e7d-9116-65df20c80148

< OSD: MAX7456 (30 x 13)
< BUILD KEY: dfdcf6b3305513b48622922f4088f684 (4.5.0-zulu)
---
> OSD: NONE (60 x 157)
> BUILD KEY: b4635dbba024181fe1baf0511fc003f7 (4.5.0-zulu)

OSD_HD is missing:

resource OSD_CS 1 B10
feature OSD
set max7456_clock = NOMINAL
set max7456_spi_bus = 2
set max7456_preinit_opu = OFF
```

@blckmn if `OSD_HD` is defined without `OSD_SD` should we `set osd_displayport_device = MSP` as default and keep enable `feature OSD` ?

SteveCEvans, 2023-03-20:
If the `MSP_SET_OSD_CANVAS` command is received and `USE_OSD_HD` is defined then `set osd_displayport_device = MSP` will be performed automatically. This requires the VTX to send this command.

mituritsyn, 2023-03-21:
> If the `MSP_SET_OSD_CANVAS` command is received and `USE_OSD_HD` is defined then `set osd_displayport_device = MSP` will be performed automatically. This requires the VTX to send this command.

How should VTX send something before been powered up? Common setting up scenario assumes battery will be plugged after all settings are done.

SteveCEvans, 2023-03-21:
The first time the VTX is powered up these settings are changed and then saved.

SteveCEvans, 2023-03-22:
Apologies. I missed the point of this issue. Please see https://github.com/betaflight/betaflight/pull/12563 for a fix.

closed 2023-03-22 by blckmn