betaflight/betaflight #12370: Latest master not working

TheIsotopes, 2023-02-15:
### Describe the bug

I'm tested now the latest master but i'm not happy. I checked back and found the culprit.

since #12355 the following problems have occurred
 - cpu load has increased by about 20%
 - led strip flashes confused
 - motors are not initialized
 - here any screenshots

[image]
[image]

 @blckmn pls can you check this

### To Reproduce

using latest master

### Expected behavior

working fc

### Support ID

```txt
*** Support data submitted *** 
Id: "e27ef9ea-1e52-40e8-ab38-1cf082393389"
```

### Flight controller

MATEKF722MINI

haslinghuis, 2023-02-15:
I suspect `#12349` - @TheIsotopes did you check that PR too or other PR's before that?

TheIsotopes, 2023-02-15:
@haslinghuis I'm just testing again.
It only seems to affect F722 and H743, on my F405 everything seems fine.

haslinghuis, 2023-02-15:
Curious about the timer assignment between last working version and failing version (and maybe also compare tasks output)

TheIsotopes, 2023-02-15:
Here i have the outputs of `resource show all` and `tasks` to compare.
with the resources so far everything is in order, only two new entrys from i2c on B10/B11.

here outputs from last working version:
```
# version
# Betaflight / STM32F7X2 (S7X2) 4.5.0 Feb 15 2023 / 20:32:52 (2dbbd1882) MSP API: 1.45

# config: YES
# board: manufacturer_id: MTKS, board_name: MATEKF722MINI

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED_STRIP
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
[... 245 lines cut ...]
DMA2 Stream 5: SPI_MOSI 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms   late    run reqd/us
00 - (         SYSTEM)     10      16       0    0.0%    0.0%         0      1    233      14
01 - (         SYSTEM)    974      19       0    1.8%    0.0%        14     60  22642      14
02 - (           GYRO)   8013      19       4   15.2%    3.2%       703      0 186335       0
03 - (         FILTER)   8013      32      19   25.6%   15.2%      3550      0 186335       0
04 - (            PID)   8013      90      61   72.1%   49.1%     11111      0 186335       0
07 - (             RX)     10      32      14    0.0%    0.0%         9      8    699      29
08 - (         SERIAL)     99    2787       2   27.5%    0.0%        11      0   2268      17
09 - (       DISPATCH)    958      19       0    1.8%    0.0%        17     35  22653      16
10 - (BATTERY_VOLTAGE)     50      19       2    0.0%    0.0%         2      1   1158      17
11 - (BATTERY_CURRENT)     50      19       2    0.0%    0.0%         2      2   1158      17
12 - ( BATTERY_ALERTS)      5       4       2    0.0%    0.0%         0      0    117       3
13 - (         BEEPER)     99      18       1    0.1%    0.0%         2      5   2268      15
18 - (      TELEMETRY)    247      39       0    0.9%    0.0%        10     28   5596      10
19 - (       LEDSTRIP)     83      53       5    0.4%    0.0%        13     38   2041      42
21 - (        VTXCTRL)      5       2       0    0.0%    0.0%         0      0    117       1
23 - (    ADCINTERNAL)      1       5       3    0.0%    0.0%         0      0     24       4
RX Check Function                  18       1                         1
Total (excluding SERIAL)                                67.5%
```

haslinghuis, 2023-02-15:
Nothing is missing. Added I2C2 resources and higher FILTER and PID utilization. Not sure why this should affect RPMFILTER.

@blckmn please check.

blckmn, 2023-02-15:
PR coming up. I think I have found the issue.

blckmn, 2023-02-15:
https://github.com/betaflight/betaflight/pull/12373

closed 2023-02-16 by blckmn