betaflight/betaflight #12374: GEPR/GEPRCF722_BT_HD(STM32F7X2) - GPS and Black box - either one or the other works

MarcinKnD, 2023-02-16:
### Describe the bug

with flash as core only in ports tab I see 5 UART, BlackBox works but GPS does not work (it does not remember settings in the configurator).

with flash as cloud build in ports I see UART6 GPS works but BlackBox does not work (adding the FLASH_W25M512 command does not help)

On 4.3.2 everything working fine and I see 6 UART's

### To Reproduce

flash with core only - black box working, GPS not working - in ports tab seeing 5 UART's

[image]
[image]
[image]
[image]

flash with cloud build (GPS,BalckBOX selected) - gps working, blackbox not working - in ports tab seeing 6 UART's

[image]
[image]
[image]
[image]

### Expected behavior

Working GPS and BlackBox as is on 4.3.2

### Support ID

```txt
# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Feb  5 2023 / 01:51:07 (4605309d8) MSP API: 1.45

# config: YES
# board: manufacturer_id: GEPR, board_name: GEPRCF722_BT_HD

# status
MCU F722 Clock=216MHz, Vref=3.28V, Core temp=56degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3627, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P
OSD: MAX7456 (30 x 13)
BUILD KEY: 7fabaee4ca71caad10e226356c805c39 (4.4.0)
System Uptime: 101 seconds, Current Time: 2023-02-16T19:20:15.289+00:00
CPU:49%, cycle time: 123, GYRO rate: 8130, RX rate: 15, System rate: 9
Voltage: 3 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
Arming disable flags: RXLOSS CLI MSP RPMFILTER NO_ACC_CAL

# flash_info
Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=0, totalSize=0 JEDEC ID=0x00000000
Partitions:
FlashFS size=0, usedSize=0

# dump hardware

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Feb  5 2023 / 01:51:07 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name GEPRCF722_BT_HD
manufacturer_id GEPR

# resources
resource BEEPER 1 C15
resource MOTOR 1 C08
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C07
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 B01
resource MOTOR 8 B00
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 C10
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 C11
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 B10
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 NONE
resource I2C_SDA 2 B11
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 C04
resource LED 2 NONE
resource LED 3 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_SCK 4 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MISO 4 NONE
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource SPI_MOSI 4 NONE
resource ADC_BATT 1 C01
resource ADC_RSSI 1 C00
resource ADC_CURR 1 C02
resource ADC_EXT 1 NONE
resource PINIO 1 C13
resource PINIO 2 C14
resource PINIO 3 B08
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 NONE
resource OSD_CS 1 B12
resource GYRO_EXTI 1 A08
resource GYRO_EXTI 2 B02
resource GYRO_CS 1 A15
resource GYRO_CS 2 C03
resource USB_DETECT 1 NONE
resource PULLUP 1 NONE
resource PULLUP 2 NONE
resource PULLUP 3 NONE
resource PULLUP 4 NONE
resource PULLDOWN 1 NONE
resource PULLDOWN 2 NONE
resource PULLDOWN 3 NONE
resource PULLDOWN 4 NONE

# timer
timer A00 AF2
# pin A00: TIM5 CH1 (AF2)
timer A03 AF3
# pin A03: TIM9 CH2 (AF3)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer C06 AF3
# pin C06: TIM8 CH1 (AF3)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer C07 AF3
# pin C07: TIM8 CH2 (AF3)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer A01 AF1
# pin A01: TIM2 CH2 (AF1)

# dma
dma SPI_MOSI 1 NONE
dma SPI_MOSI 2 NONE
dma SPI_MOSI 3 NONE
dma SPI_MOSI 4 NONE
dma SPI_MISO 1 NONE
dma SPI_MISO 2 NONE
dma SPI_MISO 3 NONE
dma SPI_MISO 4 NONE
dma SPI_TX 1 NONE
dma SPI_TX 2 NONE
dma SPI_TX 3 NONE
dma SPI_TX 4 NONE
dma SPI_RX 1 NONE
dma SPI_RX 2 NONE
dma SPI_RX 3 NONE
dma SPI_RX 4 NONE
dma ADC 1 NONE
dma ADC 2 NONE
dma ADC 3 0
# ADC 3: DMA2 Stream 0 Channel 2
dma UART_TX 1 NONE
dma UART_TX 2 NONE
dma UART_TX 3 NONE
dma UART_TX 4 NONE
dma UART_TX 5 NONE
dma UART_TX 6 NONE
dma UART_TX 7 NONE
dma UART_TX 8 NONE
dma UART_RX 1 NONE
dma UART_RX 2 NONE
dma UART_RX 3 NONE
dma UART_RX 4 NONE
dma UART_RX 5 NONE
dma UART_RX 6 NONE
dma UART_RX 7 NONE
dma UART_RX 8 NONE
dma pin A00 0
# pin A00: DMA1 Stream 2 Channel 6
dma pin A03 NONE
dma pin C08 1
# pin C08: DMA2 Stream 4 Channel 7
dma pin C06 0
# pin C06: DMA2 Stream 2 Channel 0
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin C07 1
# pin C07: DMA2 Stream 3 Channel 7
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin A01 0
# pin A01: DMA1 Stream 6 Channel 3

# feature
feature -RX_PPM
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -RANGEFINDER
feature -TELEMETRY
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -RX_SPI
feature -ESC_SENSOR
feature -ANTI_GRAVITY
feature RX_SERIAL
feature GPS
feature LED_STRIP
feature OSD
feature ANTI_GRAVITY

# serial
serial 20 1 115200 57600 0 115200
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 3 1 115200 57600 0 115200
serial 4 2 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# master
set gyro_to_use = FIRST
set adc_device = 3
set blackbox_device = NONE
set dshot_burst = ON
set dshot_bitbang = AUTO
set dshot_bitbang_timer = AUTO
set current_meter = ESC
set battery_meter = ADC
set vbat_scale = 110
set ibata_scale = 100
set beeper_inversion = ON
set beeper_od = OFF
set beeper_frequency = 0
set max7456_clock = NOMINAL
set max7456_spi_bus = 2
set max7456_preinit_opu = OFF
set led_inversion = 0
set pinio_config = 129,1,1,1
set pinio_box = 0,40,41,42
set usb_msc_pin_pullup = ON
set flash_spi_bus = 0
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_i2cBus = 0
set gyro_1_i2c_address = 0
set gyro_1_sensor_align = CW270
set gyro_1_align_roll = 0
set gyro_1_align_pitch = 0
set gyro_1_align_yaw = 2700
set gyro_2_bustype = SPI
set gyro_2_spibus = 1
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

*** Support data submitted *** 
Id: "95dac97b-4231-46dd-893a-6f1cac0fcf07"

# copy ID and provide to the betaflight team.
```

### Flight controller

Geprc Span F722-BT-HD-v2

### Other components

TBS Nano RX on uart3
GPS TBS M8.2 on UART5
Walksnail on UART1

### How are the different components wired up (including port information)

TBS Nano RX on uart3
GPS TBS M8.2 on UART5
Walksnail on UART1

haslinghuis, 2023-02-16:
Please see: https://betaflight.com/docs/wiki/configurator/firmware-flasher-tab#board-defines

If your board peripherals are not recognized please help us add the required configuration details.

To get the required information follow this procedure:

1. Flash your board with the Core Only switch enabled
2. Go to the CLI tab and click the Submit Support Data button.
3. With this generated support ID we have all required information to update board configuration files.

MarcinKnD, 2023-02-17:
OK. here You have Submit Support Data on CORE ONLY flashed GEPRC Mark4 with Span F722-BT-HD Betaflight 4.4

`# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 29 2023 / 02:32:03 (4605309d8) MSP API: 1.45

# config: YES
# board: manufacturer_id: GEPR, board_name: GEPRCF722_BT_HD

# status
MCU F722 Clock=216MHz, Vref=3.28V, Core temp=51degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 3791, max available: 16384
Devices detected: SPI:1, I2C:0
Gyros detected: gyro 1 locked dma
GYRO=ICM42688P, ACC=ICM42688P
OSD: MAX7456 (30 x 13)
BUILD KEY: d171e91b63de268fe5acf30bc7664f6c (4.4.0)
System Uptime: 24 seconds, Current Time: 2023-02-17T16:16:09.104+00:00
CPU:49%, cycle time: 131, GYRO rate: 7633, RX rate: 15, System rate: 9
Voltage: 41 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 22
SD card: Manufacturer 0x0, 491520kB, 12/2021, v0.0, 'APPSD'
Filesystem: Ready
Arming disable flags: RXLOSS CLI MSP NO_ACC_CAL

# flash_info
Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=0, totalSize=0 JEDEC ID=0x00000000
Partitions:
FlashFS size=0, usedSize=0

# dump hardware

# version
# Betaflight / STM32F7X2 (S7X2) 4.4.0 Jan 29 2023 / 02:32:03 (4605309d8) MSP API: 1.45

# config: YES

# start the command batch
batch start

board_name GEPRCF722_BT_HD
manufacturer_id GEPR

# resources
resource BEEPER 1 C15
resource MOTOR 1 C08
resource MOTOR 2 C06
resource MOTOR 3 C09
resource MOTOR 4 C07
resource MOTOR 5 B06
resource MOTOR 6 B07
resource MOTOR 7 B01
resource MOTOR 8 B00
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
[... 266 lines cut ...]
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set i2c1_pullup = OFF
set i2c1_clockspeed_khz = 800
set i2c2_pullup = OFF
set i2c2_clockspeed_khz = 800
set i2c3_pullup = OFF
set i2c3_clockspeed_khz = 800
set mco2_on_pc9 = OFF
set scheduler_relax_rx = 25
set scheduler_relax_osd = 25
set box_user_1_name = -
set box_user_2_name = -
set box_user_3_name = -
set box_user_4_name = -

# end the command batch
batch end

*** Support data submitted *** 
Id: "5a251c3a-3194-46ba-bc22-802d132a3e8f"

# copy ID and provide to the betaflight team.`

On this flash BlackBox works, and GPS connected to UART5 is not. There is no way to save GPS enable option in configuration tab , after restart is switched off

haslinghuis, 2023-02-19:
Please follow the 3 steps provided to get an ID with the correct information.

MarcinKnD, 2023-02-19:
1. I flashed Core Only
2. Support ID from Core Only is in Updated 1st post
3. What else You whant me to do?

MarcinKnD, 2023-02-19:
Please make attention that on CORE ONLY flash, there is no problem with Black Box. It works. My problem is with GPS, there is no way to activate GPS in configuration tab.

Problem with Black Box occured with cloud Flash, when gps is working and Black Box is not.

I can't get this two functions work simultaneously

haslinghuis, 2023-02-19:
Core Only contains all drivers needed for support. When above PR is merged you can flash again and blackbox should be available. We just have to get used to pick the options we need like GPS. The most obvious options are pre-selected for convenience and soon configurator will "remember" your choices for each board by getting the info from the cloud API.

closed 2023-02-20 by haslinghuis