betaflight/betaflight #12378: DFU mode

litlehead, 2023-02-17:
### Describe the bug

When I plug into B.F. Whoop goes into DFU  mode, and will not stay out of DFU mode, even after running Impulse RC Fixer .          Supperbee F4 lite Frsky V 1.2 board. Flashed to latest Firmware in B.F.

### To Reproduce

plug into BF and open up consecutive tabs

### Expected behavior

my whoop goes into DFU mde and I can't get it out untill I unpug and close BF then start BF up again.

### Support ID

```txt
Id: "8ef7fb36-cfc3-4ecd-a05d-292fbdd4db26"
```

### Flight controller

Supperbee F4 lite FRSKY V 1.2

### Other components

HDZERO Whooplite with camera and mipicable, Happy model EX0802 19,000KV motors.

### How are the different components wired up (including port information)

Whooplite board wired to Flight controler via uart 1 , motors soldered to F.C..

haslinghuis, 2023-02-19:
Try disabling VTX Administrator in ELRS settings.

automatt1c, 2023-03-04:
I have the same issue, It has to do with Betaflight 4.4 and Displayport set to MSP.  If you set it to Auto or leave it default the issue doesnt happen. Doesnt happen on all my boards though. Currently have the issue with "Skystars H7 Dual BMI270" and Walksnail VTX. With mine i can occasionally connect it without it going into DFU mode but its random when its work.

Once i set Displyport to Auto then the problem goes away  but the VTX MSP no longer works.

P-I-Engineer, 2023-03-25:
i have this issue with betafpv f405aio w/ walksnail vtx.

automatt1c, 2023-03-25:
I noticed that it will still happen if you connect the flight controller to a power bank. As long as the walksnail VTX is powered then the problem doesn't seem to happen. So if you connect battery power to quad and then connect USB that will work fine.

P-I-Engineer, 2023-03-25:
same if you unplug the cable from the walksnail, it won't boot the dfu mode. glad i'm not going crazy, this sounds like a bug.

litlehead, 2023-03-25:
I have to put a battery on to keep it from going into DFU, don’t know why.

closed 2023-04-25 by haslinghuis