betaflight/betaflight #12387: Betaflight 4.4 - Can't choose Onboard Flash for Blackbox logging

TommieTYX, 2023-02-18:
### Describe the bug

After flashing to the latest release of BF4.4, even with "Flash Storage"  option selected, under black box tab, there is no flash storage option to choose, only serial logging.

I found this thread https://github.com/betaflight/betaflight/issues/12127#issuecomment-1368372723

Tried the suggested parameter in custom defines FLASH_W25Q128FV and it works.

### To Reproduce

Install the Betaflight 4.4.0 [29-Jan-2023] FF RACEPIT flight controller using the Betaflight Configurator 10.9.0
Go into the blackbox tab
Try to select "Onboard Flash" for the Blackbox logging device

### Expected behavior

Expect to be able to choose "Onboard Flash for Blackbox logging as was possible in previous stable releases

### Support ID

```txt
b9d77161-7cef-4f3a-9581-6f190dad3ee3
```

### Flight controller

FuriousFPV Racepit

David-OConnor, 2023-02-21:
Same issue on G473. Shows "No dataflash chip found" in the system status bar at the top.

haslinghuis, 2023-02-21:
@David-OConnor target? It's all macro driven now with some magic.

David-OConnor, 2023-02-21:
I was able to sort it out on my end by compiling with `make TARGET=STM32G47X EXTRA_FLAGS="-DUSE_FLASH -DUSE_FLASH_M25P16"`

Before ~a week ago, a unified G4 target (eg from the Nightlies page, or by compiling without those flags) worked.

closed 2023-02-22 by blckmn