betaflight/betaflight #12391: SPI ELRS 3.0 + LEDs Wont dont work together.

lesharodin, 2023-02-19:
### Describe the bug

Since 4.3.1 BF we cant fly whoops with leds, thats sad.
Last issue https://github.com/betaflight/unified-targets/pull/662 was fixed with just disabling LED STRIP in defaults https://github.com/betaflight/unified-targets/pull/662
Thats not a fix ;)

### To Reproduce

Try to fly 4.3.1/4.3.2/4.4 with SPI ELRS and  LED Strip, get random FS everytime

### Expected behavior

Expecting to fly with SPI ELRS and LED Strip on a whoop

### Support ID

```txt
nothing interesting here
```

### Flight controller

CRAZYBEEF4SX1280

haslinghuis, 2023-02-19:
Did you try: https://github.com/betaflight/betaflight/pull/12057

lesharodin, 2023-02-19:
> Did you try: #12057

Is this commit in released 4.4? So yes, still FS randomly with LED Strip enabled.

haslinghuis, 2023-02-19:
Milestone is 4.3,3

lesharodin, 2023-02-19:
> Milestone is 4.3,3
Its broken since 4.3.1 to 4.4, even newer builds of 4.5(?) https://github.com/betaflight/betaflight/pull/12231 its still the same

haslinghuis, 2023-02-19:
Please add output of 'resource show all' both for working 4.3 and not working 4.3.1

lesharodin, 2023-02-19:
> Please add output of 'resource show all' both for working 4.3 and not working 4.3.1

4.3.0 (working led strip and elrs)
```
A00: LED_STRIP
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: RX_SPI_EXPRESSLRS_RESET
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: RX_SPI_EXPRESSLRS_BUSY
A14: PREINIT 2
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: PREINIT 4
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1 : DSHOT_BITBANG 2
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1 : LED_STRIP
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 1: FREE
DMA1 Stream 2: LED_STRIP
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: DSHOT_BITBANG 2
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

4.4 (broken elrs with led strip):

```
A00: LED_STRIP
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: RX_SPI_EXPRESSLRS_RESET
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: RX_SPI_EXPRESSLRS_BUSY
A14: SWD
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3 : MOTOR 1
TIM3: FREE
TIM4:
    CH1 : MOTOR 2
    CH2 : MOTOR 3
    CH3 : MOTOR 4
TIM5:
    CH1 : LED_STRIP
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 2
DMA1 Stream 1: MOTOR 1
DMA1 Stream 2: LED_STRIP
DMA1 Stream 3: MOTOR 3
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: FREE
DMA1 Stream 7: MOTOR 4
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

haslinghuis, 2023-02-19:
Please try with `set dshot_bitbang = ON`

Can have a `diff` please.

haslinghuis, 2023-02-20:
Working config is using bitbang. To sum up:

```
WORKING CONFIG:

Currently active Timers:
-----------------------
TIM1:
    CH1 : DSHOT_BITBANG 2
TIM5:
    CH1 : LED_STRIP

Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 2: LED_STRIP
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA2 Stream 0: ADC 1
DMA2 Stream 1: DSHOT_BITBANG 2

NOT WORKING:
==========

# timer show

Currently active Timers:
-----------------------
TIM2:
    CH3 : MOTOR 1
TIM4:
    CH1 : MOTOR 2
    CH2 : MOTOR 3
    CH3 : MOTOR 4
TIM5:
    CH1 : LED_STRIP

=================

# dma show

Currently active DMA:
--------------------
DMA1 Stream 2: LED_STRIP
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1

```

haslinghuis, 2023-02-20:
@SteveCEvans can you give some light on this issue?

lesharodin, 2023-02-20:
Tried with bitbang = on and led strip on, still random FS on SPI ELRS
DIFF:
```
board_name CRAZYBEEF4SX1280
manufacturer_id HAMO
mcu_id 0044004b5047500620333539
signature 

# name: @lesharodin

# resources
resource MOTOR 1 B08
resource MOTOR 2 B07
resource MOTOR 3 B06
resource MOTOR 4 B10

# feature
feature -RX_SERIAL
feature -SOFTSERIAL
feature -AIRMODE
feature RX_SPI

# serial
serial 1 8192 115200 57600 0 115200

# map
map TAER1234

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 13 3 1700 2100 0 0
aux 3 28 1 1700 2100 0 0
aux 4 35 2 1700 2100 0 0

# vtxtable
vtxtable bands 7
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWRACE  L CUSTOM  5333 5373 5413 5453 5493 5533 5573 5613
vtxtable band 7 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 5
vtxtable powervalues 1 2 25 100 400
vtxtable powerlabels 0 RCE 25 100 400

# master
set gyro_lpf1_static_hz = 0
set dyn_notch_count = 0
set dyn_notch_q = 450
set gyro_lpf1_dyn_min_hz = 0
set acc_lpf_hz = 1
set acc_trim_pitch = 4
set acc_calibration = -107,50,-1,1
set rc_smoothing_auto_factor = 25
set rc_smoothing_auto_factor_throttle = 25
set serialrx_provider = CRSF
set dshot_idle_value = 600
set dshot_bidir = ON
set dshot_bitbang = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set align_board_yaw = 90
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 310
set vbat_warning_cell_voltage = 320
set yaw_motors_reversed = ON
set crashflip_motor_percent = 100
set small_angle = 180
set report_cell_voltage = ON
set osd_logo_on_arming = ON
set osd_tim2 = 2577
set osd_rssi_pos = 405
set osd_link_quality_pos = 2548
set osd_rssi_dbm_pos = 2528
set osd_tim_1_pos = 233
set osd_tim_2_pos = 2517
set osd_flymode_pos = 2533
set osd_throttle_pos = 2490
set osd_vtx_channel_pos = 385
set osd_ah_sbar_pos = 205
set osd_current_pos = 2464
set osd_mah_drawn_pos = 2553
set osd_craft_name_pos = 2537
set osd_warnings_pos = 2505
set osd_avg_cell_voltage_pos = 2496
set osd_canvas_width = 30
set osd_canvas_height = 16
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 4
set vtx_freq = 5917
set vcd_video_system = PAL
set expresslrs_uid = 50,88,33,71,2,57
set expresslrs_domain = ISM2400
set expresslrs_rate_index = 1
set craft_name = @lesharodin
set pilot_name = @LeshaRodin

profile 0

# profile 0
set vbat_sag_compensation = 50
set crash_time = 250
set iterm_relax_cutoff = 20
set p_pitch = 71
set i_pitch = 63
set d_pitch = 67
set f_pitch = 249
set p_roll = 68
set i_roll = 60
set d_roll = 62
set f_roll = 239
set p_yaw = 68
set i_yaw = 60
set f_yaw = 239
set angle_p_gain = 70
set angle_limit = 65
set angle_earth_ref = 50
set d_min_roll = 62
set d_min_pitch = 67
set d_max_advance = 0
set thrust_linear = 20
set feedforward_averaging = 2_POINT
set feedforward_smooth_factor = 35
set feedforward_jitter_factor = 4
set feedforward_boost = 18
set feedforward_max_rate_limit = 95
set dyn_idle_min_rpm = 40
set simplified_master_multiplier = 105
set simplified_i_gain = 50
set simplified_d_gain = 200
set simplified_pi_gain = 145
set simplified_dmax_gain = 0
set simplified_feedforward_gain = 190
set simplified_pitch_d_gain = 95
set tpa_mode = PD
set tpa_breakpoint = 1250

profile 1

profile 2

profile 3

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set rateprofile_name = MCK
set roll_rc_rate = 22
set pitch_rc_rate = 18
set yaw_rc_rate = 16
set roll_expo = 56
set pitch_expo = 56
set yaw_expo = 56
set roll_srate = 75
set pitch_srate = 60
set yaw_srate = 53

rateprofile 1

# rateprofile 1
set rateprofile_name = Justice
set roll_rc_rate = 20
set pitch_rc_rate = 20
set yaw_rc_rate = 20
set roll_expo = 50
set pitch_expo = 50
set yaw_expo = 25
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50

rateprofile 2

rateprofile 3

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

haslinghuis, 2023-02-20:
Think we need TIM3 instead of TIM5 as before:

```
Currently active DMA:
--------------------
DMA1 Stream 0: SPI_MISO 3
DMA1 Stream 2: LED_STRIP
DMA1 Stream 3: SPI_MISO 2
DMA1 Stream 4: SPI_MOSI 2
DMA1 Stream 5: SPI_MOSI 3
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 2: SPI_MISO 1
DMA2 Stream 3: SPI_MOSI 1

# timer show

Currently active Timers:
-----------------------
TIM2:
    CH3 : MOTOR 1
TIM4:
    CH1 : MOTOR 2
    CH2 : MOTOR 3
    CH3 : MOTOR 4
TIM5:
    CH1 : LED_STRIP
```

Made a local build with CRAZYBEEF4SX1280, DSHOT, LED_STRIP.

Please check carefully (as I only have a bare sample board for testing):

[betaflight_4.5.0_STM32F411_CRAZYBEEF4SX1280.zip](https://github.com/betaflight/betaflight/files/10788181/betaflight_4.5.0_STM32F411_CRAZYBEEF4SX1280.zip)

haslinghuis, 2023-02-21:
@lesharodin please test the hex provided above.

mchelifpv, 2023-02-25:
Tested master branch with RX_EXPRESSLRS_TIMER_INSTANCE=TIM3 - works fine on two whoops with DiamondF4-ELRS. But I have a new (or maybe an old) problem. Current sensor shows 1A in postflight stats. Current divider is set to 1175.

haslinghuis, 2023-02-25:
Fixed in https://github.com/betaflight/unified-targets/pull/915

closed 2023-02-25 by haslinghuis