betaflight/betaflight #12396: No "On Board" Blackbox for HAKRC F4120 AIO via BF4.4 Cloud Build Config

Nanopilotz, 2023-02-21:
### Describe the bug

"On Board" Blackbox cannot be enabled on HAKRC F4120 AIO Flight Controller via BF4.4 Cloud Build Config. 
Previously in BF4.2 it can be enabled in CLI via:

resource FLASH_CS 1 A00
set blackbox_device = SPIFLASH
set flash_spi_bus = 2

*Note: The CLI can be done after using "Core only" mode but not via Flash Storage option in Cloud Build mode. 

HAKRC F4120 AIO FC is using BF firmware target: MATEK411. It has:
a) Baro (BMP280)
b) Blackbox (Winbond W25Q128JV)

Reference:
[image]

### To Reproduce

1. Update HAKRC F4120 AIO flight controller to BF4.4 (Firmware Target: MATEK411) via BF Configurator 10.9
2. Enable On Board Blackbox in BF Configurator 10.9 - Blackbox tab.

### Expected behavior

On Board Blackbox shall be available in the BF Configurator 10.9 - Blackbox tab option.

### Support ID

```txt
Id: "8eefb587-82e7-4ad1-858d-1b18ab35a30f"
```

### Flight controller

HAKRC F4120 AIO (20mm x 20mm holes)

### Add any other context about the problem that you think might be relevant here

HAKRC F4120 AIO Flight Controller requires dedicated Firmware Target like what BF Team did for HAKRC F722 DM Flight Controller using FW Target: HAKRC722MINI (https://github.com/betaflight/betaflight/issues/10769). The old FW Target: MATEK722SE no longer valid for HAKRC 722 DM.

haslinghuis, 2023-02-21:
HAKRCF411D is almost identical.

Try adding `BARO_BMP280` (you can test by enabling expert mode and put it in the custom defines field).

Nanopilotz, 2023-02-22:
As per requested, I have tried Cloud Build Mode with FW Target: HAKRCF411D, however it becomes worst (Gyro, Acc, & Baro are not available/detected) although i have included all of them in Custom Define.

My Cloud Build mode:
[image]

My Cloud Build Result:
[image]

The good thing about HAKRCF411D, the "On Board Flash" for Blackbox is available:
[image]

Due to the above, I have also tried the "Core Only" mode, and the result is amazing. Everything are available (Gyro, Acc, Baro, OnBoard Blackbox) except GPS (noted that GPS will be disable in "Core Only" mode because its coding size is quite big for the F411 & F722 MCU flash memory)
[image]

[image]

However, i cannot run my drone in "Core Only" mode because i need the GPS. Appreciate your help. Thanks.

haslinghuis, 2023-02-22:
1. Enable expert mode
2. Select HAKRC-F411D
3. Add GYRO_SPI_MPU6000 ACC_SPI_MPU6000 BARO_BMP280 to custom defines
4. Load firmware online
5. Flash

```
# status
MCU F411 Clock=108MHz (PLLP-HSE), Vref=3.31V, Core temp=39degC
Stack size: 2048, Stack address: 0x2001fff0
Configuration: CONFIGURED, size: 3751, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 1 locked dma
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MAX7456 (30 x 13)
BUILD KEY: 5b6cd516daf3658a24bf02f77fcd595f (4.4.0)
System Uptime: 24 seconds, Current Time: 2023-02-21T07:38:01.950+00:00
CPU:25%, cycle time: 125, GYRO rate: 8000, RX rate: 15, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: RXLOSS CLI MSP MOTOR_PROTO

# flash_info
Flash sectors=0, sectorSize=0, pagesPerSector=0, pageSize=0, totalSize=0 JEDEC ID=0x00000000
Partitions:
FlashFS size=0, usedSize=0
```

Nanopilotz, 2023-02-22:
The problem still remain the same (no Gyro, no Acc, no Baro). I think the problem in the Cloud Build is due to the Unified Target for HAKRC411D was not updated with:

#Define USE_GYRO_SPI_MPU6000
#Define USE_ACC_SPI_MPU6000
#Define USE_BARO_BMP280

[image]

link: https://github.com/betaflight/unified-targets/blob/master/configs/default/HARC-HAKRCF411D.config

Kindly note, due to worldwide chip shortage, manufacturer tend to change their Gyro chip from MPU to ICM or even BMI. If BF Team only add #define GYRO_SPI_ICM42688P for the latest Flight Controller batch, the previous Flight Controller batch with MPU6000 might not be able to use the same FW Target: HAKRCF411D. 

Is it possible for BF Team to load all types of GYRO use on this chip in the Unified Target: HAKRCF411D.config?

In this case the Custom Define is very helpful for future proof should the manufacturer keep changing their chip. However, user must identify the correct chip and load it manually in the Custom Define for the FC to function.

closed 2023-02-22 by blckmn

Nanopilotz, 2023-02-22:
After the unified targets update, the Gyro, Acc, Baro & On board Blackbox are functioning. However the GPS icon in BF Configurator is not functioning although the GPS feature was enabled. 

[image]

[image]

[image]

In my case, the BN-180 GPS is connected via Soft Serial 1 at 9600 Baud Rate (the only speed that can make it work via soft serial). 

Previously with BF4.2 (FW Target: MATEKF411), the GPS is functioning. Below are the resources in CLI:

resource Serial Tx_11 B02 #Without pysical pad. B02 also used for Beeper
resource Serial Rx_11 B03 #Rx pad (a.k.a RT6 in the picture)

[image]

Nanopilotz, 2023-02-23:
After the unified target update, the Gyro, Acc, Baro, Onboard Flash (Blackbox) and GPS icons are functioning.

User shall add GYRO_SPI_MPU6000 ACC_SPI_MPU6000 BARO_BMP280 FLASH_W25Q128FV in the Custom Define during the firmware update.

BN-180 GPS connected via Soft serial 1 (9600 Baud rate). 

In the CLI:
Resource Serial_Tx 11 B02 (No physical connection. GPS only use 3 wires)
Resource Serial_Rx 11 B03 (physical connection on RT6 pad to receive signal)

In Configuration tab:
1. GPS = enabled
2. GPS protocol = NMEA (for BN-180)
3. Auto Baud = Disabled
4. Auto Config = Enabled

However based on my flight test, the Blackbox is not recording.

[image]