betaflight/betaflight #12398: CRSF baud rate is incorrect.

hydra, 2023-02-21:
### Describe the bug

The baudrate for CRSF is defined as 420,000baud.  This is incorrect.  Trappy from TBS confirmed to me today that CRSF uses 416,666 and always has.  I've not verified this statement however.

When the wrong baud rates are uses (e.g. RX/FC mismatch) then FCs will get framing errors, data corruption, CRC mismatches and packet loss for RC channel data. (Verified by inspecting UART `ISR` register and looking at the `FE` flag).

Related information:
* Old official TBS CRSF documentation (at least up to version 7) specified a 400,000baud ("400kbaud") which was corrected in revision 10 to be 416,666baud.
* ELRS has a default baud rate of 420,000baud.

Code:
BF - https://github.com/betaflight/betaflight/blob/master/src/main/rx/crsf_protocol.h#L30
ELRS - https://github.com/ExpressLRS/ExpressLRS/blob/c5f67bd28a9f559a0bba3a4bcf509a4340445df5/src/lib/OPTIONS/options.cpp#L84

Rev10 CRSF docs:
[image]

Rev7 CRSF docs:
[image]

### To Reproduce

Use mismatched baud rates on RX and FC.

### Expected behavior

No data corruption or packet loss.

* Agreement with BF, ELRS and TBS on the correct baud rate to use.
* Notification to users that it's changed, if/when it changes.
* Added to the list of known-issues until corrected.

### Support ID

```txt
N/A
```

### Flight controller

N/A

hydra, 2023-02-21:
Note that users of ELRS using SPI-based RX will be unaffected by any baud rate changes as there's no CRSF UART involved.

hydra, 2023-02-21:
After a bit more digging, CRSF V3 users may be less affected because the baud rate is auto-negociated, and the Rx tells the FC what baud rate to use.

However, for CRSF V3, if there are 3 CRC errors in a row after baud-rate negotiation BF will drop-back to the pre-negotiated baud rate without the receiver knowing and the receiver somehow has to handle it, but from the little information I have (in the official specs) it's unclear how the receiver is supposed to handle a baud-rate-change on the FC, if at all.

It can also do this during flight, after arming, as I couldn't find any arming status check before dropping back to the pre-negotiated baud rate.

See also:

See also:
* crc check and counter increment - https://github.com/betaflight/betaflight/blob/master/src/main/rx/crsf.c#L459-L474
* setCrsfDefaultSpeed - https://github.com/betaflight/betaflight/blob/1a6fa4ddbd5a53c79579c2770402e039f6591908/src/main/telemetry/crsf.c#L129-L137
* crsfRxUpdateBaudrate - https://github.com/betaflight/betaflight/blob/master/src/main/rx/crsf.c#L661-L666
* https://github.com/betaflight/betaflight/blob/master/src/main/rx/crsf.c#L64

klutvott123, 2023-02-21:
Yes, it's true that the correct baud rate is 416666. 
For CRSF V3, the receiver uses telemetry from the FC to detect a mismatch. If telemetry is disabled, the FC will send heartbeat frames instead, and if the telemetry wire isn't connected it wouldn't be able to negotiate the baud rate anyway. Both incomplete frames and crc errors are counted as errors.

hydra, 2023-02-21:
Looks like Kiss V1 was also affected by this, at least according to this comment in the ELRS codebase (link below), perhaps @fedorcomander can confirm what baud rate new KISS firmware is using for CRSF these days?

https://github.com/ExpressLRS/ExpressLRS/blob/c5f67bd28a9f559a0bba3a4bcf509a4340445df5/src/user_defines.txt#L48-L50

fedorcomander, 2023-02-21:
Kiss 400000

Ultra 400000 and 2000000 v2/v3 auto negotiation. (afair on KISS F3 cpus its 1000000 max, but frankly never seen it), follow TBS protocol specs. With auto negotiation.

All my ERLS works fine ;)

fedorcomander, 2023-02-21:
In fact receiver may request any baud rate. :) on v3.

howels, 2023-02-22:
> 

Do you need to customise ELRS to use the 400K baud rate on RX or is that automatic too?

hydra, 2023-02-22:
> follow TBS protocol specs. With auto negotiation.

@fedorcomander Which version of the spec document did you follow? As the baud rate changed between Rev 7 and Rev 10, see OP above.

fedorcomander, 2023-02-22:
oh. interesting. i was not aware of that. will poke trappy. thanks. so far none of my clients had any issues btw. ---Cheers,Alex

fedorcomander, 2023-02-22:
anyhow. tbs is tad faster then 400kb. but it works fine on 400 withing range. since i dont have any issues with og devices - nog gonna change it. ---Cheers,Alex

SteveCEvans, 2023-02-22:
@hydra please post links to the TBS CRSF specifications.

hydra, 2023-09-22:
@haslinghuis Please can you re-open this, official confirmation from TBS is here:

https://github.com/tbs-fpv/freedomtx/issues/26#issuecomment-1731202619

klutvott123, 2023-09-22:
we should just change it

hydra, 2023-09-26:
> we should just change it

this is my feeling too, it's currently incorrect and should be corrected.

I'm not sure how happy other projects will be about it though, as they will have to communicate this to their users.

Suggest co-ordinating the change with the devs of at least the ELRS project.

fedorcomander, 2023-09-26:
i plan to make it configurable on ultra. ;) just in case ;) may be you should do the same. ---Cheers,Alex

hydra, 2023-09-26:
> i plan to make it configurable on ultra. ;) just in case ;) may be you should do the same.

Not sure if that's a good idea, traditionally all RX protocols, that I can remember, have used a fixed baud rate associated with the protocol.

Every additional option only serves to complicate things, confuse the user and generate more support issues.

KissUltra, 2023-09-26:
you dont say ;) still. run on 400 with zero issues for last 6 years. i think uart allows 5% error easy. ok. shutting up. game changes dramatically if cpu clocked internally. especially h7. ask me how i know. ;)---Cheers,Alex

ctzsnooze, 2023-12-19:
Closing because the difference is small.  ELRS deals running their baud rate report no errors.  A project to refresh CRSF is underway.  This specific issue can be laid to rest for now.

hydra, 2023-12-29:
why not just correct it in BF if the difference is so small that the error difference will not be noticable?  it's kinda silly to drift off-spec like this and currently just serves to confuse people.

ledvinap, 2023-12-29:
@ctzsnooze : There may be other error sources (internal RC and possibly others). Wasting baudrate margin is not a good idea

howels, 2023-12-29:
> @ctzsnooze : There may be other error sources (internal RC and possibly others). Wasting baudrate margin is not a good idea

The current situation provably works and eLRS and others have standardised on 420K as this was the observable implementation of CRSF at the RX. So please can we close this and move on? This is pedantry and not related to observable issues. 420K is an easy to use number already implemented without issue.

faryna1993, 2024-05-01:
I have a request, add the following command.
**set serialrx_baund 115200**

KissUltra, 2024-05-01:
> eLRS and others have standardised on 420K

Unfortunately only @hydra makes sense here. Pedantry to keep to the standard instead of reinventing your own square wheel and call it OK :) That is not pedantry. Thats laziness and ignorance. God luck. :)

faryna1993, 2024-05-01:
It seems that this is too difficult a task for you. Therefore, it is easy to attribute it to pedantry. Good luck. :)

KissUltra, 2024-05-01:
i can say exactly the same about your task. ;)

hydra, 2024-05-03:
> The current situation provably works and eLRS and others have standardised on 420K as this was the observable implementation of CRSF at the RX. So please can we close this and move on? This is [...]

In my recent testing here, as the hardware designer of at least 4 UART based ELRS RX's now, using both the ESP32 Pico D4 and the ESP32S3 I can 100% say that you DO notice observable errors and corruption of the data when using the wrong baud rate.

haslinghuis, 2024-05-03:
@hydra did you try #13614

hydra, 2024-05-03:
> @hydra did you try #13614

no, but happy to see the PR!

KissUltra, 2024-05-04:
Thank you @hydra and @haslinghuis

SunjunKim, 2024-05-08:
> In my recent testing here, as the hardware designer of at least 4 UART based ELRS RX's now, using both the ESP32 Pico D4 and the ESP32S3 I can 100% say that you DO notice observable errors and [...]

Do you have any data showing how serious the problem is?

fedorcomander, 2024-05-08:
i have arc error counter on the osd. at some flight was more then 1000 errors. apparently some packets were OK after arc check, but had WRONG channel data, so it was affecting flights.

> Do you have any data showing how serious the problem is?

CapnBry, 2024-05-08:
> i have arc error counter on the osd

What is the "arc error counter" that can be placed on the OSD?

KissUltra, 2024-05-08:
crc---Cheers,Alex

SunjunKim, 2024-05-09:
Sorry this is not much informative.. the conversation is like it’s A. -> what is it? -> it’s A.

Providing your gear setup comprehensively will be better helpful.

hydra, 2024-05-14:
Just commenting that #13614 does _not_ fix/complete this issue and that BF is not following the CRSF specification outlined in the OP here: https://github.com/betaflight/betaflight/issues/12398#issue-1593355393

Users wanting a build of BF that follows the CRSF spec have to add `#define USE_CRSF_OFFICIAL_SPEC` as it is not, currently, the default.

hydra, 2026-11-04:
Just a quick 'Hi' to everyone involved in this.  👋 and also to add this link to the public CRSF spec so that anyone coming here has the link:

https://github.com/tbs-fpv/tbs-crsf-spec

The published spec still lists 416666 as the default baud rate.

https://github.com/tbs-fpv/tbs-crsf-spec/blob/main/crsf.md#dual-wire-full-duplex-uart

As of 2026/11/04 both ELRS and BF still use 420000 for the default.

BF:
https://github.com/betaflight/betaflight/blob/master/src/main/rx/crsf_protocol.h#L33-L37

ELRS:
https://github.com/ExpressLRS/ExpressLRS/blob/master/src/user_defines.txt#L44-L47
https://github.com/ExpressLRS/ExpressLRS/blob/master/src/lib/OPTIONS/options.cpp#L197

Additionally, ELRS uses 400,000 by default for TX modules and depending on the MCU will either autodetect the baud rate or cycle through a list of baud rates.

KissUltra, 2026-11-04:
Yeh, for now, i detect elrs, and switch fc serial to 420k, at least its not bothering users. :) Hi Dominic. Its been a while :)

haslinghuis, 2026-11-04:
@hydra you still can manually override to use the spec using https://github.com/betaflight/betaflight/pull/13614

Personal would like to use the spec by default.

KissUltra, 2026-11-04:
Honestly, its a elrs issue, not betaflight nor ultra. They should follow the specs. fc is just a consumer. this story goes on for 3 years now.

haslinghuis, 2026-11-04:
Guess it should be 

```re
#ifdef USE_ERLS_NOT_USING_CRSF_OFFICIAL_SPEC
#define CRSF_BAUDRATE           420000
#else
#define CRSF_BAUDRATE           416666
#endif
```

CapnBry, 2026-11-04:
Oh this old chestnut eh? TBS didn't even know what baud rate is the correct baud rate and changing the default in ExpressLRS only breaks everyone's system so tell me again why we should change it?

When everyone is on the "wrong" value, is it the wrong value? :-D

fedorcomander, 2026-11-04:
yes. it is wrong.

haslinghuis, 2026-11-04:
There is no wrong or right value here - just using the official spec is the only right path forward - the PR has a fallback for ELRS

fedorcomander, 2026-11-04:
yes. there is right and wrong. its tbs protocol. everything not in it by definition IS wrong. its pointless discussion for sure. neither tbs or elrs will change anything. tbs because why? and elrs because it used to be like this since we made it and bf supports it. whole situation is rather comical. one implements competitor protocol without permission, does it wrong (i know, they did not knowing specs) and call it standard. well. works on bf - makes it standard ;) i solved my problem. using different baudrates. can detect receiver id and act accordingly. not a rocket science. no defines. no configuration. ;) all automatic. its a mess. i had enough. ;)

haslinghuis, 2026-11-04:
Why - just be neutral - new PR should go in the right direction - sure it will time for ELRS to adapt - but specs should be implemented.

KissUltra, 2026-11-04:
couldnt agree more.

hydra, 2026-11-04:
> 26\. Oh this old chestnut eh?

@CapnBry yeah, that's what I thought when I got asked about this today..  I had to refer myself back to this issue so I could point at the current state of things.

> 27\. yes. it is wrong

@fedorcomander 💯 

> 30\. specs should be implemented

@haslinghuis 💯 

> 22\. Hi Dominic. Its been a while :)

@KissUltra sure has! 👋 

everyone:

It would be nice, that perhaps both BF and ELRS devs could agree to change it at the same time, say perhaps on January 1st 2027. So that we can all get on the same page and not have to deal with this, or the ensuing code bloat from autodetection systems as a result.

perhaps we can ask the AI to summarise the error rates for the MCUs and oscillators we all use in both projects at 416666 and 420000 baud and see just which devices might actually be affected by this change, given that the actual baud rate difference is small and that not all MCU/oscillator combinations will actually able to achieve one or the other or both baud rates? especially since there are now many new platforms in BF since this issue was created.  @haslinghuis perhaps that's one for you?

CapnBry, 2026-11-05:
The problem with changing it is that users plug a receiver into BF and it works currently. Even changing both sides to the standard the user is still going to get receivers with old firmware for probably years to come, or grabbing an existing receiver will have the wrong baud rate as well. So everyone now needs to keep in their mind that X version only works with Y version and what versions work with what again? It is like the nightmare of SPI ELRS support and matching version numbers and build flags. It would be a source of immense frustration for years to come as things that used to work don't after upgrading one side. And the proposed solution is that the user should just memorize this weird string they have to type into the defines field in the BF flasher? Nobody is going to remember what that string is supposed to be, and a typo fails silently, and the option isn't even presented to users if they're not in 'expert mode'. That solution is laughable.

And for what? I do agree that we've all decided that the spec has a different number in it now but what does changing betaflight get for changing it? There's no advantage to the different number. You will not get an award for Most Improved TBS Spec Adjuster. Only pain for the user comes.

tbs-trappy, 2026-11-05:
> You will not get an award for Most Improved TBS Spec Adjuster.

hey now. don't give me ideas :)

haslinghuis, 2026-11-05:
@hydra here are the numbers you asked for. They're calculated from source, not measured. For each MCU I took the UART clock from Betaflight's clock setup, then applied the vendor library's divider code, including how it rounds. On the receiver side I used the framework versions that ExpressLRS pins.

### Flight controller: actual rate vs requested

Default clock settings are in **bold**. Where two values are given for F4/F7/APM32, they are APB2 / APB1: USART1/6, then the other UARTs.

| MCU / clock | UART clock | Error at 416666 | Error at 420000 |
|---|---|---|---|
| **F405 @168** | 84 / 42 MHz | −0.198% | 0.000% |
| F405 @216 (overclock) | 108 / 54 MHz | +0.077 / −0.308% | +0.056 / +0.446% |
| **F411 @108** | 108 / 54 MHz | +0.077 / −0.308% | +0.056 / +0.446% |
| F446 | 90 / 45 MHz | 0.000% | +0.134% |
| F7 @216 | 108 / 54 MHz | +0.077 / −0.308% | +0.056 / −0.332% |
| **G474 @168** | 168 MHz | +0.050% | 0.000% |
| H743 rev Y / rev V | 100 / 120 MHz | 0.000% | +0.040 / −0.100% |
| H723 / H730 | 130 MHz | 0.000% | −0.154% |
| H5 | 250 MHz | 0.000% | +0.040% |
| N6 (USART1 runs from the 64 MHz HSI) | 64 MHz | −0.260% | +0.251% |
| AT32F435 | 144 MHz | −0.115% | −0.042% |
| APM32F405 @168 | 84 / 42 MHz | −0.198% | 0.000% |
| APM32F425 @240 | 120 / 60 MHz | 0.000% | −0.100% |
| GD32F460 | 120 / 60 MHz | 0.000% | −0.100% |
| GD32H7 | 300 / 150 MHz | 0.000% | +0.040% |
| RP2350 (UART / PIO) | 150 MHz | 0.000% | −0.030 / −0.004% |

The F4 rows (F405, F411, F446) use ST's standard peripheral library. It truncates the divider before rounding, which is why F411 and the overclocked F405 land at +0.446% instead of −0.332% at 420000.

### ExpressLRS receivers: actual rate

ELRS opens the port with a plain `Serial.begin()`, so the divider comes from the framework. Figures are from ELRS master (espressif32 6.12.0 / arduino-esp32 2.0.17, espressif8266 4.2.1 / core 3.1.2), plus 3.6.4 for the STM32 receivers, which were removed in 4.0.

| Receiver MCU | Clock / divider | Set to 416666 | Set to 420000 (default) |
|---|---|---|---|
| ESP8285 | 80 MHz, whole-number divider, truncates | 416667 | 421053 (+1.05% vs spec) |
| ESP32 | APB 80 MHz, 1/16 fraction, truncates | 416667 | 420085 |
| ESP32-C3 / S3 | XTAL 40 MHz, 1/16 fraction, truncates | 416667 | 420223 |
| STM32F103 USART1 @72 (R9MM/R9mini, 3.x) | F1 HAL two-stage rounding | 416185 | 421053 |
| STM32F301/F373 USART2 @36 (3.x) | | 418605 | 418605 |

### FC-to-receiver mismatch

A 16x-oversampled STM32 UART tolerates about 3.41–3.75% of total deviation, depending on whether the divider has a fractional part (RM0090 Tables 144 and 145). Betaflight doesn't set ONEBIT, so those are the figures that apply.

| Scenario | Worst case (crystal-clocked FC) |
|---|---|
| Today: BF 420000 ↔ ELRS 420000 | ≤0.78% (≤0.58% on ESP receivers) |
| Today: BF 420000 ↔ receiver at exactly 416666 | 1.25% (F411 / overclocked F405 on APB1) |
| BF 416666 ↔ ELRS left at 420000 | 1.36% (F411/F7 APB1 ↔ ESP8285) |
| BF 416666 ↔ ELRS set to 416666 | ≤0.31% on ESP receivers (≤0.78% on the old F3 receivers) |

So with crystal clocks, every combination has more than 2% of margin. Switching only Betaflight makes things slightly worse for ELRS users who don't change anything (0.6% → 1.4%). Switching both sides gives the best result of all (≤0.31%), because every ESP-based ELRS receiver produces 416666 exactly. ELRS users can already choose this: the receiver web UI has a free-form "CRSF baud" field (`rcvr-uart-baud`).

### Where it can actually bite: RC-oscillator clocks

- **F4 and G4 boards without `SYSTEM_HSE_MHZ`.** Betaflight builds their PLL from the 16 MHz HSI unless `system_hse_mhz` is set. At the config commit Betaflight currently pins, that's 46 configs: 25 F405, 12 F411, 1 F446 and 8 G474. The F405 datasheet gives the HSI as ±1% at 25 °C, ±4% from −10 to 85 °C, and −8%/+4.5% from −40 to 105 °C. Across the temperature range that is already at or beyond the UART's tolerance on its own, whichever baud rate we choose. For those boards the right fix is configuring the crystal, not the baud rate.
- **N6 USART1** always runs from its 64 MHz HSI.
- **H5** falls back to the 64 MHz HSI if the crystal fails to start.
- **ELRS L432/L433 receivers** (3.x only) also run from internal RC oscillators.

### History

420000 started in Betaflight. Commit 818a91b6c (Martin Budden, November 2016) changed `CRSF_BAUDRATE` from 400000 to 420000. ExpressLRS later took it from there.

### Way forward

I'm in favour of the coordinated switch to 416666 you suggested. #15972 makes the change in Betaflight and keeps 420000 behind `USE_CRSF_LEGACY_BAUDRATE`. It has no milestone, so it can be lined up with an ExpressLRS release that defaults to 416666, plus a release note telling existing ELRS users they can set `rcvr-uart-baud` to 416666 now.

What's still open:
- **TBS receivers:** I couldn't find any public source for their MCU or clock source, and the CRSF spec doesn't state a tolerance.
- **Measurements:** everything above is calculated. Scope captures, especially from HSI-clocked F4/G4 boards, would be welcome.

haslinghuis, 2026-11-05:
@CapnBry fair points. The numbers I posted above answer some of them.

**Mismatched versions keep working.** Nothing turns into an "X only works with Y" matrix. Take a Betaflight FC at 416666 and an old ELRS receiver left at 420000. With a crystal-clocked FC, the worst case is 1.36% (F411/F7 APB1 with an ESP8285). The STM32 UART tolerates about 3.4–3.75%. The reverse case is the same: an old Betaflight at 420000 with a receiver at 416666 is at most 1.25%, and spec-rate receivers run like that today without complaint. Mismatched firmware loses some margin, but the link still works. That's quite different from the SPI ELRS case, where a version mismatch means no link at all.

The exception is F4/G4 boards with no crystal configured, which run from the HSI. There the RC oscillator's temperature drift alone can use up the budget, whichever baud rate we pick. The fix for those boards is setting `system_hse_mhz`, not the CRSF baud rate.

**What it gains.** I'll be honest: with crystal clocks the gain is margin, not a fix for a link that doesn't work. A matched pair at 416666 is within ≤0.31%, because every ESP-based ELRS receiver generates 416666 exactly. A matched pair at 420000 is within ≤0.58% on ESP receivers. Today, spec-rate receivers paired with Betaflight are at up to 1.25%. The other gain is that every CRSF implementation would agree on one number, so this doesn't come up again.

**The build define.** Point taken: it isn't something users should have to remember or type correctly. That's one reason #15972 has no milestone. It only makes sense together with ELRS, not as a Betaflight-only switch.

haslinghuis, 2026-11-05:
I've opened an alternative as a draft: #15978.

It replaces the build define with a runtime setting, `crsf_baudrate` (`AUTO` / `416666` / `420000`). `AUTO` is the default:
- It starts at 420000. Receivers that don't answer stay there, so nothing changes for them.
- ExpressLRS receivers (identified from their device info reply) stay at 420000.
- Other receivers move to the spec rate of 416666.
- It only switches while disarmed, and never while a CRSF V3 negotiated speed is active.

So nothing changes for existing ELRS setups, spec receivers get the spec rate, and nobody has to type a build define. It hasn't been bench-tested yet, so feedback and test reports are welcome.

hydra, 2026-11-05:
@haslinghuis yes! this is the exact kind of stuff that AI is great for (given the usual 'pinch of salt' of course).

I agree that @CapnBry has fair points indeed, I'm definitely not disputing them.

It's not just an ELRS issue though, since BF is already out-of-spec when talking to every single CRSF receiver that's NOT ELRS on the wrong baud rate, and IMHO that alone is worth fixing on principle regardless of market share, etc.

We now have a clearer picture on the effect of the change. 

As I suspected, the ELRS MCUs are actually better at using 416666 as there is less error, perhaps that's why TBS changed the spec too (confirm @tbs-trappy ? 👋 )

given the table in 'FC-to-receiver mismatch' above, it seems like the change should be made in BF and ELRS should follow too and then we can finally put this issue to bed where it belongs.

@haslinghuis in light of the new information and the investigation above and the PR can you re-open this issue please?

haslinghuis, 2026-11-13:
Agreed. TBS owns the CRSF spec, and the spec says 416666. Betaflight should follow it, and ELRS should follow too. The numbers above show that moving both sides gives the lowest error of any combination (≤0.31% on ESP receivers), and old ELRS receivers left at 420000 still work with plenty of margin (worst case 1.36% on crystal-clocked FCs).

I'll go ahead with #15972 (416666 by default, 420000 still available via `USE_CRSF_LEGACY_BAUDRATE`) and close #15978. ELRS users can set `rcvr-uart-baud` to 416666 today. @CapnBry, would ELRS be willing to switch its default to 416666 in a future release so the two can line up?

haslinghuis, 2026-11-16:
Updated https://github.com/betaflight/betaflight/issues/12398#issuecomment-6663455603

PR #13614 (2024-05-11): Update CRSF spec
- Fixes #12398

- According to spec CRSF uses a baudrate of 416666
- Please test carefully