betaflight/betaflight #12401: [SITL] BF4.5.0 Stuck in RX_FAILSAFE if rc packages are received during boot

AJ92, 2023-02-21:
### Describe the bug

If the SITL target is started while the UDP socket for rc packages is fed with data the flight controller is stuck in RX_FAILSAFE and cannot be rest with valid rc data.

I assume statements set in sitl.c L249-253 are overwritten somewhere after they where set. Maybe opening the UDP sockets has to be delayed, as reseting those statements again fixes the problem.

### To Reproduce

Missbehavior:
- spin up a UDP socket that sends valid rc data
- start SITL target
- connect to BFC and check under Setup -> Info -> Armig disabled Flags:  RX_FAILSAFE

terminal output:
`[SITL] The SITL will output to IP 172.28.144.1:9002 (Gazebo) and 172.28.144.1:9001 (RealFlightBridge)
[system]Init...
[SITL] init PwmOut UDP link to gazebo 172.28.144.1:9002...0
[SITL] init PwmOut UDP link to RF9 172.28.144.1:9001...0
[SITL] start UDP server @9003...0
[SITL] start UDP server for RC input @9004...0
start UDP server...0
[FLASH_Unlock] loaded 'eeprom.bin', size = 32768 / 32768
[timer]Init...
Initialized motor count 4
[SITL] new fdm 144 t:98.680000 from 0.0.0.0:0
[SITL] new rc 40: t:98.680000 AETR: 1497 1111 1111 1111 AUX1-4: 1111 1111 1111 1111
bind port 5761 for UART1
unusedPinsInit
New connection on UART1, 0
[NEW]UART1: 1,1`

current work around:
- first start SITL target
- wait until target booted up
- spin up a UDP socket that sends valid rc data
- connect to BFC and check under Setup -> Info -> Armig disabled Flags:  THROTTLE

terminal output:
`[SITL] The SITL will output to IP 172.28.144.1:9002 (Gazebo) and 172.28.144.1:9001 (RealFlightBridge)
[system]Init...
[SITL] init PwmOut UDP link to gazebo 172.28.144.1:9002...0
[SITL] init PwmOut UDP link to RF9 172.28.144.1:9001...0
[SITL] start UDP server @9003...0
[SITL] start UDP server for RC input @9004...0
start UDP server...0
[FLASH_Unlock] loaded 'eeprom.bin', size = 32768 / 32768
[timer]Init...
Initialized motor count 4
bind port 5761 for UART1
unusedPinsInit
[SITL] new rc 40: t:0.000000 AETR: 1111 1111 1111 1111 AUX1-4: 1111 1111 1111 1111
[SITL] new fdm 144 t:0.000000 from 0.0.0.0:0
New connection on UART1, 0
[NEW]UART1: 1,1`

notice position of first received rc package in both cases

### Expected behavior

running UDP socket does not trigger RX_FAILSAFE, or at least is cleared again.

### Support ID

```txt
-
```

### Flight controller

SITL 4.5.0 master@166ff9c9b362d4f6bac8430567152731015ee61e

### Add any other context about the problem that you think might be relevant here

running under WSL

haslinghuis, 2023-02-21:
@ahcorde can you replicate this behavior?

ridavv, 2023-03-01:
Im having this issue, my fc gives me rx_failsafe when all hardware is ok

Prajwal-Koirala, 2023-12-26:
This is an issue report about the Betaflight flight controller software getting stuck in failsafe mode when receiving RC (radio control) packets during startup. The issue occurs in the SITL (Software in the Loop) simulator. The reporter suggests that the UDP sockets for RC packets are being opened too early, and that this is causing the problem. As a workaround, the reporter suggests starting the SITL target first, then waiting until it has booted up before starting the UDP socket. 

Here are the key points:
* The Betaflight flight controller software can get stuck in failsafe mode when receiving RC packets during startup.
* This issue occurs in the SITL simulator.
* The problem is likely caused by the UDP sockets for RC packets being opened too early.
* There is a workaround that involves starting the SITL target first, then waiting until it has booted up before starting the UDP socket.

I hope this summary is helpful! Let me know if you have any other questions.

Tuxliri, 2024-01-08:
I can reproduce this, while working on [this](https://github.com/osrf/vehicle_gateway/pull/133) pr.

Tuxliri, 2024-01-08:
Actually it seems that I get a `segmentation fault` error sometimes, currently investigating the reason.

AJ92, 2024-01-08:
on my custom SITL version i don't have this problem anymore. could check later what is different in my fork.

closed 2026-09-26 by blckmn